NXPNFCLinux/linux_libnfc-nci #1: Example with Android Phone in HCE mode

diegosueiro, 2015-10-20:
Is there an example for using this lib or nfcDemoApp to read data from Android phone ain HCE mode?

I'm using the PN7120.

jgeslin, 2015-10-22:
The nfcDemoApp allows detecting remote contactless card and extract NDEF data if any. The android phone in HCE is emulating ISO14443-4 card. 
Then depending on the android application running on top of the HCE it can emulates NFC Forum type 4 tag to present NDEF data or implement dedicated application.

Do you expect to share NDEF data or something else ?

diegosueiro, 2015-10-22:
The idea is exchange data between PN7120 and an Android NFC enabled phone.
The communication must be bi-directional and the PN7120 should be the "master".

I believe I can use 2 approaches:
* PN7120 as a NFC reader and Android phone as a TAG (HCE);
* PN7120 and Android phone using P2P (SNEP on NXP side and Android Beam on phone side).

This is my first contact with NFC technology, so I'm a little bit lost in how to proceed with this.

jgeslin, 2015-10-22:
If you want to exchange data in both way I would advise you to use the PN7120 as reader and phone as HCE. Indeed the Android Beam allows to send or receive NDEF data but prevent to send after it has received !

From PN7120 side you'll have to use the nfcTag_transceive API to exchange following ISO7816-4 protocol (for application selection via AID). If needed I can provide an example of android app making use of HCE android API.

diegosueiro, 2015-10-22:
@jgeslin ,

Thanks for the information. It should be really great if you can provide me an example of android app that uses the HCE API.

Do you know if it is possible to have this communication encrypted?

I did some tests in P2P mode and I realized that I just can send data from Android to PN7120 if I depart and approach again.

jgeslin, 2015-10-22:
Please find the example there: https://nxp.box.com/s/ffz4phb8v8wnnsf80hxwho6e1ywgo95d.
It is an emulation of the first exchanges of VISA payment applet selection.

After the application selection procedure is done, you can define your own encryption procedure/key exchange.

diegosueiro, 2015-10-22:
@jgeslin,

Thanks. I'll take a look on that.

closed 2015-12-04 by jgeslin

lemimu, 2018-05-03:
Hi @jgeslin,
Could you share with me the example please?
Thanks, it will be very helpful for me.