NXPNFCLinux/linux_libnfc-nci #2: Trouble getting PN7120 to work

yaoz39, 2015-12-03:
The OM5577/PN7120S works with Rasperry Pi 2.

Then I ran into problems with a ARM based embedded system.

The driver is loaded fine and /dev/pn544 is created.  But when running nfcDemoApp, I get the following:
# ./nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

pn54x_dev_write : i2c_master_send returned -6
pn54x_dev_write : i2c_master_send returned -6
pn54x_dev_write : i2c_master_send returned -6
pn54x_dev_write : i2c_master_send returned -6
pn54x_dev_write : i2c_master_send returned -6
pn54x_dev_write : i2c_master_send returned -6

After changing all the debug levels to 0x03 in /etc/libnfc-nxp.conf and run it again, the output looks like:

# ./nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpTml:     Opening port=/dev/pn544

NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     phTmlNfc_i2c_reset(), VEN level 0
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     PN54X - Tml Reader Thread Started................

NxpTmlpn54x_dev_write : i2c_master_send returned -6
:     PN54X - Tml Writer Thread Started................

NxpTml:     PN54X - Tml Writer Thpn54x_dev_write : i2c_master_send returned -6
read Running................
NxpTml:
NxpTml:     PN54X - Read requested.....

NxpTml:pn54x_dev_write : i2c_master_send returned -6
     PN54X - Invoking I2C Read.....

NxpHal:     thread started
PN54X - Write requested..pn54x_dev_write : i2c_master_send returned -6
...

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
Npn54x_dev_write : i2c_master_send returned -6
xpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write messapn54x_dev_write : i2c_master_send returned -6
ge.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     write error status = 0x1ff
PN54X - Tml Writer Thread Running................

NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     write error status = 0x1ff
PN54X - Tml Writer Thread Running................

NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     write error status = 0x1ff
PN54X - Tml Writer Thread Running................

NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     write error status = 0x1ff
PN54X - Tml Writer Thread Running................

NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode (max count = 0x6)
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     phTmlNfc_i2c_reset(), VEN level 0
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpHal:     PN54X Reset - SUCCESS

NxpHal:     phNxpNciHal_write failed for hal ext
NxpHal:     Force FW Download, NFCC not coming out from Standby

Can you help to check what is wrong?

jgeslin, 2015-12-04:
Hi,

The "i2c_master_send returned -6" trace indicates something is going wrong at i2c level:
    - #define ENXIO        6  /* No such device or address */

How did you connected the OM5577/PN7120S demoboard to your ARM based embedded system (to which pin of your MCU is the board connected) ?
Did you updated accordingly platform_data or device_tree details (I2C bus used, GPIO assignements of both IRQ and VEN pins) ?

Regards,
Jeremy.

yaoz39, 2015-12-04:
Thank you for your quick reply.

There are two i2c buses in out embedded system.  The PN7120S demo board is connected to i2c bus 0.  There are also other devices on the i2c bus 0.  All the other i2c devices on bus 0 work fine.

The relevant device tree looks like
                ps7_i2c_0: ps7-i2c@e0004000 {
                        bus-id = <0>;
                        clock-frequency = <50000>;
                        clocks = <&clkc 38>;
                        compatible = "cdns,i2c-r1p10";
                        interrupt-parent = <&ps7_scugic_0>;
                        interrupts = <0 25 4>;
                        reg = <0xe0004000 0x1000>;
                        xlnx,has-interrupt = <0x0>;
                        xlnx,i2c-reset = "";

                        pn547: pn547@28 {
                                compatible = "nxp,pn547";
                                reg = <0x28>;
                                clock-frequency = <50000>;
                                interrupt-gpios = <&ps7_gpio_0 2 0>;
                                enable-gpios = <&ps7_gpio_0 3 0>;
                        } ;
                } ;

The clock frequency is reduced to 50K because of other devices in the i2c bus 0.

I also tried to use i2cdetect to see if pn7120 is responding.  i2cdetect shows all the other i2c devices on the bus, but not pn7120.  Even with Raspberry Pi, i2cdetect does not show pn7120, although the demo program works fine on Raspberry Pi.

Thanks again for your help.

yaoz39, 2015-12-04:
We made some progress today.  The i2c seems to be working and when using a scope, I see VEN toggles at the beginning of nfcDemoApp.  This is today's debug output:
# ./nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpTml:     Opening port=/dev/pn544

NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     phTmlNfc_i2c_reset(), VEN level 0
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     PN54X - Tml Reader Thread Started................

NxpTml:     PN54X - Tml Writer Thread Started................

NxpTml:     PN54X - Tml Writer Thread Running................
NxpTml:     NxpHal:
NxpTml:     thread started
PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

PN54X - Read requested.....

NxpTml:     PN54X - Invoking I2C Read.....

NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpHal:     hal_extns_write_rsp_timeout_cb - write timeout!!!
NxpHal:     Response timer stopped
NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
NxpHal:     NCI_CORE_RESET: Failed
NxpHal:     Power Cycle
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     phTmlNfc_i2c_reset(), VEN level 0
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpHal:     PN54X Reset - SUCCESS

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     write successful status = 0x0
PN54X - Tml Writer Thread Running................

NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpHal:     hal_extns_write_rsp_timeout_cb - write timeout!!!
NxpHal:     Response timer stopped
NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
NxpHal:     NCI_CORE_RESET: Failed
NxpHal:     Power Cycle
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     phTmlNfc_i2c_reset(), VEN level 0
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpHal:     PN54X Reset - SUCCESS

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpHal:     hal_extns_write_rsp_timeout_cb - write timeout!!!
NxpHal:     Response timer stopped
NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
NxpHal:     NCI_CORE_RESET: Failed
NxpHal:     Power Cycle
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     phTmlNfc_i2c_reset(), VEN level 0
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpHal:     PN54X Reset - SUCCESS

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     PN54X - Tml Writer Thread Running................

write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpHal:     hal_extns_write_rsp_timeout_cb - write timeout!!!
NxpHal:     Response timer stopped
NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
NxpHal:     NCI_CORE_RESET: Failed
NxpTml:     PN54X - Write request NOT enabled

jgeslin, 2015-12-07:
Hi,

Regarding your comment:
"I also tried to use i2cdetect to see if pn7120 is responding. i2cdetect shows all the other i2c devices on the bus, but not pn7120. Even with Raspberry Pi, i2cdetect does not show pn7120, although the demo program works fine on Raspberry Pi."
Indeed, the PN7120 has a standby mode, enabled by default. To leave this standby mode, the first I2C request must be retransmitted shortly so the PN7120 can answers the second time. Running I2cdetect a second time fastly allows to see the 0x28 address being acked.

jgeslin, 2015-12-07:
From your last logs, the first I2C write "20000101" is properly acknowledged by the PN7120 which means I2C bus is correctly connected as well as the VEN pin. You should then check the behavior of IRQ pin: it should raise up after "20000101" frame is sent to indicate a response is ready to be read.

What is the status of kernel debug trace ("dmesg") ?

yaoz39, 2015-12-08:
Thanks again for your help.

We checked the IRQ pin and made sure it is connected correctly.  We don't
know if it is active low or high by default though.

Attached the scope captures during "nfcDemoApp poll".  The file
nfc-initial.jpg is at the beginning of the demo app.  The nfc-fail.jpg is
when the app hangs.  The traces are: C1 - IRQ, C2 - SDA, C3 - VEN, C4 - SCL.

Also, with all the debug level on, the output of the app is in file
nfc-1.log.

It looks like the i2c read failed with timeout:
pn54x_dev_read: i2c_master_recv returned -110

Since the system locked up after running nfcDemoApp, I was not able to get
dmesg.

jgeslin, 2016-03-25:
I2C kernel driver issue, closed.

closed 2016-03-25 by jgeslin

rajesh181092, 2016-08-01:
why is it closed. I am facing the same problem. may i get the correction done for this issue

jgeslin, 2016-08-09:
This issue was not related to linux_libnfc-nci stack but due to I2C kernel driver of the platform.
In case you need help please open a new issue and post you logs with all available information.

libchap, 2017-11-20:
I'm facing the same or similar issue. In a custom board with PN7150 installed, the nfcDemoApp exits after trying three times like this:
```
NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpHal:     hal_extns_write_rsp_timeout_cb - write timeout!!!
NxpHal:     Response timer stopped
NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
NxpHal:     NCI_CORE_RESET: Failed
```
Please attach more information regarding the mentioned "I2C kernel driver issue" - which driver was used, how to detect its issue and when it was fixed (incl. evtl. link to its bug tracker). Thanks!

jgeslin, 2017-11-20:
Hi,

Author of the issue confirmed the problem was related to the I2C driver of the platform, explaining the decision to close it.

However, according to your log, you are facing same behavior. The Host can send the first command to PN71xx, which seems to be acknowledge but not answered. 
The first thing should be to check the behavior of the PN71xx IRQ signal: it should goes high slightly after the "20000101" command is sent, and remain until HW rest is applied (during recovery procedure).
If it indeed goes high it probably indicates an error in the GPIO assignment to the IRQ pin.
By the way, what configuration are you using (during configure step) ? --enable-i2c, --enable-alt or default one ?

Regards,
Jeremy.

libchap, 2017-11-21:
Hi,
Thanks much for such quick reply!
I'm using `--enable-alt` configuration. Today I found out my mistake: I configured `phTmlNfc_alt.h` like this:
```
#define I2C_BUS         "/dev/i2c-0" 
#define I2C_ADDRESS     0x28 
#define PIN_INT         23 
#define PIN_ENABLE      24 
```
but the IRQ and VEN pins on the PN7150 are not connected to any GPIO anyway.

Actually, I have no free GPIO to be used for this at all :(

Is there a way to communicate with PN7150 (using libnfc-nci library) using just the I2C, without any IRQ ?

Thanks for any suggestion!

jgeslin, 2017-11-21:
This is not recommended, anyway you could give it a try.
You'll have to connect PN7150 VEN pin to PVDD or VBAT and implement I2C read polling instead of IRQ monitoring. This is done in file src\halimpl\pn54x\tml\i2c\phTmlNfc_i2c.c, in function wait4interrupt().

Another possibility would be to use PN7150 USB dongle: https://shop.mikroe.com/nfc-usb-dongle. This is being supported by the libnfc-nci linux stack using --enable-lpcusbsio option.

libchap, 2017-11-22:
Thanks again! The physical connection of the chip was already correct: VEN pin to power source and IRQ pin to GND. Power consumption is not so big deal here. I modified the `wait4interrupt()` function to behave simply like `usleep(50 * 1000);` and it seems to be working somehow now (i just had to get rid of errno=79 error messages): after initialization, it prints
```
NxpFwDnld:  Clock frequency value is wrong in config file, setting it as default
BrcmHcpX8103
BrcmHcpR8180
BrcmHcpX810103020304
BrcmHcpR8180
BrcmHcpX81010156c6180006c61800
BrcmHcpR8180
BrcmHcpX810204
BrcmHcpR818000
Waiting for a Tag/Device...
```
and then it detects Mifare cards. However, sometimes it enters bug mode, printing
```
NxpFwDnld:  Clock frequency value is wrong in config file, setting it as default
NxpHal:     Sending DH and NFCC core connection command as raw packet!!
NxpHal:     Sending DH and NFCC mode set as raw packet!!
NxpHal:     Sending UICC Select Command as raw packet!!
NxpHal:     Sending Set Screen ON State Command as raw packet!!
NxpHal:     Sending discovery as raw packet!!
NxpHal:     Sending last command for Recovery 
NxpHal:     Invoking data callback!!
```
and no longer detecting anything. Do you have any idea how can I avoid this? I can only see that `phNxpNciHal_core_initialized()` was called with `*p_core_init_rsp_params == 1` but dunno what it means.

Additionally, do you know about any specific danger caused by reading I2C without catched IRQ?

Anyway, I see that I've diverted from the topics of this issue, if you'd like to move this discussion anywhere, please do.

jgeslin, 2017-11-22:
First of all you should prevent connecting PN7150 IRQ pin to GND, because PN7150 will drive this pin high to inform host of data to be read. So I recommend to let it floating if you don't use it !

Then, you may extend the timeout, to receive NCI answer, leading to recovery mechanism (I don't get your complete log but I am suspecting it). Update NFC_CMD_CMPL_TIMEOUT parameter (in src\libnfc-nci\include\nfc_target.h file) to 5 for instance.

libchap, 2017-11-23:
OK, the IRQ pin no longer connected to GND, but hanging.

I found the cause of the "bug mode": it was a result of reading a malformed message (probably wrong read when the chip was not ready). I filter these wrong messages by returning `-1` from `phTmlNfc_i2c_read()` if the message starts with `0xff`. Your suggestion of modifying `NFC_CMD_CMPL_TIMEOUT` did not have any positive impact.

Thanks again much for your support, you helped to make things working for me! :)

Anyway, could you please once more think of any (obvoius) dangers from using PN7150 like this (no IRQ connection, just I2C, every 50millis read error, discarding obviously malformed messages) ?

jgeslin, 2017-11-23:
Indeed, I confirm that I2C frame starting with 0xFF can be filtered since NCI protocol defines frame cannot start with this value.

Despite the CPU overload, I can only foresee one drawback, considering increased power consumption is not a problem for you: not being able to driver VEN prevent HW reset of the PN7150 which may be useful during recovery of the stack (in some case there can be mismatch in the state machines of the stack and the PN7150, then a recovery mechanism is implemented to fix those potential issues).

Regards,
Jeremy.

flgcpwd, 2019-06-26:
> We made some progress today. The i2c seems to be working and when using a scope, I see VEN toggles at the beginning of nfcDemoApp. This is today's debug output:
>
> # ./nfcDemoApp poll [...]

I have the same problem, VEN is not driven when i start nfcDemoApp. How did you do to get VEN well driven ?
Thanks.