NXPNFCLinux/linux_libnfc-nci #5: Polling frequently freezes or stops working after a few scans

traic, 2015-12-31:
I'm experiencing an issue where polling might not working when starting up the demo app, or will freeze after a few scans. I'm trying to isolate whether this is a hardware issue or software issue. Have you guy's experience this on a Raspberry Pi using the linux image that you provide?

jgeslin, 2016-01-04:
I never seen such behavior on the used platforms (Raspberyy Pi, beagle bone, and a couple of others).
You can easily activate traces for debug at runtime by setting NXPLOG_XXX_LOGLEVEL parameters to value 0x03 inside /etc/libnfc-nxp.conf file on the target.

traic, 2016-01-08:
ok will give that a try, maybe it's a power supply issue, or faulty reader.

jgeslin, 2016-03-25:
Any update on the issue ? Can we close it ?

jgeslin, 2016-08-17:
Closed because of no further feedback about this issue.

closed 2016-08-17 by jgeslin