NXPNFCLinux/linux_libnfc-nci #6: multi-user

ContribOpenSource, 2016-02-07:
Hello,

it seems that this library uses (many) global variables (EXAMPLES: gSyncMutex,
gTagCallback, etc.) and triggers some issues:
*) it is a library, so it would be helpful that all symbols have a restricted scope.
   It would avoid potential symbols clash/ambiguity for the linker.
   EXAMPLE: add a namespace to C++ objects dedicated to this library (like Mutex)
*) what if the same application process needs to handle several NFC devices?
   Besides callbacks stored in (single) global variables, there is no way to specify
   the device file handle (/dev/pn544 is hard-coded in: src/halimpl/pn54x/hal/phNxpNciHal.c).

Is it possible to fix these issues?

Moreover, from which version of Android (parts of) this library is based on?
https://android.googlesource.com/platform/packages/apps/Nfc/+/master/nci/jni/NativeNfcTag.cpp

Thanks for your help

jgeslin, 2016-02-08:
Hi,

We'll take into account your feedback about global variables.
Regarding multiple NFC devices handling, the target of this library is embedded systems where only one NFC device is considered. The library is designed to be based on the /dev/pn54x kernel driver explaining the hard link. However, the library being delivered as source, user can modify it if he wants to use outside the original definition.

This library is a porting of the NXP's Android delivery originaly based on AOSP (kitKat).

Regards

jgeslin, 2016-08-17:
Considered closed.

closed 2016-08-17 by jgeslin