NXPNFCLinux/linux_libnfc-nci #7: build time functionality inclusion/exclusion

ContribOpenSource, 2016-02-08:
Hello,

for embedded system, it may be useful to be able to include/exclude some functionalities
at build time (like kernel configuration for example) in order to limit footprint,
but also for security reasons (help to minimize vulnerabilities).

Is it possible to configure the library, like tag reader only for instance?

Thanks

jgeslin, 2016-02-09:
Compile flags exist which can be set (in the Makefile.am) to include/exclude some functionalities (e.g. NXP_NFC_NATIVE_ENABLE_HCE, NXP_HW_SELF_TEST...).
Furthermore, configuration files on the target (libnfc-brcm.conf and libnfc-nxp.conf) allows runtime setting of the library (e.g. technologies considered for discovery).

ContribOpenSource, 2016-02-09:
So what can be done in order to only compile source code for "tag reader only" feature?

Thanks

jgeslin, 2016-03-25:
The current version of the library doesn't provides this flexibility.

jgeslin, 2016-08-17:
Considered closed.

closed 2016-08-17 by jgeslin