NXPNFCLinux/linux_libnfc-nci #8: access to shared i2c bus

ContribOpenSource, 2016-02-08:
Hello,

How to protect access to i2c bus (if other devices, handled by other applications, use the same i2c bus) ?
Is there a way to handle this mutual exclusion in the library?

Thanks for your help

jgeslin, 2016-02-09:
The potential confilct has to be managed at the I2C driver level, not in the scope of the NFC driver which handle only one device.

ContribOpenSource, 2016-02-09:
> The potential confilct has to be managed at the I2C driver level

How kernel space will be able to know the end of transactions?
Userspace may use several and different system calls, like:
   _ioctl(fd, SLAVE_ADDRESS, ...)
   read()/write()
   ...
   read()/write()_
It is the entire sequence that need to be protected against concurrent access,
sequence that is done by userspace.

So how can we handle mutual exclusion for access to shared i2c bus with this library?

Thanks for your help

jgeslin, 2016-02-10:
Not sure I understand your concern. The NFC library access the /dev/pn54x kernel driver through dedicated Ioctl calls. The /dev/pn54x driver handles the request and uses I2C and GPIO drivers of the platform (direct calls to I2C and GPIO drivers functions). It protects against concurrent access to it ressources thanks to mutex. The I2C and GPIO drivers shall themselves protect against concurrent access to their own ressources.

jgeslin, 2016-08-17:
Considered closed because of no further feedback.

closed 2016-08-17 by jgeslin