NXPNFCLinux/linux_libnfc-nci #12: nfcDemoApp with PN7120 on Jetson TK1

jldempsey9, 2016-04-26:
I have an issue similar to #2 except I don't believe it is an I2C kernel driver issue.  I have tested this with a few user space applications and other I2C devices and they worked fine.  I decided to make a separate issue from #11 because I believe I have device tree configured correctly with VEN and IRQ.  

The issue: After 20000101 is sent and IRQ is pulled high, there is no activity on the I2C bus and I get write timeouts.  I have adjusted the timeout up to 10 seconds to make sure that it was sufficiently long, but that didn't fix the issue.  Here is the output of running the nfcDemoApp (I **did** configure with --enable-pn7120 when building):

> root@tegra-ubuntu:/home/ubuntu/linux_libnfc-nci# ./nfcDemoApp poll
> #########################################################################################
> ##                                       NFC demo                                      ##
> #########################################################################################
> ##                                 Poll mode activated                                 ##
> #########################################################################################
>                               ... press enter to quit ...
> 
> NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)
> 
> NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)
> 
> NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)
> 
> NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)
> 
> NxpHal:     Entering phNxpNciHal_init_monitor
> NxpHal:     Returning with SUCCESS
> NxpTml:     Opening port=/dev/pn544
> 
> NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
> NxpTml:     phTmlNfc_i2c_reset(), VEN level 0
> NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
> NxpTml:     PN54X - Tml Reader Thread Started................
> 
> NxpTml:     PN54X - Read requested.....
> 
> NxpTml:     PN54X - Invoking I2C Read.....
> 
> NxpTml:     PN54X - Tml Writer Thread Started................
> 
> NxpTml:     PN54X - Tml Writer Thread Running................
> 
> NxpTml:     PN54X - Write requested.....
> 
> NxpTml:     PN54X - Invoking I2C Write.....
> 
> NxpHal:     thread started
> NxpNciX:     len =   4 > 20000101
> NxpTml:     PN54X - I2C Write successful.....
> 
> NxpTml:     PN54X - Posting Fresh Write message.....
> 
> NxpTml:     PN54X - Tml Writer Thread Running................
> 
> NxpHal:     write successful status = 0x0
> NxpHal:     Response timer started
> NxpHal:     Waiting after ext cmd sent
> NxpHal:     hal_extns_write_rsp_timeout_cb - write timeout!!!
> NxpHal:     Response timer stopped
> NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
> NxpHal:     NCI_CORE_RESET: Failed
> NxpHal:     Power Cycle
> NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
> NxpTml:     phTmlNfc_i2c_reset(), VEN level 0
> NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
> NxpHal:     PN54X Reset - SUCCESS
> 
> NxpTml:     PN54X - Write requested.....
> 
> NxpTml:     PN54X - Invoking I2C Write.....
> 
> NxpNciX:     len =   4 > 20000101
> NxpTml:     PN54X - I2C Write successful.....
> 
> NxpTml:     PN54X - Posting Fresh Write message.....
[... 135 lines cut ...]
> [ 7248.506687] pn54x_dev_ioctl, cmd=1074063617, arg=1
> [ 7248.506701] pn544_enable power on
> [ 7251.619127] pn54x_dev_ioctl, cmd=1074063617, arg=1
> [ 7251.619656] pn544_enable power on
> [ 7251.822186] pn54x_dev_ioctl, cmd=1074063617, arg=0
> [ 7251.822665] pn544_disable power off
> [ 7252.025128] pn54x_dev_ioctl, cmd=1074063617, arg=1
> [ 7252.025601] pn544_enable power on
> [ 7255.138881] pn54x_dev_ioctl, cmd=1074063617, arg=1
> [ 7255.139047] pn544_enable power on
> [ 7255.342608] pn54x_dev_ioctl, cmd=1074063617, arg=0
> [ 7255.342775] pn544_disable power off
> [ 7255.545230] pn54x_dev_ioctl, cmd=1074063617, arg=1
> [ 7255.545717] pn544_enable power on
> [ 7274.433861] pn54x_dev_release : closing 10,41

Everything looks normal to me on the software side except for the timeouts.  The result of a logic analyzer on the signal is shown in the attached pictures:
[image]
[image]
[image]

Some info on my platform:
I am using the custom Grinch kernel, based on the 3.10.24 mainline kernel:
[grinch kernel from nvidia forums](https://devtalk.nvidia.com/default/topic/766303/jetson-tk1/-customkernel-the-grinch-19-3-8-for-jetson-tk1-developed/)
I am running Ubuntu 14.04 and my hardware is the Jetson TK1.

jgeslin, 2016-04-27:
Hi,

1. From the kernel logs we can see that 3 instances on the driver are loaded (the last 2 ones are furthermore not able to load properly because of the reserved GPIO from the first one). Can you check why 3 instances are loaded ?

2. Can you check the correct mapping of the irq_gpio line (GPIO_PU0 / gpio160) to the IRQ number 337 with respect to your platform ?

3. Is the GPIO_PU0 well defined as GPIO function in the pinmux section of the device tree ?

4. in the pn5xx_kernel module (pn5xx_i2c.c), you could verify if the issue relates to interrupt handling replacing the related piece of code by irq_gpio polling:

```
...
		while (1) {
			pn54x_dev->irq_enabled = true;
			enable_irq(pn54x_dev->client->irq);
			ret = wait_event_interruptible(
					pn54x_dev->read_wq,
					!pn54x_dev->irq_enabled);

			pn54x_disable_irq(pn54x_dev);

			if (ret)
				goto fail;

			if (gpio_get_value(pn54x_dev->irq_gpio))
				break;

			pr_warning("%s: spurious interrupt detected\n", __func__);
		}
...
```

by something like:

...
		while (1) {
			if (gpio_get_value(pn54x_dev->irq_gpio))
				break;
                        usleep(500);
		}
...

jldempsey9, 2016-04-27:
Hello,

1. I solved the 3 instances of the driver issue (my device tree had 3 nodes with the same "compatible" string).  Here is the new dmesg output when I load the driver:

> root@tegra-ubuntu:/home/ubuntu/nxp-pn5xx# dmesg
> [  503.768312] pn54x_dev_init
> [  503.768417] pn54x_probe
> [  503.768432] of_get_named_gpio_flags exited with status 161
> [  503.768436] of_get_named_gpio_flags: can't parse gpios property
> [  503.768444] pn544 1-0028: FIRM GPIO <OPTIONAL> error getting from OF node
> [  503.768448] of_get_named_gpio_flags exited with status 160
> [  503.768452] of_get_named_gpio_flags: can't parse gpios property
> [  503.768456] pn544 1-0028: CLKREQ GPIO <OPTIONAL> error getting from OF node
> [  503.768513] regulator_get() failed for (1-0028,nxp,pn54x-pvdd), -19
> [  503.800409] pn54x_get_pdata: could not get nxp,pn54x-pvdd, rc=-19
> [  503.815965] regulator_get() failed for (1-0028,nxp,pn54x-vbat), -19
> [  503.829273] pn54x_get_pdata: could not get nxp,pn54x-vbat, rc=-19
> [  503.835560] regulator_get() failed for (1-0028,nxp,pn54x-pmuvcc), -19
> [  503.842019] pn54x_get_pdata: could not get nxp,pn54x-pmuvcc, rc=-19
> [  503.848297] regulator_get() failed for (1-0028,nxp,pn54x-sevdd), -19
> [  503.854647] pn54x_get_pdata: could not get nxp,pn54x-sevdd, rc=-19
> [  503.860820] pn54x_probe: request irq_gpio 160
> [  503.860834] pn54x_probe: request ven_gpio 161
> [  503.861050] pn54x_probe : requesting IRQ 337
> root@tegra-ubuntu:/home/ubuntu/nxp-pn5xx# 

2. I am currently trying to verify that this mapping is correct, but the technical reference manual does not contain information for the IRQ numbers.  I'll keep trying to confirm this.
3. GPIO_PU0 is well defined as a GPIO function in the pinmux section of the device tree.  Furthermore, I can acess gpios 160 and 161 from user space via sysfs (/sys/class/gpio)
4. I replaced 
```
        while (1) {
            pn54x_dev->irq_enabled = true;
            enable_irq(pn54x_dev->client->irq);
            ret = wait_event_interruptible(
                    pn54x_dev->read_wq,
                    !pn54x_dev->irq_enabled);

            pn54x_disable_irq(pn54x_dev);

            if (ret)
                goto fail;

            if (gpio_get_value(pn54x_dev->irq_gpio))
                break;

            pr_warning("%s: spurious interrupt detected\n", __func__);
        }
```
with 
```
                while (1) {
                        if (gpio_get_value(pn54x_dev->irq_gpio))
                        {
                                pr_debug("gpio_get_value returned");
                                break;
                        }
                        udelay(500);
                }
```
in the pn5xx_i2c module.  I never get the "gpio_get_value returned" string that I'm expecting and the while loop never quits (nfcDemoApp won't quit with ctrl+c).  This confirms it to be an interrupt handling issue, correct?  Assuming 337 is the correct IRQ number, what can I do to debug it?

Thank you for your help

jgeslin, 2016-04-28:
`gpio_get_value(pn54x_dev->irq_gpio)` function call don't relates to interrupt handling.
If it never returns TRUE while the related GPIO line is HIGH, it means there is a problem with the GPIO pad configuration or the GPIO driver.

Are you able to read the GPIO value to HIGH from the user space ? By the way you should not be able to access GPIO_PU0 and GPIO_PU1 from user space when the pn5xx_i2c driver is loaded since it reserve those ressources for its usage.

jldempsey9, 2016-04-29:
Hi,

I was not able to access GPIO_PU0 and GPIO_PU1 while the driver is loaded (I unloaded the module first). But I found out that I was not able to read the IRQ GPIO value as HIGH from user space.  I switched the irq_gpio to a different pin and the demo app works.  It must have been a problem with the GPIO pad configuration or a hardware issue.  

Thank you for helping me on this issue, I could not have done it without your help.  Issue closed.

closed 2016-04-29 by jldempsey9