NXPNFCLinux/linux_libnfc-nci #13: Where the location of nfa_dta_api.h file?

mbdroid99, 2016-05-23:
Hello, I tried to compile linux **libnfc-NCI-R2.1** using **Android Studio 2.1.1** (NDK environment), I found some bugs because losing nfa_dta_api.h file.

Where the location of nfa_dta_api.h file? This file does not exist in your program.

Thank you,
mb

jgeslin, 2016-05-23:
Hi,
nfa_dta_api.h is useless. 
Indeed it is included from nfa_dta_int.h file but this one is not included from anywhere.
You should not get any error  when compiling the library.

FYI, this code is ported from Android AOSP from where nfa_dat_api.h file is missing as well.

Regards,
Jeremy.

mbdroid99, 2016-05-23:
Hello Jeremy, 

From nfa_dta_int.h I get error message:

Can't resolve type tNFA_DTA_FL_POLL_LISTEN in struct tNFA_DTA_CONFIG
Can't resolve type tNFA_DTA_EMVCO_PCD_MODE in struct tNFA_DTA_CONFIG

Can't resolve type tNFA_DTA_CBACK in struct tNFA_DTA_API_ENABLE

Can't resolve type tNFA_DTA_CFG_ITEM in struct tNFA_DTA_API_CONFIG
Can't resolve type tNFA_DTA_CFG in struct tNFA_DTA_API_CONFIG

I think the definition of the type mentioned above is in nfa_dta_api.h file because that file is missing. But when you say that nfa_dta_api.h useless, What the types above is useless,  can be eliminated?

Thank You,
edha

jgeslin, 2016-05-23:
What I mean is that nfa_dta_int.h should not be included from anywhere. So you could simply remove it from the directory tree.

mbdroid99, 2016-05-23:
Thank You, my problem solved.

closed 2016-05-23 by mbdroid99