NXPNFCLinux/linux_libnfc-nci #14: Read nfc service via native without having root android phone

mbdroid99, 2016-06-10:
1. What we can to read NFC service via native, without having to root android phone?

2. What we can use the NFC service that is already available on Android, without create self NFC services?

Thank You.

jgeslin, 2016-06-10:
If you want to integrate PN7120 support on Android recommendation is rather to use our Android related delivery based NFC service part of google AOSP.
More details here: [http://www.nxp.com/documents/application_note/AN11690.pdf](http://www.nxp.com/documents/application_note/AN11690.pdf)
- for Kitkat: [https://github.com/NXPNFCLinux/nxpnfc_android_kitkat](https://github.com/NXPNFCLinux/nxpnfc_android_kitkat)
- for Lollipop: [https://github.com/NXPNFCLinux/nxpnfc_android_lollipop](https://github.com/NXPNFCLinux/nxpnfc_android_lollipop)

mbdroid99, 2016-06-10:
Sorry, maybe this is different topic. I want to ask about nfc:

To read card in android phone, we can use [NFC API](https://developer.android.com/guide/topics/connectivity/nfc/nfc.html) with java. 

I want to protect communication between android phone (NFC reader) with card with native. How to read card in android phone with native NDK (C++) ? Thank You.

jgeslin, 2016-08-17:
Question not related to Linux libnfc-nci stack.

closed 2016-08-17 by jgeslin