NXPNFCLinux/linux_libnfc-nci #15: Issue running nfcDemoApp

traic, 2016-08-17:
I get the following error message when trying to poll an NFC card:

[  411.962799] pn54x_dev_init
[  411.965820] pn54x_probe
[  411.968414] pn54x_probe: request irq_gpio 51
[  411.972991] pn54x_probe: request ven_gpio 50
[  411.994415] pn54x_probe : requesting IRQ 211
[ 9398.014190] pn54x_dev_open : 10,58
[ 9398.017822] pn54x_dev_ioctl, cmd=1074063617, arg=1
[ 9398.022949] pn544_enable power on
[ 9398.231353] pn54x_dev_ioctl, cmd=1074063617, arg=0
[ 9398.236419] pn544_disable power off
[ 9398.441253] pn54x_dev_ioctl, cmd=1074063617, arg=1
[ 9398.446319] pn544_enable power on
[ 9398.552642] pn54x_dev_write : i2c_master_send returned -121
[ 9399.564300] pn54x_dev_ioctl, cmd=1074063617, arg=1
[ 9399.569396] pn544_enable power on
[ 9399.781250] pn54x_dev_ioctl, cmd=1074063617, arg=0
[ 9399.786315] pn544_disable power off
[ 9399.991241] pn54x_dev_ioctl, cmd=1074063617, arg=1
[ 9399.996307] pn544_enable power on
[ 9401.103851] pn54x_dev_ioctl, cmd=1074063617, arg=1
[ 9401.108947] pn544_enable power on
[ 9401.321258] pn54x_dev_ioctl, cmd=1074063617, arg=0
[ 9401.326324] pn544_disable power off
[ 9401.531250] pn54x_dev_ioctl, cmd=1074063617, arg=1
[ 9401.536285] pn544_enable power on
[ 9402.643829] pn54x_dev_ioctl, cmd=1074063617, arg=1
[ 9402.648925] pn544_enable power on
[ 9402.861267] pn54x_dev_ioctl, cmd=1074063617, arg=0
[ 9402.866333] pn544_disable power off
[ 9403.071258] pn54x_dev_ioctl, cmd=1074063617, arg=1
[ 9403.076324] pn544_enable power on
[ 9516.743927] pn54x_dev_release : closing 10,58

traic, 2016-08-17:
root@pcm051:~/linux_libnfc-nci# ./nfcDemoApp  poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated         [ 2121.529113] pn54x_dev_open : 10,58
                [ 2121.533355] pn54x_dev_ioctl, cmd=1074063617, arg=1
        ##
####[ 2121.539581] pn544_enable power on
#####################################################################################
                              ... press enter to quit ...

NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpTml:     Opening port=/dev/pn544

NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     [ 2162.413330] pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_res[ 2162.419250] pn544_disable power off
et(), VEN level 0
NxpTml:     [ 2162.633239] pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_res[ 2162.639160] pn544_enable power on
et(), VEN level 1
NxpTml:     PN54X - Tml Read[ 2162.755126] pn54x_dev_write : i2c_master_send returned -121
er Thread Started................

NxpTml:    [ 2162.765686] pn54x_dev_write : i2c_master_send returned -121
 PN54X - Read requested.....

NxpTml:     PN54[ 2162.775421] pn54x_dev_write : i2c_master_send returned -121
X - Invoking I2C Read.....

NxpTml:     PN54X [ 2162.785461] pn54x_dev_write : i2c_master_send returned -121
- Tml Writer Thread Started................

N[ 2162.795532] pn54x_dev_write : i2c_master_send returned -121

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1f[ 1956.173187] pn54x_dev_ioctl, cmd=1074063617, arg=0
f
NxpHal:     w[ 1956.178375] pn544_disable power off
rite_unlocked failed - PN54X Maybe in Standby Mode (max count = 0x6)
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     phTmlNfc_i2c_reset(), VEN level 0
NxpTml:     [ 1956.393157] pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_res[ 1956.399108] pn544_enable power on
et(), VEN level 1
NxpHal:     PN54X Reset - SUCCESS

NxpHal:     phNxpNciHal_write failed for hal ext
NxpHal:     Force FW Download, NFCC not coming out from Standby

jgeslin, 2016-08-17:
Hi,

From the traces, the issue looks to be related to the I2C communication. The I2C kernel driver raises issue 121 on the first write operation (#define EREMOTEIO   121 /* Remote I/O error */).
Can be that:
- PN7120 VEN pin is not properly managed (it shall be high to allow PN7120 operating) because of wrong GPIO assignement in device tree or platform data.
- I2C address defined in the device tree or platform data is not consistent with PN7120 I2C slave address selection

Which HW platform is used ?
How PN7120 is connected to the platform (I2C bus, GPIOs ...) ?
What's the device tree of platform definition ?

Good way to investigates would be to spy I2C communication together with VEN signal.

Regards,
Jeremy

traic, 2016-08-17:
Hey @jgeslin,

I updated the logs above, and I'm not getting the -121 any longer, and the i2s is able to write now. however, I am getting a write timeout. is your advice still valid on how I should proceed? I'm using the AM335x phycore chip and carrier board, the following is my platform data, which seems correct:

50B X_GPIO_1_9 IO 3.3 V AM335x GPIO_1_9 (IRQ using pin 50)
51B X_GPIO_1_8 IO 3.3 V AM335x GPIO_1_8 (VEN using pin 51)

Board info uses registered address 0x28

+/* Convert GPIO signal to GPIO pin number */
+#define GPIO_TO_PIN(bank, gpio) (32 * (bank) + (gpio))

+static struct pn544_i2c_platform_data nfc_pdata = {
+	.irq_gpio = GPIO_TO_PIN(1,19),
+	.ven_gpio = GPIO_TO_PIN(1,18),
+	.firm_gpio = GPIO_UNUSED,
+	.clkreq_gpio = GPIO_UNUSED,
+};

 	{
+		I2C_BOARD_INFO("pn547", 0x28),
+		.platform_data = &nfc_pdata,
+	},

traic, 2016-08-17:
[   36.382904] pn54x_dev_init
[   36.385925] pn54x_probe
[   36.388519] pn54x_probe: request irq_gpio 51
[   36.393066] pn54x_probe: request ven_gpio 50
[   36.414184] pn54x_probe : requesting IRQ 211
[   73.552886] pn54x_dev_open : 10,58
[   73.557617] pn54x_dev_ioctl, cmd=1074063617, arg=1
[   73.563934] pn544_enable power on
[   73.772064] pn54x_dev_ioctl, cmd=1074063617, arg=0
[   73.777984] pn544_disable power off
[   73.991943] pn54x_dev_ioctl, cmd=1074063617, arg=1
[   73.997894] pn544_enable power on
[   74.118835] pn54x_dev_write : i2c_master_send returned -121
[   75.134216] pn54x_dev_ioctl, cmd=1074063617, arg=1
[   75.139587] pn544_enable power on
[   75.351959] pn54x_dev_ioctl, cmd=1074063617, arg=0
[   75.357879] pn544_disable power off
[   75.571960] pn54x_dev_ioctl, cmd=1074063617, arg=1
[   75.577880] pn544_enable power on
[   76.695495] pn54x_dev_ioctl, cmd=1074063617, arg=1
[   76.701019] pn544_enable power on
[   76.911956] pn54x_dev_ioctl, cmd=1074063617, arg=0
[   76.917907] pn544_disable power off
[   77.131958] pn54x_dev_ioctl, cmd=1074063617, arg=1
[   77.137939] pn544_enable power on
[   78.255493] pn54x_dev_ioctl, cmd=1074063617, arg=1
[   78.261016] pn544_enable power on
[   78.471954] pn54x_dev_ioctl, cmd=1074063617, arg=0
[   78.477905] pn544_disable power off
[   78.691986] pn54x_dev_ioctl, cmd=1074063617, arg=1
[   78.697937] pn544_enable power on
[   90.254089] pn54x_dev_release : closing 10,58
root@pcm051:~#

traic, 2016-08-18:
I attached a more recent output:

root@pcm051:~# /usr/sbin/nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated         [ 1083.084045] pn54x_dev_open : 10,58
                [ 1083.088256] pn54x_dev_ioctl, cmd=1074063617, arg=1
        ##
####[ 1083.094604] pn544_enable power on
#####################################################################################
                              ... press enter to quit ...

NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpTml:     Opening port=/dev/pn544

NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     [ 1083.302337] pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_res[ 1083.308258] pn544_disable power off
et(), VEN level 0
NxpTml:     [ 1083.522277] pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_res[ 1083.528198] pn544_enable power on
et(), VEN level 1
NxpTml:     PN54X - Tml Reader Thread Started................

NxpTml:     PN54X - Read requested.....

NxpTml:     PN54X - Invoking I2C Read.....

NxpTml:     PN54X - Tml Writer Thread Started................

NxpTml:     PN54X - Tml Writer Thread Running................

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpHal:     thread started
NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpHal:     [ 1087.765808] pn54x_dev_ioctl, cmd=1074063617, arg=1
hal_extns_write_[ 1087.771331] pn544_enable power on
rsp_timeout_cb - write timeout!!!
NxpHal:     Response timer stopped
NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
NxpHal:     NCI_CORE_RESET: Failed
NxpHal:     Power Cycle
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     [ 1087.982238] pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_res[ 1087.988189] pn544_disable power off
et(), VEN level 0
NxpTml:     [ 1088.202270] pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_res[ 1088.208190] pn544_enable power on
et(), VEN level 1
NxpHal:     PN54X Reset - SUCCESS

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpHal:     hal_extns_write_rsp_timeout_cb - write timeout!!!
NxpHal:     Response timer stopped
NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
NxpHal:     NCI_CORE_RESET: Failed
NxpTml:     PN54X - Write request NOT enabled

traic, 2016-08-18:
root@pcm051:~# cat /sys/kernel/debug/gpio
GPIOs 0-31, gpio:
 gpio-6   (mmc_cd              ) in  lo
 gpio-31  (poweroff            ) in  hi

GPIOs 32-63, gpio:
 gpio-40  (nfc_ven             ) out hi
 gpio-41  (nfc_int             ) in  lo
 gpio-54  (bt_en
              ) out lo
 gpio-56  (wlan_en             ) out hi

GPIOs 64-95, gpio:

GPIOs 96-127, gpio:
 gpio-100 (rtc-tps65910-irq    ) in  lo

traic, 2016-08-18:
also, does the I2C_BOARD_INFO("pn547", 0x28) need to be named "pn544" instead since the device is mounted to /dev/pn544?

traic, 2016-08-19:
@jgeslin I enabled the PN7120 chip in the configure script for the driver, but I did not install the LLCP 3.0 protocol, do I need a secure element for this to work properly?

jgeslin, 2016-08-19:
LLCP1.3 support doesn't requires a secure element. It is using [openSSL component](https://www.openssl.org/) (must be installed on the platform and related path referenced at configuration step).
For more details about library setup with LLPC1.3 support please refer to the [documentation](https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/master/doc/AN11697%20-%20PN71x0%20Linux%20Software%20Stack%20Integration%20Guidelines.pdf).

traic, 2016-08-19:
@jgeslin right, but in general is the LLCP1.3 library required for the nxp nfc library to function properly? Can I run the library without LLCP just to do regular tag reading (non-P2P stuff)?

jgeslin, 2016-08-22:
The library supports by default LLCP1.2 which is enough for P2P communication between 2 devices but also with any NFC compatible smartphone. LLPC1.3 is adding security (P2P communication is then encrypted) between the devices. However this is optional and only takes place when the 2 devices support it. Otherwise the communication is done following LLCP1.2 specification (unencrypted).

jgeslin, 2016-09-01:
Can we consider this item closed ?

traic, 2016-09-14:
@jgeslin yes the issue ended up being a bad interrupt configuration on the board script, thx.

closed 2016-09-14 by traic