NXPNFCLinux/linux_libnfc-nci #16: Issue reading NFC tags

MarkoLainovic, 2016-08-22:
Hi,

I am unable to read/write NFC tags with this lib, and I was wondering if you could perhaps help me identifying the cause behind it.

Here's what works:

- drivers are properly loaded into kernel:
`pn54x_dev_init`
`pn54x_probe`
`pn544 0-0028: FIRM GPIO <OPTIONAL> error getting from OF node`
`pn544 0-0028: CLKREQ GPIO <OPTIONAL> error getting from OF node`
`0-0028 supply nxp,pn54x-pvdd not found, using dummy regulator`
`0-0028 supply nxp,pn54x-vbat not found, using dummy regulator`
`0-0028 supply nxp,pn54x-pmuvcc not found, using dummy regulator`
`0-0028 supply nxp,pn54x-sevdd not found, using dummy regulator`
`pn54x_probe: request irq_gpio 102`
`pn54x_probe: request ven_gpio 44`
`pn54x_probe : requesting IRQ 152`

- IRQ interrupts' count can be seen in /proc/interrupts, caused by pn54x

- library has been built and transferred to the target system, demo app has been successfully built,  executed and debug session has been successfully initiated.

- using the poll argument, i am able to read successfully the tag's UID and the tag's type:
`Type : 		Type A - Mifare Classic`
`NFCID1 :    	5E 22 87 1A `

- Polling the Android phone also works, lib recognizes the device, Android Beam activates and the received NDEF message is displayed over the target console:
`Device Found`
`    NDEF Message Received : `
`            Type :              'URI'`
`            URI :               'http://play.google.com/store/apps/details?id=at.mroland.android.apps.nfctaginfo&feature=beam'`
`    142 bytes of NDEF data received :`
`    91 01 56 55 03 70 6C 61 79 2E 67 6F 6F 67 6C 65 2E 63 6F 6D 2F 73 74 6F 72 65 2F 61 70 70 73 `
`    2F 64 65 74 61 69 6C 73 3F 69 64 3D 61 74 2E 6D 72 6F 6C 61 6E 64 2E 61 6E 64 72 6F 69 64 `
`    2E 61 70 70 73 2E 6E 66 63 74 61 67 69 6E 66 6F 26 66 65 61 74 75 72 65 3D 62 65 61 6D 54 `
`    0F 22 61 6E 64 72 6F 69 64 2E 63 6F 6D 3A 70 6B 67 61 74 2E 6D 72 6F 6C 61 6E 64 2E 61 6E `
`    64 72 6F 69 64 2E 61 70 70 73 2E 6E 66 63 74 61 67 69 6E 66 6F `
`Device Lost`

What doesn't work:

- reading tags never works, the complete message is always:  

`NFC Tag Found`
    `Type :      'Type A - Mifare Classic'`
    `NFCID1 :        '5E 22 87 1A '`
    `NDEF Content : NO`
    `RAW Tag transceive failed`
`NFC Tag Lost`

- trying to write also fails in similar fashion.

Two things I've noticed:

- while debugging, I was able to observe that, before the library even checks for NDEF content, NfcTag::resetTechnologies() and similiar function calls happen in between, effectively erasing all the information about the tag, and by the time library actually checks for any content, all the necessary fields are zeroed out; it might be that my debugging session slows the process down way too much, but all in all, I was never able to reach the part of the library where a tag's NDEF content is actually read. In other words, function `nfcTag_isNdef()` always returns 0, since NfcTag object fields used for various checks are always zeroed out.

- Over at the debug console, i am receiving a ridiculous amount of "spurious interrupt" messages all the time:
`pn54x_dev_read: spurious interrupt detected`

Could any of this info help you to identify the issue with reading tags? Tags are formatted and contain simple text NDEF records, and this has been confirmed by several Android apps and a couple of different smartphones.

Here's a thread I made over at NXP community forums, no luck so far:
https://community.nxp.com/thread/432818

Thanks very much!
Marko

jgeslin, 2016-08-23:
Hi,

Regarding the "spurious interrupt" messages. It comes from the pn5xx_i2c kernel driver, whenever the PN71xx IRQ pin related interrupt is fired but the pin is not High. This may be due because the interrupt line is shared with other source. Anyway this should not have any impact on functionality.

Regarding "nfcTag_isNdef()" function returning 0 with those tags, for deeper analyzis please provide more traces of the scenario. To do this update /etc/libnfc-nxp.conf file on the target to set all "NXPLOG_xxx_LOGLEVEL" flags to 0x03 instead of 0x00.

Regards,
Jeremy.

MarkoLainovic, 2016-08-23:
Hey Jeremy,

Thanks very much for your response!

I've updated the configuration file and included for you three gists below:

- the first one is the part of the output while trying to scan the tag (unsuccessfully):
https://gist.github.com/MarkoLainovic/71d6d85096311d285ad65c7414fc51d0

- the second one is the part of the output while trying to scan the device (successfully):
https://gist.github.com/MarkoLainovic/a119dba4195de68863cd3c8946a92160

- the third gist is the .xml file I exported from an Android app, to show you the content of the tag I am trying to scan. It contains one text record with the text "jelradisad", you can search for it in the gist:
https://gist.github.com/MarkoLainovic/26f5a287a27e44659f79be78e91c6237

Thanks again for any help, much appreciated!
Marko

MarkoLainovic, 2016-08-23:
Just a small update:

Here's an updated gist for unsuccessful tag reading with API traces enabled as well:
https://gist.github.com/MarkoLainovic/9a0dc6762984c80aefba25c3f63a0abd

MarkoLainovic, 2016-08-31:
Hi, I'd just like to report that the issue has been solved!

It was hardware-related; the module hasn't been receiving proper battery power supply, which caused it to work  unpredictably, i.e. I2c interface was working well and P2P would sometimes work, tag reading would never work, etc.

Thanks again for you help,
Marko

closed 2016-08-31 by MarkoLainovic

lucar91, 2020-02-21:
Hi, 
I have an error when I try to use the nfcDemoApp to read a tag. I'm working in a 'imx6' device, with my controllers (pn7150) and my tag (ntag216 20mm 144byte).  Sometimes he reads and sometimes he doesn't.

**The error:**
[image]

**The log file (log_level set to 0x01):**

NxpTml:     System already has pin /sys/class/gpio/gpio125

NxpTml:     System already has pin /sys/class/gpio/gpio133

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpHal:     phNxpNciHal_deriveChipType HwVersion : 0x88
NxpHal:     phNxpNciHal_deriveChipType NxpNci > Product : PN548C2
NxpHal:     i2c_fragmentation_status existing
NxpFwDnld:  Ignoring Config :NFC FW Binary Path:

NxpFwDnld:  Ignoring Config:NFC FW Binary :libpn548ad

NxpFwDnld:  NFC FW Binary :/libpn548ad_fw.so

NxpFwDnld:  NULL handler : unable to load the binary file, verify binary path
NxpFwDnld:  Image extraction Failed - invalid imginfo or imginfolen!!
NxpFwDnld:  Error loading libpn54x_fw !!

NxpFwDnld:  Freeing Mem for Dnld Context..
NxpHal:     phNxpNciHal_deriveChipType HwVersion : 0x88
NxpHal:     phNxpNciHal_deriveChipType NxpNci > Product : PN548C2
NxpFwDnld:  Clock frequency value is wrong in config file, setting it as default
NxpFunc:    initialize:HOST_LISTEN_TECH_MASK=7;
NxpFunc:    RoutingManager::nfaEeCallback: unknown event=9 ????
NxpFunc:    nativeNfcTag_doTransceive: already deactivated
NxpFunc:    nativeNfcTag_doTransceive: already deactivated

**The problem** occurs when the tag is not positioned correctly (but is still in the direction of the antenna) or when tag is at the limit distance (3-4cm, read/not read tag content).
Why he read UID but doesn't read  the content? Can it be related to a library stack problem or is a NFC  technology problem related to distance?

I also attach the file with log_level 0x03
[debug_error.log](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/4235456/debug_error.log)

jgeslin, 2020-02-24:
Hi,

Unfortunately the complete log (log_level 0x03) doesn't match the scenario detailled (there is no discovery of any card nor any "nativeNfcTag_doTransceive" call in it). This prevent further analysis.

However, I can easily consider that the card UID is being reported (because the card was being properly activated) but further communication to read the tag content fails if the tag is at the limit of the reading distance.
Indeed, the complete card activation is done in three RF exchanges within less than 1ms (and only reported if activation completed), while the card reading occurs a few hundred of milliseconds later (library execution time). So the card reading may fail while the card activation succeed.

Regards,
Jeremy.

lucar91, 2020-02-24:
Thank you, I was just reading the various posts related to distance ( #68 ), to check the possibility of improving the reader sensitivity and something about speeding up the detection ( #48) (correct me if I'm wrong).  I guess there's no way, through the library, to increase the coverage range of the NFC controller, right?

jgeslin, 2020-02-25:
Hi,

If you are using PN7150 demokit (OM5578) the settings delivered by default in the library (NXP_RF_CONF_BLK_1 parameter from libnfc-nxp-pn548.conf file) are optimized for the distance, but if you are using your own design you can refer to [AN11755](https://www.nxp.com/docs/en/application-note/AN11755.pdf) to learn how to optimize those settings.

Regarding speeding the detection, by default the library is configured to search for all NFC technologies at a frequency of 2Hz (every 500ms). You can change this by updating the technologies the library poll for (for instance if you are only interested in detecting Ntag216 you can set parameter POLLING_TECH_MASK from libnfc-nci.conf file to 0x01), and you can increase the polling frequency by defining parameter NFA_DM_DISC_DURATION_POLL in libnfc-nci.conf file (for instance to increase the frequency to 4Hz, add "NFA_DM_DISC_DURATION_POLL=256" in libnfc-nci.conf file).

Regards,
Jeremy.