NXPNFCLinux/linux_libnfc-nci #17: Using the API to enter Low Power Discovery Mode

MarkoLainovic, 2016-08-31:
Hi,

Is there an easy way to configure PN7120 for Low Power Discovery mode, using the API?

From the application note AN11562 and the User Manual, I see a lot of parameters which need to be modified, but I fail to see how to do this programatically; I cannot find any of these commands in the sources.

Thanks!

jgeslin, 2016-08-31:
There is no API for configuring the Low Power discovery mode. This is done through the libnfc-nxp.conf configuration file.
In the file provided as reference for PN7120, the field NXP_CORE_CONF_EXTN already contains a default configuration:
NXP_CORE_CONF_EXTN={20, 02, 11, 04,
    A0, 5E, 01, 01,
    A0, 40, 01, 01, 
    A0, 41, 01, 04,
    A0, 43, 01, 00
}

A0, 40, 01, 01 <- indicates Tag Detector enabled
A0, 41, 01, 04 <- set threshold to 4 (somehow in the middle of the sensitivity)
A0, 43, 01, 00 <-  disable the hybrid mode

So by default you are already running in Low Power Discovery mode. The only thing you could wish is to tune the threshold in order to optimize the distance of tag detection with a trade off on power consumption. For this the AN11562 provides guidelines.

MarkoLainovic, 2016-08-31:
Ah, I see!

I must say these configuration options look a bit intimidating, but I'm getting a hang of it.
Is there a list of ALL the configuration options we can use with .conf files, along with their descriptions? I haven't managed to find it, and I could use it very much

P.S. Also, if you don't mind, I'd just like to ask you one more thing, unrelated to this issue; I saw that PN7120 doesn't support the encryption functionality, so currently I'm encrypting data manually and then I'm writing it down as plain text to the card. Could you perhaps suggest a better way? Is there one?

I'm still new to this, but I've seen that a lot of newer generation cards support AES encryption. Currently, I am only in possession of MIFARE Classic tag cards. Would that mean that, if I should use MIFARE DESFire cards, for example, data encryption would be performed by the tag microcontroller, once the data-transfer is finished, and saved as such? In that case, how should I configure PN7120 to be the only one able to receive decrypted data?

Thanks in advance!
Marko

jgeslin, 2016-09-01:
Regarding configuration files, there is a small description of the fields in [AN11697](https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/master/doc/AN11697%20-%20PN71x0%20Linux%20Software%20Stack%20Integration%20Guidelines.pdf) (chapter §4.4). Additionally you can find PN7120 parameters description in the related User Manual [UM10819](http://www.nxp.com/documents/user_manual/UM10819.pdf).

Regarding Encryption, PN7120 only embeds MIFARE classic crypto. This allows encrypting data which are going over the RF link to/from the card (data are going in plain on the I2C link). For MIFARE DESFire, the crypto has to be handled on the host side (data are encrypted on the I2C link and forwarded as is on the RF link) and are decrypt/encrypted by the card (key exchange must be done prior to this).

jgeslin, 2016-10-12:
Can we consider this topic closed ?

MarkoLainovic, 2016-10-18:
Hi,
Sorry for the late response.
Yes, I forgot to mark the issue as closed, since LPCD mode seems to be working well.
Thanks again for your help!

closed 2016-10-18 by MarkoLainovic