NXPNFCLinux/linux_libnfc-nci #18: Not compatible with newer OpenSSL (1.0.2g tested)

r1mikey, 2016-09-23:
OpenSSL no longer exposes the ec_lcl.h header, and therefore no longer exposes the internals of the EC_KEY structure, which is used in nfc/include/cipher.h and nfc/cipher/cipher.c.

This prevents this library compiling against newer OpenSSL versions as used in the Yocto project (and anybody else on the OpenSSL security update train).

Any ideas, or in-progress patches?

jgeslin, 2016-10-12:
Sorry for the late response.
Please find attached propose patch: 
[patch_OpenSSL.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/524786/patch_OpenSSL.txt)

Your feedback about the patch would be really appreciated.

r1mikey, 2016-10-17:
Brilliant!

I'll apply this to our tree.  We're going through EVT at the moment, and
we're struggling to get the clocks up on our design (our hardware guys are
chatting with NXP via our ODM to resolve), so I'll be able to give proper
feedback on the patch once we have those issues resolved.

Thanks again.
Michael

jgeslin, 2016-12-07:
Any feedback ?
Can I close this issue ?

r1mikey, 2016-12-08:
I can confirm that it fixes the build issue, but we're still having problems on our hardware, so I haven't been able to test it yet.

I'm OK for this to be closed.

closed 2016-12-08 by r1mikey