NXPNFCLinux/linux_libnfc-nci #22: support for both PN7120 & PN7150 ?

ContribOpenSource, 2016-12-05:
Hello,

> from: AN11697 PN71x0 Linux Software Stack Integration Guidelines
> chapter 4.2.2 Generating the Makefile
>    --enable-pn7120 or enable-pn7150: enable NFC Controller type to PN7120 or
>    PN7150. Default NFC Controller selection is PN7120.
>    ...
>    Pay attention to properly generate the Makefile according to the targeted NFC
>    Controller IC using appropriate “–enable-xxx” option of the configure script. Wrong
>    option may lead to unexpected behavior during code execution.

Moreover, if we try to enable both chips: ./configure --enable-pn7120 --enable-pn7150
> configure: error: Can not enable multiple chip type

So it seems that libnfc-nci can not be compiled in order to handle both controllers (PN7120 & PN7150), right ?
What if we use a custom board, but with different versions/features (one based on PN7120 and another based on PN7150): do we have to compile this library twice ?

Regards,
Contrib

jgeslin, 2016-12-06:
Hi,

Indeed the library is built with the support of only one controller type. There is no runtime detection of the controller type.

Regards,
Jeremy

jgeslin, 2018-08-13:
Latest release R2.4 now dynamically detect the NFC Controller. There is no more selection required at configure step.

closed 2018-08-13 by jgeslin