NXPNFCLinux/linux_libnfc-nci #24: nfcManager_enableDiscovery() vs. NFA_SetRfDiscoveryDuration()

ContribOpenSource, 2016-12-20:
Hello,

it seems that calls to **nfcManager_enableDiscovery()** trigger a call to **NFA_SetRfDiscoveryDuration()** with a different value.
CONTEXT: application is a **tag reader only**, using lib R2.1, based on this "skeleton":
1. **nfcManager_doInitialize()** and **nfcManager_registerTagCallback()** are called once at application startup
1. polling is started in reader only mode:
         **nfcManager_enableDiscovery(DEFAULT_NFA_TECH_MASK, reader_only_mode=1 , hce=0,  restart=0)**
1. a breakpoint in **NFA_SetRfDiscoveryDuration()** shows that argument is equal to: **200**
1. few seconds later, radio is disabled: **nfcManager_disableDiscovery()**
1. few seconds later, redio is re-enabled, with the same call used in step **(2)**. This time, breakpoint hit shows a different value: **500**
1. each time enable/disable discovery cycle is done, **NFA_SetRfDiscoveryDuration()** is alternatively called with either 200 or 500 as argument.

Q1) is it normal ? **NFA_SetRfDiscoveryDuration()** seems to be used in order to set NCI parameter **TOTAL_DURATION**, right ?

Q2) in code path, **NFC_NXP_NOT_OPEN_INCLUDED** macro is defined (by default): what is this feature ?

Thanks,
Contrib

jgeslin, 2016-12-21:
Hi,

Q1: Indeed the function NFA_SetRfDiscoveryDuration() sets the TOTAL_DURATION NCI parameter. FYI, the Linux_libnfc-nci is a porting of the Android AOSP libnfc-nci to linux. The reported behavior is also present in the AOSP stack from google. If you wish to change this I would suggest to just reset the flag "sReaderModeEnabled" inside the nativeNfcManager_disableDiscovery() funtion. From my mind this is missing from the Android stack.

Q2: The pieces of code, included thanks to NFC_NXP_NOT_OPEN_INCLUDED compile flag, are additions to the Android AOSP NFC stack for the support of NXP's NFC Controllers proprietary features (for instance support of MIFARE). The recoomendtion is to keep this flag defined to benefit of the best experience with NXP's NFC Controllers.

Regards,
Jeremy.

ContribOpenSource, 2016-12-22:
Hi,

Thanks.

Q3: do you plan to fix [Q1] (= TOTAL_DURATION toggling) in future release ?

Q4: what is the best duration for a tag reader only: 200 or 500ms ?

Regards,
Contrib

jgeslin, 2016-12-23:
Hi,

Q3: yes we plan to fix, as suggested by reseting "sReaderModeEnabled" flag in nativeNfcManager_disableDiscovery() funtion, in future release.

Q4: The TOTAL_DURATION configuration depends on the targetted application requirements. Basically it is a trade off between fastness reaction to device detection and power consumption. Indeed the NFC function power consumption significantly relates to the discovery loop frequency (you can refer to [AN11757](http://www.nxp.com/documents/application_note/AN11757.pdf) for more information). FYI  portable NFC devices, powered from batteries, usually implement a TOTAL_DURATION between 300ms and 500ms.

Regards,
Jeremy.

jgeslin, 2018-08-14:
Fix integrated in latest delivery.

closed 2018-08-14 by jgeslin