NXPNFCLinux/linux_libnfc-nci #25: How to get NDEF text record size?

ContribOpenSource, 2016-12-22:
Hello,

Q1: In lib R2.1, how to know the exact size of NDEF text record content? It seems that only NDEF message size is available in ndef_info_t (text content is included in this value, but how to get it? call to strlen() may not be enough if NUL chars are included in the middle and all raw bytes need to be read. So is it possible to get text data size contained in a NDEF Text record ?)

Q2) even if buffer sizes are defined in nfcTag_readNdef() and in ndef_readText(), buffer boundaries are ignored : do you plan to fix it in future release?

Thanks,
Contrib

jgeslin, 2016-12-23:
Hi,

Q1: Indeed the nfcTag_isNdef() API only returns, inside ndef_info_t structure, the Ndef message length. To know the Text field length you need to parse the Ndef message. This is the purpose of nfcTag_readNdef() API (to retrieve the Ndef message content and type), and then ndef_readText() API (in case Ndef message is of Text type).

Q2: I'm not sure to understand your concern about buffer boundaries. Indeed, the application is responsible to provide buffer of size enough to contain the targeted data (Ndef message or Text field). In case the buffer size indicated as APIs parameter is lower than the required length to store the data, the API will not fill the buffer and will return with an error.

Regards,
Jeremy.

ContribOpenSource, 2016-12-23:
Hi,

About your answer to Q1:
ndef_readText() is just calling ans returning nativeNdef_readText() function. There is no way to get text content size because only OK or FAIL is returned.

About Q2:
Again, in nativeNdef_readText(), there is no check on buffer boundaries.

Regards,
Contrib

jgeslin, 2016-12-27:
Hi,

Q1: Currently the only way to retrieve safely the Text field length is by parsing the ndef message inside the application itself. We'll consider updating the readText API in further release to make it returning the field length.

Q2: Indeed, the boundaries are not verified in readText while there are in readNdef. This will be fixed in further release.

Regards,
Jeremy.

jovere, 2017-09-19:
Hi,

I'm not entirely sure if this is the same request of Contrib, however, I would like to know the number of bytes read from the ndef message when calling functions such as `ndef_readText()`, `ndef_readUrl()`, etc.  Right now, they return either `0` for success or `-1` for failure.  I would suggest a positive number of read bytes on success.  This is because there can be more than one record in a message.

Right now, I can only decode the first record, since I don't know where the first one ends and the second one begins.  However, the raw data in the demo clearly shows that all the data is there.

If I am not mistaken, changing the `nativeNdef_readText`, `nativeNdef_readUrl`, etc. functions from returning 0 to returning `payloadLength` would make this possible.

Regards,
Jeremy

jgeslin, 2017-09-25:
Hi,

Please find attached proposed patch allowing ndef_readText() and ndef_readUrl() functions to return the content length (text or URL) -> [patch_returnLength.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1329876/patch_returnLength.txt)

For others related APIs (ndef_readHandoverSelectInfo and ndef_readHandoverRequestInfo) it doesn't make sense since the output is a structure with potentially two separated content (Wifi and bluetooth information).

Regards,
Jeremy.

jovere, 2017-09-25:
Hi Jeremy,

Thank you for looking at this.

Unfortunately, I don't believe that will work.  I need to know the number of bytes *read*, not the number of bytes written.  The header needs to be included in that number of bytes in order to read the next record.  I suggest leaving the null terminator and sending the *read* bytes, like the following:
 [0001-Reading-text-url-now-returns-bytes-read-to-allow-rea.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1330165/0001-Reading-text-url-now-returns-bytes-read-to-allow-rea.txt)

Regards,
Jeremy

jgeslin, 2017-09-26:
I understand know your request: being able to parse several records inside an NDEF message.
Then I would suggest to first identify the NDEF records locations (using MB/ME and Type Length/Pyyload Length fields) and then call library APIs on the NDEF record instances.

jovere, 2017-09-26:
In that case, there should be an API function which gives me an NDEF record, such as
```uint8_t * GetNdefRecord(uint8_t *buffer, size_t *size)```
and returns the pointer and size of the record.

All of the data for doing that is internal to the library, the user of the library shouldn't need to reinvent that code.

jgeslin, 2018-08-13:
No updates so considered closed.

closed 2018-08-13 by jgeslin