NXPNFCLinux/linux_libnfc-nci #26: infinite loops when no NFC device (100% CPU)

ContribOpenSource, 2016-12-27:
Hello,

in libnfc-nci R2.1, there are 2 infinite loops, with 100% CPU used, when:
- NFC device is not plugged
- pn5xx_i2c driver is loaded
- nfcDemoApp is launched

This misbehaviour can also be reproduced on a computer thanks to a dedicated pn5xx_i2c_stub driver.
This stub does not use i2c bus (so can be executed on any computer) and always returns error in read()/write() syscalls.

These infinite loops are executed by i2c read thread (= phTmlNfc_TmlThread() function):
   https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/R2.1/src/halimpl/pn54x/tml/phTmlNfc.c#L325

First infinite loop occurs because this thread always re-executes itself when error is returned by phTmlNfc_i2c_read(),
so retries are not limited (and no waiting time in the path, that's why 100% CPU is used):
   https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/R2.1/src/halimpl/pn54x/tml/phTmlNfc.c#L367
After limiting these retries and returning an error (in tTransactionInfo.wStatus),
problem was still not fixed...

...because a second infinite loop occurs, triggered by callback: phNxpNciHal_read_complete()
*) in this callback, upper layer was not informed of i2c read error:
   https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/R2.1/src/halimpl/pn54x/hal/phNxpNciHal.c#L921
*) a new call to read is set, with itself as callback (which creates an infinite chain of calls):
   https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/R2.1/src/halimpl/pn54x/hal/phNxpNciHal.c#L932
There are many levels of callbacks, semaphores, etc. so a summary is below:
phTmlNfc_i2c_read() [libnfc-nci/src/halimpl/pn54x/tml/phTmlNfc.c]
phTmlNfc_DeferredCall()
   --> phDal4Nfc_msgsnd()
          --> sem_post(&pQueue->nProcessSemaphore);
              => phDal4Nfc_msgrcv(): sem_wait(&pQueue->nProcessSemaphore);
                    <-- phNxpNciHal_client_thread():
                           case PH_LIBNFC_DEFERREDCALL_MSG:
                              phTmlNfc_ReadDeferredCb() [libnfc-nci/src/halimpl/pn54x/tml/phTmlNfc.c]
                                 --> gpphTmlNfc_Context->tReadInfo.pThread_Callback(gpphTmlNfc_Context->tReadInfo.pContext, pTransactionInfo);
                                     = phNxpNciHal_read_complete() {for caller: phNxpNciHal_open()}

Dedicated pn5xx_i2c_stub driver + bug fix patch will be added soon.
nfcManager_doInitialize() will then return an error when NFC device is not plugged (without this patch,
nfcManager_doInitialize() did not return and caller was "stuck").

Q1) Can you please review this patch, and fix this bug in future release ?

Q2) What if an error occurs in nfcManager_enableDiscovery() call: there is no return value, so how caller can be informed of this problem ?

Q3) idem for nfcManager_disableDiscovery() ?

Regards,
Contrib

ContribOpenSource, 2016-12-27:
proposed fix in: https://github.com/NXPNFCLinux/linux_libnfc-nci/pull/27

jgeslin, 2016-12-28:
Hi,

Thanks for the valuable feedback.

Q1: We'll consider proposed patch for integration in further release.

Q2: Indeed there is today no return code. We'll consider add this in next release. As an intermediate solution you could make use of an internal function, declaring it this way:
`extern BOOLEAN isDiscoveryStarted();`
It returns the status whether discovery is running or not.

Q3: same as for Q2

Regards,
Jeremy.

jgeslin, 2018-08-13:
Latest release R2.4 now includes the fix.

closed 2018-08-13 by jgeslin