NXPNFCLinux/linux_libnfc-nci #28: Type 2 Tag only

ContribOpenSource, 2017-01-11:
Hello,

Q1) with lib R2.1, how to exclude tags that are not NFC Forum Type 2 Tag ?

Q2) in nfcManager_enableDiscovery(), techMask shall be NFA_TECHNOLOGY_MASK_A, right?

Q3) In ndef_info_t there is "technology" field, do we have to check both: TARGET_TYPE_MIFARE_UL and TARGET_TYPE_ISO14443_3A for T2T ?

Thanks for your help,
Contrib

jgeslin, 2017-01-11:
Hi,

Q1: There is no way to configure the stack in order to exclude discovery of all cards which are not NFC Forum Type Tag. Only the technology can be set (type A and/or type B and/or type F ...).

Q2: Indeed NFC Forum Type 2 tags are specified as type A technology and as such are discovered when nfcManager_enableDiscovery() is called with NFA_TECHNOLOGY_MASK_A set.

Q3: All NXP's NFC Forum type 2 tags (MIFARE Ultralight cards but also NTAG tags) will be identified as technology type "TARGET_TYPE_MIFARE_UL" while NFC Forum Type 2 tags from other IC manufacturers (for instance my-D move tags from Infineon) will be recognized as "TARGET_TYPE_ISO14443_3A".

Regards,
Jeremy.

ContribOpenSource, 2017-01-11:
Thanks for these explanations.
Just to be sure about your reply to Q3:
Q4) check on **TARGET_TYPE_MIFARE_UL** or **TARGET_TYPE_ISO14443_3A** handles all Type 2 Tags, or other values have also to be checked ?

Q5) check on **TARGET_TYPE_ISO14443_3A** will not trigger "false" positive if it is a Type 1 Tag, or Type 4 Tag for example ?

Regards,
Contrib

jgeslin, 2017-01-12:
You're right, Type 1 tags as well as Type 4 (of tech A) tags will also be identified as technology=TARGET_TYPE_ISO14443_3A. However you can easily distinguish Type 2 tags (whatever the IC manufacturer is) by looking to the protocol field of nfc_tag_info_t structure which will be equal to NFA_PROTOCOL_T2T (0x2).

closed 2017-05-04 by jgeslin