NXPNFCLinux/linux_libnfc-nci #30: When the device is opened and get filedescriptor 6 the lib wont work

d99roric, 2017-01-17:
Of some weird reason NFCSTATUS_INVALID_DEVICE is used to indicate that the device handle is invalid.

That define happens to be defined to 6:
#define NFCSTATUS_INVALID_DEVICE                              (0x0006)

But what if open returns fd 6? I can tell you the answer, read and write will fail since its considered being an invalid device....

jgeslin, 2017-01-18:
Thank you for raising the issue. 
It was already detected and planned to be fix in next release.

kallehuttunen, 2017-03-03:
This one bit us hard recently... We spent quite a while trying to debug what on earth is happening to our message queue. Well, the queue happened to have FD 6, and libnfc-nci decided to close that FD.

Any ETA on the release where this will be fixed?

jgeslin, 2017-03-06:
The fix will be included in next release for which with have no planned date yet.
In the meantime, see related patch here: [patch_TML_InvalidDevice.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/820641/patch_TML_InvalidDevice.txt)

darander, 2017-06-08:
@jgeslin If it's fixed, why isn't the patch pushed? Making an official release is one thing, pushing fixes to the git repo is another...

jgeslin, 2017-06-13:
Now integrated in release R2.2.

closed 2017-06-13 by jgeslin