NXPNFCLinux/linux_libnfc-nci #34: Fail to send APDU in a test minimal application.

ChristopheBelpaire, 2017-02-01:
Hello,
I'm trying to make a minimal application that read a page of a iso 15693 tag.
The tag is successfully detected, but the function nfcTag_transceive always return 0.
I tried the same APDU in the exmple app and it seems to work.
Any idea of what I'm doing wrong ?
Thanks in advance.

```
#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>
#include <ctype.h>
#include <string.h>
#include "linux_nfc_api.h"

static nfcTagCallback_t g_TagCB;
static nfc_tag_info_t g_TagInfo;
static time_t t0;
static time_t t1;

void onTagArrival(nfc_tag_info_t *pTagInfo){
    printf("\n-----\nTag detected\n");

    int res;
    int i;
    unsigned char cmd_send[3] = {
        0x02,
        0x20,
        0x00,
    };

    unsigned char cmd_response[255];
    memset(cmd_response, 0x00, 255);

    printf("SEND APDU: ");
    for(i=0; i < sizeof(cmd_send); i++){
        printf("%02X ",cmd_send[i]);
    }
    printf("\n");

    res = nfcTag_transceive((*pTagInfo).handle, cmd_send, 3, cmd_response, 255, 500);

    printf("APDU RESPONSE: ");
    for(i=0; i<res; i++){
        printf("%02X ",cmd_response[i]);
    }
    printf("\n");

}

void onTagDeparture(void){
    printf("Tag removed\n");
}

int main(int argc, char ** argv) {
    g_TagCB.onTagArrival = onTagArrival;
    g_TagCB.onTagDeparture = onTagDeparture;
    int res = nfcManager_doInitialize();
    nfcManager_registerTagCallback(&g_TagCB);
    printf("Callbacks registred\n");
    nfcManager_enableDiscovery(DEFAULT_NFA_TECH_MASK, 0x00, 0, 0);
    printf("Discovery enabled\n");
    getchar();
    nfcManager_doDeinitialize();
}
```

jgeslin, 2017-02-06:
Hi,

The libnfc-nci is making use of Threads, and the callback functions (onTagArrival, onTagDeparture, ...) are called from the main libnfc-nci thread preventing to call libnfc-nci API from there then.
The recommendation is to call the libnfc-nci API from the application thread (or a newly created one).

See [on NXP community forum](https://community.nxp.com/thread/442871) the exact same issue reported and related response.

Regards,
Jeremy.

mRamiroGonzalez, 2017-02-07:
I have checked the code on the forum and here is an answer that uses a mutex to wait for a tag and then read the first page of its memory (I am using an NXP ICODE DNA).
Everything is in a while true loop to continue looking for tags

```
#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>
#include <ctype.h>
#include <string.h>
#include <pthread.h>
#include "linux_nfc_api.h"

pthread_cond_t condition = PTHREAD_COND_INITIALIZER;
pthread_mutex_t mutex = PTHREAD_MUTEX_INITIALIZER;

nfcTagCallback_t g_TagCB;
nfc_tag_info_t g_tagInfos;

void onTagArrival(nfc_tag_info_t *pTagInfo){
    printf("Tag detected\n");
    int i = 0;
    printf("UID (inverted): ");
    for(i = 0x00; i < (*pTagInfo).uid_length; i++){
        printf("%02X ", (unsigned char) (*pTagInfo).uid[i]);
    }
    printf("\n");
    g_tagInfos = *pTagInfo;
    pthread_cond_signal(&condition);
}

void onTagDeparture(void){
    printf("Tag removed\n-------------\n");
}

int main(int argc, char ** argv) {
    g_TagCB.onTagArrival = onTagArrival;
    g_TagCB.onTagDeparture = onTagDeparture;
    nfcManager_doInitialize();
    nfcManager_registerTagCallback(&g_TagCB);
    nfcManager_enableDiscovery(DEFAULT_NFA_TECH_MASK, 0x01, 0, 0);

    unsigned char page = 0x00;
    int i;
    int res = 0x00;
    unsigned char cmd_send[3] = {0x02, 0x20, page};
    unsigned char cmd_response[255];
    printf("\n-------------\nWaiting for tag...\n");
    do{
        pthread_cond_wait(&condition, &mutex);
        memset(cmd_response, 0x00, 255);

        printf("SEND APDU: ");
        for(i=0; i < sizeof(cmd_send); i++){
            printf("%02X ",cmd_send[i]);
        }
        printf("\n");
        res = nfcTag_transceive(g_tagInfos.handle, cmd_send, 3, cmd_response, 255, 500);
        printf("RECV APDU (%d): ",res);
        for(i=0; i<res; i++){
            printf("%02X ",cmd_response[i]);
        }
        printf("\n");
    }while(1);
    nfcManager_doDeinitialize();
}
```

Output:
-------------
Waiting for tag...
Tag detected
UID (inverted): 01 9B 4F B1 18 01 04 E0
SEND APDU: 02 20 00
RECV APDU (5): 00 E1 40 20 00
Tag removed
-------------

closed 2017-05-04 by jgeslin

JuanAngelSG, 2017-12-01:
Hi, 

I have been testing this program with EMVCo Polling and also with LPCD polling, and I have noticed that after the activation of the card (after RATS), there are ~100 ms where nothing happens before the first APDU is sent. 

[image]

(Q1) Does anyone knows what is the cause of this? Maybe it has something to do with changing threads?
(Q2) Can I reduce this time to make the transaction/execution faster? If yes, how?

Thank you!
BR,
Juan

jgeslin, 2017-12-04:
Hi,

This may be due to the card itself. Indeed, the ISO14443 protocol defines a guard time between the card activation and the first APDU exchange. This is defined to allow the card to be ready for application exchanges. This guard time is refered as SFGT and specified by the card in the response to RATS. 

Can you copy here the NCI frame of the card activation "61 05 ..." ? From it we can extract the SFGT parameter to validate the assumption.

Regards,
Jeremy.

JuanAngelSG, 2017-12-04:
Hi,

Many thanks for the answer! 

I haven´t thought of that, but yes it could be the reason. The ATS is: 0E 78 77 **84** 03 66 52 57 45 ...
The card has SFGI = 4, so the next command should arrive much earlier (SFGT=4,8 ms).

NxpNciR:     len =  37 > 61052201020400FF010904000485E554C801200000000E**0D78778403665257453250524F4D**

In the meantime, I have observed that the application sends something between RATS and my APDU (around 8 ms after RATS). I am trying to find what this command is. Can I find this out using somehow the NCI Log (see below)?

[image]
[image]

I am a complete newbie with the linux_libnxp-nci, I hope I am not missing something absolutely obvious.

Best Reagards,
Juan

-----------------------------------------------------------------------------------------------
NxpNciX:     len =   4 > 20000101
NxpNciR:     len =   6 > 400003001001
NxpNciX:     len =   3 > 200100
NxpNciR:     len =  26 > 40011700030E03000600010203818202C800FF02000405080202
NxpNciX:     len =   6 > 20030301A005
NxpNciR:     len =   9 > 4003060001A0050100
NxpNciX:     len =   4 > 20000101
NxpNciR:     len =   6 > 400003001001
NxpNciX:     len =   3 > 200100
NxpNciR:     len =  26 > 40011700030E03000600010203818202C800FF02000405080202
NxpNciX:     len =   3 > 2F0200
NxpNciR:     len =   8 > 4F02050000007A5D
NxpNciX:     len =  10 > 20030703A002A003A004
NxpNciR:     len =  17 > 40030E0003A0020101A0030108A0040101
NxpNciX:     len =   8 > 20020501A0030108
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =  26 > 20021702A0EA08FFFFFFFFFFFFFFFFA0EB08FFFFFFFFFFFFFFFF
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =   8 > 20020501A0440100
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =  20 > 20021104A05E0101A0400101A0410104A0440101
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =   8 > 20020501A0620101
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =   4 > 2F000101
NxpNciR:     len =   4 > 4F000100
NxpNciX:     len =  50 > 20022F0E2801002101003001083101033304010203045401065001025B010060010E80010181010182010E18010100023200
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =  10 > 21010700010103000105
NxpNciR:     len =   4 > 41010100
NxpNciX:     len =  13 > 21000A03040302050303800180
NxpNciR:     len =   4 > 41000100
NxpNciX:     len =   4 > 22000101
NxpNciR:     len =   5 > 4200020001
NxpNciR:     len =   8 > 6200050101010100
NxpNciX:     len =   9 > 200406030101020101
NxpNciR:     len =   7 > 40040400FF0103
NxpNciX:     len =   5 > 0300028103
NxpNciR:     len =   6 > 600603010301
NxpNciR:     len =   5 > 0300028180
NxpNciX:     len =   7 > 030004810103C0
NxpNciR:     len =   6 > 600603010301
NxpNciR:     len =   5 > 0300028180
NxpNciX:     len =  14 > 03000B810101AD740E00A8740E00
NxpNciR:     len =   6 > 600603010301
NxpNciR:     len =   5 > 0300028180
NxpNciX:     len =   5 > 2201020101
NxpNciR:     len =   4 > 42010100
NxpNciR:     len =   8 > 6200050202018000
NxpNciX:     len =   6 > 030003810204
NxpNciR:     len =   6 > 600603010301
NxpNciR:     len =   6 > 030003818000
NxpNciX:     len =  19 > 20021005320120300104310100380101500100
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =   8 > 200205010002C800
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =   6 > 210303010001
NxpNciR:     len =   4 > 41030100
NxpNciR:     len =  37 > 61052201020400FF010904000485E554C801200000000E**0D78778403665257453250524F4D**
NxpNciX:     len =   8 > 000005C0B0000038 **This is my APDU**
NxpNciR:     len =   6 > 600603010001
NxpNciR:     len =  61 > 00003A0000C00889504E470D0A1A0A0000000D49484452000000FA000000B80806000000CA7CF15A000000017352474200AECE1CE900000006624B9000

jgeslin, 2017-12-05:
Hi,

Please instrument logs with timestamp to check the stack behavior, replacing linux_libnfc-nci/src/halimpl/pn54x/log/phNxpLog.c file by this one [phNxpLog.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1531526/phNxpLog.zip).
Then enable all logs withing libnfc-nxp.conf file, setting NXPLOG_xxxx_LOGLEVEL flags to 0x03.
The logs are output to stderr, so if you wish to route them to a file instead of stdout, run you app this way:
`./YourApp your_parameters 2>log.txt`

Regards,
Jeremy.

JuanAngelSG, 2017-12-06:
Hi jgeslin,

Thanks for the new message. This is my output file. I have been checking it, but unfortunately, I do not understand all of it.

[log.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1534025/log.txt)

The file records around 10 seconds and 2x card tap (card is type A, but not MIFARE). I am sending two APDUs one after the other, one requires the card to response with 56 bytes, the second APDU is answered with 3 bytes.

Best Regards,
Juan

jgeslin, 2017-12-06:
We are missing timestamp in the log. Did you replaced linux_libnfc-nci/src/halimpl/pn54x/log/phNxpLog.c with the one from phNxpLog.zip and rebuild the library ?
Without timestamp information in the log it will be difficult to see where are lost the 100ms.

JuanAngelSG, 2017-12-06:
Oh, sorry. I replaced the file on the first attempt, but my makefile for some reason omitted the file.

It is quite cool the timestamp logging. I just generated this new log, and managed to find what was the cause. I include the log for completion: 
[log1.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1535013/log1.txt)

So, thanks to the new log you provided, I saw the line "Posting read message....", and I though that it may be related to the two printf´s that I had in my code that were delaying the communication. I should have known this, even without the log, from other projects I have worked on in the past.

In any case, now the time between the end of ATS and the next APDU is only 9.3 ms.

Thankyou so much for the support!
Juan

jgeslin, 2017-12-06:
Can we close the issue then ?

JuanAngelSG, 2017-12-06:
Sure! 
I thought it was already closed.
Thanks again for your support and best regards.