NXPNFCLinux/linux_libnfc-nci #37: Time between 2 tag detection

ChristopheBelpaire, 2017-03-20:
Hello,
I'm using the PN7150 to read the serial number of iso 15693 ICODE SLIX tags with 
the nfcTagCallback_t functions.
My problem is that the time between 2 tags reading (so, the time between onTagArrival and onTagDeparture) is in average 1200ms.
Is there a way do decrease this time? 

Thanks in advance!

jgeslin, 2017-03-20:
Hi,
This delay is due to the timeout on the InventoryRequest command used to check the presence of the tag (to detect when it is being removed). This timeout can be reduced (by default 1s) setting  RW_I93_MAX_TIMEOUT parameter in rw_api.h file. However, this timeout applies to any communication with ISO15693, so decreasing it can lead to unexpected communication abort if the card requires more time to answer. A possible workaround could be to update the rw_i93.c implementation to define a new timeout value specific to the "PresenceCheck" command.

Regards,
Jeremy.

ChristopheBelpaire, 2017-03-27:
Hi,
Thanks for the answer, I'll try your solution.
I was also wondering, does your lib and the reader support anti-collision ?
If it is the case, would it be possible to handle multiple tags at once ?
I thought that the handle parameter in the nfc_tag_info_t was used for anti-collision.

Thanks in advance!

jgeslin, 2017-03-27:
Hi, 
The library handles anti-collision but for simplicity it only reports (through the callback function) the discovery of the first one (according the anti-collision order).

mRamiroGonzalez, 2017-03-27:
Is there a way to bypass that and get the id of multiple tags at once ?

jgeslin, 2017-03-28:
Not without rework of the libnfc-nci code (service layer).

closed 2017-05-04 by jgeslin