NXPNFCLinux/linux_libnfc-nci #43: Communicate through Serial Port

hazahashim, 2017-05-18:
Hi There , I am trying to connect the the NXP OM5577/PN 7120 S with raspberry pi through serial port . I have connected the TTL USB to the J8 of the Raspberry PI TTL USB RX - GPIO 14 (TXD) TTL USB TX - GPIO 15 ( RXD) GND - GND The OS install in the SD card for the Raspberry PI is OM5577-PN7120S_Rpi_Linux_demo However , i do not see any response from the serial port and raspberry PI .. I have chose the right COM Port and Baud rate (115200) I tried to run the serial port communication by using the NOOBS Raspberry PI OS , and run it with Teraterm , the serial port communication is working fine. Is there any special config for the NXP OM5577/PN7120 S if we want to communicate via serial port ? Note that I already add enable_uart=1 at the config file,

Why the serial communication fail when using OM5577 OS ?

jgeslin, 2017-05-18:
Hi,

PN7120 only offers I2C interface (no serial UART). As such the OM5577 demo kit cannot connected over serial port of the board. The setting of the "enable_uart=1" parameter in configuration file is useless for PN7120 (the presence of this parameter in configuration file comes form the porting of the NFC stack from the Android AOSP).

Regards,
Jeremy.

hazahashim, 2017-05-18:
Hi jgselin,

Is that mean eventhough I connected the I2C of the OM5577 to the raspberry PI , and connect the RX and TX of the OM5577 to the serial cable , it wont work?

Is there any way we could send the command of the I2C with Windows OS ? We are trying to run the OM5577 without Linux .

jgeslin, 2017-05-18:
Hi,

Indeed, the OM5577 can only be driven via I2C.
We are currently working on a new PN7120 demoboard driven via USB (including a USB to I2C converter).
This will allow driving PN7120 from windows laptop.
I will let you know here when it will be available.

Regards,
Jeremy.

hazahashim, 2017-05-18:
Hi , may I know the cause of NfcService Init Failed error message
Thanks ! 
Really appreciate the feedback

hazahashim, 2017-05-19:
Hi Jeremy ,
By saying that PN7120 only offers I2C interface , is there any limitation of the hardware / software .? Is the signal TX and RX from your have been disabled or occupied (connection or source code)?
I am just trying to simulate what I ran in linux of this through RX and TX signal (Pin 10 and 8)

Refer picture below

[image]

jgeslin, 2017-05-23:
I'm a bit confused by your comments. 
From the above picture it seems you was able to run the demo application running on raspberry pi with OM5577. Interface between RPi and PN7120 is I2C and can only be I2C (PN7120 HW only supports I2C interface to host).
If you are willing to use UART interface between RPi and PN7120 in place of the I2C interface, this is not definitely possible: PN7120 only supports I2C, then the libnfc-nci provided library is obviously only implementing I2C support (despite mention of "enable_uart" flags in configuration files).

hazahashim, 2017-05-25:
Yes, we are able to run the demo app on the Raspberry Pi with OM5577. I do know the interface between RPi and PN7120 is I2C and can only be I2C.

I'm not trying to use UART interface between Rpi and PN7120 in place of the I2C. I want to use the console function on the RPi while running the DemoApp and  I2C interface between RPi and PN7120. I need help to get the serial console working so I can connect to a PC via serial port.

jgeslin, 2017-05-29:
Thanks for the clarification.
However I still don't understand the relation between PN7120 and the serial communication of the RPi with a PC.
Can you please elaborate about the issue you are facing ?

hazahashim, 2017-05-30:
Hi Jeremy ,
understand tht RPI able to communicate with serial console too. From the PN7120 board , we take the RX TX pin and connect it to TTL USB (PC) . The Linux console was able to run on the windows PC via Putty software.
However , when we run " ./nfcDemoApp poll 
It will give out message "NfcService Init Failed error message" on the windows PC , but no error found when we ran it on Linux . 

[image]

jgeslin, 2017-06-13:
Can you please enable debug traces (set all NXPLOG_xxx_LOGLEVEL flags to 0x03 inside /etc/libnfc-nxp.conf file) and provide logs ?

jgeslin, 2018-11-07:
Closing since no update for one year.

closed 2018-11-07 by jgeslin