NXPNFCLinux/linux_libnfc-nci #44: IRQ use in PN7150

prattom, 2017-05-19:
Based on reading document http://www.nxp.com/documents/application_note/AN11756.pdf and by studying driver code I assume IRQ gpio in PN7150 is used to detect tag reading. I wanted to use this IRQ to wake-up my board when tag comes near PN7150 controller and thus I register this IRQ gpio for board wake-up and used some other gpio for IRQ in dts(not connected to PN7150 and thus empty) just for some testing. But now whenever I call nfcDemoApp poll for reading tag the NFC doesn't work. It tries many times power up and power down but fails.
Thus my question is is IRQ being used for some other use other than tag detection? Can I avoid using IRQ in PN7150 just to perform tag read operation(I have some other method to detect tag and thus not relying on PN7150IRQ)?

My log after disabling IRQ
nfcDemoApp poll
NxpTml:     pn54x_dev_open : 10,57
Opening port=/dev/pn544

pn54x_dev_ioctl, cmd=1074063617, arg=1
NxpTml:     phTmlNfc_i2c_reset(),pn544_enable power on
 VEN level 1
NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_reset(), VEN level 0pn544_disable power off

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_reset(), VEN level 1pn544_enable power on

NxpTml:     PN54X - Tml Reader Thread Started................

NxpTml:     PN54X - Read requested.....

NxpTml:     PN54X - Invoking I2C Read.....

NxpTml:     PN54X - Tml Writer Thread Started................

NxpTml:     PN54X - Tml Writer Thread Running................

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_reset(), VEN level 1pn544_enable power on

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_reset(), VEN level 0pn544_disable power off

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_reset(), VEN level 1pn544_enable power on

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_reset(), VEN level 1pn544_enable power on

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_reset(), VEN level 0pn544_disable power off

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_reset(), VEN level 1pn544_enable power on

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_reset(), VEN level 1pn544_enable power on

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_reset(), VEN level 0pn544_disable power off

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_reset(), VEN level 1pn544_enable power on

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpTml:     PN54X - Write request NOT enabled

jgeslin, 2017-05-23:
The IRQ pin is used by the PN7150 to signal to the host that data are pending on the I2C interface and that the host must initiate I2C read request to get the data from PN7150. It indeed indicates that a tag has been discovered, since it triggers an RF_DISCOVER_NTF NCI frame, but not only.
However, in the nxp-pn5xx kernel module, you can avoid using the IRQ in the pn54x_dev_read by instead doing regular I2C read (polling) request (i2c_master_recv function call). The PN7150 will respond with wrong values (first byte will be either its I2C address or 0xFF) until it has something to send (valid NCI frame).

jgeslin, 2018-11-07:
Closing since no update for one year.

closed 2018-11-07 by jgeslin

jahnf, 2021-03-11:
@prattom @jgeslin Was there any solution to this? I also would like to know how to configure my device tree to be able to wake up the system from suspend.

prattom, 2021-03-12:
@jahnf Yes it's possible to wakeup system from sleep. I modified driver https://github.com/NXPNFCLinux/nxp-pn5xx/blob/master/pn5xx_i2c.c file to generate interrupt on gpio when tag is detected to wake up system. I think I added few lines to this file to generate interrupt. So in device tree you just need to add gpio interrupt wake up after changing driver.

jahnf, 2021-03-12:
@prattom Thank you for the reply - would it be possible for you to share these changes? This would be really appreciated.