NXPNFCLinux/linux_libnfc-nci #47: How to change NCI parameter?

kroegerama, 2017-08-03:
I need to change the parameter called 'TOTAL_DURATION'. Is there any way to set it to a different value?

jgeslin, 2017-08-04:
Hi,

This can easily be done by through libnfc-nxp.conf file.
Parameter NXP_CORE_CONF must be updated to add TOTAL_DURATION setting (see update in bold):

NXP_CORE_CONF={ 20, 02, **2F**, **0E**,
    28, 01, 00,
    21, 01, 00,
    30, 01, 08,
    31, 01, 03,
    33, 04, 01, 02, 03, 04,
    54, 01, 06,
    50, 01, 02,
    5B, 01, 00,
    60, 01, 0E,
    80, 01, 01,
    81, 01, 01,
    82, 01, 0E,
    18, 01, 01,
    **00, 02, 00, 01**
}

Last line relates to set Tag #0 (TOTAL_DURATION tag value) to 0x0100 (in ms, little endian) so 256 ms. The third and fourth bytes must be updated to reflect the CORE_SET_CONFIG command payload size and number of parameters.

Regards,
Jeremy.

kroegerama, 2017-08-04:
Thanks a lot! This should be way better documented, though ;)

closed 2017-08-04 by kroegerama