NXPNFCLinux/linux_libnfc-nci #48: How to switch to continuous detection and disable emulation mode

kroegerama, 2017-08-04:
How can I disable the emulation mode and also enable continuous detection of NFC tags? I need a way to detect and read tags as fast as possible. In my scenario, the tags are in the range of the antenna for a very short time.

I already tried to change the TOTAL_DURATION (see #47 ), but it did not change anything. The chip just sends bursts every 500ms.

robert1356, 2017-08-04:
You need to set the stack to poll ONLY for the card type you're using.  For example, if you're polling for TypeA, disable TypeB and TypeF.  Then you also need to set the total duration to a small value.  When polling is not active the stack will be in emulation mode.

kroegerama, 2017-08-04:
Okay, I set `POLLING_TECH_MASK=0x01` and it only scans for Type A now.

I also set
**NXP_CORE_CONF={ 20, 02, 2F, 0E,
    28, 01, 00,
    21, 01, 00,
    30, 01, 08,
    31, 01, 03,
    33, 04, 01, 02, 03, 04,
    54, 01, 06,
    50, 01, 02,
    5B, 01, 00,
    60, 01, 0E,
    80, 01, 01,
    81, 01, 01,
    82, 01, 0E,
    18, 01, 01,
    00, 02, 32, 00
    }**
but I still see the bursts every 500ms.

I also tried a few more things:
- Changing `A0, 40, 01, 01` to `A0, 40, 01, 00` in `NXP_CORE_CONF_EXTN` does not help
- Setting `NXP_CORE_STANDBY={2F, 00, 01, 00}` does not help

jgeslin, 2017-08-07:
Hi,

This is strange indeed since "00, 02, 32, 00" should set the discovery loop period to approximately 50ms.
Are you sure the new setting applies ? To verify, enable the logs by setting parameters NXPLOG_NCIX_LOGLEVEL and NXPLOG_NCIR_LOGLEVEL to 0x03 in libnfc-nxp.conf and provide trace of the initialization here.

Regards,
Jeremy.

kroegerama, 2017-08-07:
Thanks a lot for your response! Here is my log output:

```
NxpNciX:     len =   4 > 20000101
NxpNciR:     len =   6 > 400003001001
NxpNciX:     len =   3 > 200100
NxpNciR:     len =  26 > 40011700030E03000600010203818202C800FF02000405080202
NxpNciX:     len =   6 > 20030301A005
NxpNciR:     len =   9 > 4003060001A0050100
NxpNciX:     len =   4 > 20000101
NxpNciR:     len =   6 > 400003001001
NxpNciX:     len =   3 > 200100
NxpNciR:     len =  26 > 40011700030E03000600010203818202C800FF02000405080202
NxpNciX:     len =   3 > 2F0200
NxpNciR:     len =   8 > 4F02050000007A5D
NxpNciX:     len =  10 > 20030703A002A003A004
NxpNciR:     len =  17 > 40030E0003A0020101A0030108A0040101
NxpNciX:     len =   8 > 20020501A0030108
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =  26 > 20021702A0EA08FFFFFFFFFFFFFFFFA0EB08FFFFFFFFFFFFFFFF
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =   8 > 20020501A0440100
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =  20 > 20021104A05E0101A0400101A0410104A0430100
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =   8 > 20020501A0620101
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =   4 > 2F000101
NxpNciR:     len =   4 > 4F000100
NxpNciX:     len =  50 > 20022F0E2801002101003001083101033304010203045401065001025B010060010E80010181010182010E18010100023200
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =  10 > 21010700010103000105
NxpNciR:     len =   4 > 41010100
NxpNciX:     len =  10 > 21000702040302050303
NxpNciR:     len =   4 > 41000100
NxpNciX:     len =   4 > 22000101
NxpNciR:     len =   5 > 4200020001
NxpNciR:     len =   8 > 6200050101010100
NxpNciX:     len =   9 > 200406030101020101
NxpNciR:     len =   7 > 40040400FF0103
NxpNciX:     len =   5 > 0300028103
NxpNciR:     len =   6 > 600603010301
NxpNciR:     len =   5 > 0300028180
NxpNciX:     len =   7 > 030004810103C0
NxpNciR:     len =   6 > 600603010301
NxpNciR:     len =   5 > 0300028180
NxpNciX:     len =  14 > 03000B810101ADD8FFFFA8D8FFFF
NxpNciR:     len =   6 > 600603010301
NxpNciR:     len =   5 > 0300028180
NxpNciX:     len =   5 > 2201020101
NxpNciR:     len =   4 > 42010100
NxpNciR:     len =   8 > 6200050202018000
NxpNciX:     len =   6 > 030003810204
NxpNciR:     len =   6 > 600603010301
NxpNciR:     len =   6 > 030003818000
NxpNciX:     len =  26 > 20021701291446666D010111020207FF03020003040164070103
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =  19 > 20021005320120300104310100380101500100
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =   8 > 200205010002F401
NxpNciR:     len =   5 > 4002020000
NxpNciX:     len =  10 > 21030703000180018101
NxpNciR:     len =   4 > 41030100
```

jgeslin, 2017-08-08:
Hi,

I just realized that the TOTAL_DURATION setting, if inserted in NXP_CORE_CONF parameter, is overwritten with 500ms default value just before the discovery loop is started:
> NxpNciX:     len =   8 > 200205010002F401

In order to change the default value applied, parameter NFA_DM_DISC_DURATION_POLL must be defined in libnfc-brcm.conf file and set to the required value (in ms). For instance: `NFA_DM_DISC_DURATION_POLL=50`.

Sorry for the confusion.

Regarding the minimum value, 0 is not supported by PN7120. It would lead to incomplete polling sequence. You can set TOTAL_DURATION to 1 but, due to internal execution time, the minimum LISTEN phase period will always be in the range of 16 ms.

Regards,
Jeremy.

kroegerama, 2017-08-08:
Thanks a lot for your kind help! This worked perfectly! :)

closed 2017-08-08 by kroegerama

JuanAngelSG, 2017-12-01:
Hello,

I don´t quite follow what "continous detection" is, but I think if you change the polling to EMVCo polling (instead of LPCD) you should be even faster. I tried changing the libnfc-nxp.conf file to this:

NXP_CORE_CONF_EXTN={20, 02, 11, 04,
    A0, 5E, 01, 01,
    A0, 40, 01, 01,
    A0, 41, 01, 04,
    **A0, 44, 01, 01**
}

BR,
Juan

jgeslin, 2017-12-04:
Indeed, parameter "A0 44" allows setting the Discovery profile. In the EMVCo profile, according EMVCo specification, there is no listen phase in the discovery loop. However it prevents polling for other types than passive A and passive B types.