NXPNFCLinux/linux_libnfc-nci #49: Python integration

aeyuxi, 2017-08-24:
Hello, I was wondering if there is a way to use this library with Python. How would it be the best way to integrate it? Thanks beforehand.

jgeslin, 2017-08-25:
Hi,

Similar question was already asked in this issue: https://github.com/NXPNFCLinux/linux_libnfc-nci/issues/39#issuecomment-289456193

Perhaps the @prattom find a solution ?

Regards,
Jeremy.

prattom, 2017-08-25:
I am using python subprocess.

jgeslin, 2017-08-25:
Any example you could share ?

aeyuxi, 2017-08-25:
@prattom is there any code that you could share? Any insight or clue in how to accomplish this will be helpful.

aeyuxi, 2017-08-25:
I think I got it, let me try to explain what I think you are doing. You are calling the program nfcDemoapp. Then, you can use the different methods such as polling, writing, emulate and push by using the python subprocess. 

It would be nice to be able to call this functions from python by wrapping the library/program. It worths a try. 

I have also seen that there is a library called [nxppy](https://github.com/svvitale/nxppy). Would it be suitable for this device too (PN7120)?

prattom, 2017-08-26:
@aeyuxi Yes You are correct regarding use of subprocess
`subprocess.call(["/usr/sbin/nfcDemoApp", "poll"])`

Also I don't think nxppy will work with this NFC since nxppy is PN512 type NFC. So currently there is no solution other than using python subprocess if you want to use this library with python.

aeyuxi, 2017-08-28:
@prattom, thanks for your help. I'm having some issues with the subprocess. Do you think you could share your code? could I give you an e-mail? Probably sharing it here?

aeyuxi, 2017-08-28:
@prattom 

I have tried to follow your suggestion.
`import subprocess
out=subprocess.Popen('./nfcDemoApp poll', shell=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
(stdout, stderr) =out.communicate()`

However, the code it is interrupted immediately, it does not wait for the tag to be read. This is the output in the python shell.

#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Leaving ...

prattom, 2017-08-29:
@aeyuxi 
I am using following simple code
subprocess.call(["/usr/sbin/nfcDemoApp", "poll"])

aeyuxi, 2017-08-29:
@prattom Thanks. I'll give it a try. How are you printing the information into the console? Would you mind if I bother you by requesting to send me the code to anfedres.86@gmail.com? That would help me a lot. Thanks.

prattom, 2017-08-31:
Hi @aeyuxi  I changed my nfcDemoApp code to print tag data in file and then I process it. Thus in python code I am simply using  `subprocess.call(["/usr/sbin/nfcDemoApp", "poll"])`. My nfcDeamoApp generates a tagData file with nfc tag contents and then I process it using python code.

prattom, 2017-08-31:
I have a python code if you want to process nfc tag content
`from subprocess import Popen, PIPE, STDOUT
import pty
import os
cmd = '/usr/sbin/nfcDemoApp poll'

master, slave = pty.openpty()

p = Popen(cmd, shell=True, stdin=PIPE, stdout=slave, stderr=slave, close_fds=True)
stdout = os.fdopen(master)
line = stdout.readline();
while (line != ''):
    print stdout.readline()`

This is some basic python code for running nfcDemoApp. I was using this code earlier but later on modified nfcDemoApp code for writing tag content in file.

aeyuxi, 2017-09-04:
I will give it a try. It would be nice that this library could be integrated with Python using some wrapper. However, it seems that the task is not that easy. Thanks to all of you for your answer.

doronhorwitz, 2019-02-16:
I realise this is closed, but I hope I can still contribute, to help out other Raspberry Pi Python developers who get here.
Not wanting to modify the nfcDemoApp, I took @prattom's idea and got it working as a nice reusable module. You can find it here: https://gist.github.com/doronhorwitz/fc5c4234a9db9ed87c53213d79e63b6c

To use it 

```
# also works for PN7120
from pn7150 import PN7150

pn7150 = PN7150()

# assign a callback for each time an NFC tag is read
pn7150.when_tag_read = lambda text: print(text)

# the following line launches a thread that launches nfcDemoApp
# as a subprocess and reads its output text for every NFC tag read
pn7150.start_reading()

# to stop reading - be sure to do this before shutting down your
python app or script - otherwise the subprocess with nfcDemoApp
running will stay alive
pn7150.stop_reading()