NXPNFCLinux/linux_libnfc-nci #51: Emulate tag using a cellphone

aeyuxi, 2017-09-07:
Greetings @jgeslin. I saw in previous post #1 and #16 that is possible to send data to the PN7120.

To give you some context. I have tried the nfcDemoapp using the poll option, I have a cell phone that supports NFC communication (Samsung Galaxy 7). I'm trying to send a message from the phone to the board. However, the only thing I see is "Device Found"/"Device Lost". I can´t get any message. Is there a way that I can send data from Android phone and read it using th PN7120 board? 

Thank you beforehand for your help. Also, thanks for the answer on previous posts.

jgeslin, 2017-09-07:
Hi,
From the Android phone there 2 possibility to send data to PN71xx:
- Android phone emulating a card
- Android phone communicating in P2P mode
For more details, please have a look to: https://developer.android.com/guide/topics/connectivity/nfc/index.html

However, since the Android phone by default always support P2P (when Host Card Emulation is enabled) and that also from PN71xx side P2P mode is also supported, P2P communication will always be favored.
I would recommend you to have a look to the Android Beam API (https://developer.android.com/guide/topics/connectivity/nfc/nfc.html#p2p). Using setNdefPushMessage function, your application can set the message to send to remote P2P when beam action will be fired.
As an example of use of this API: 
- open your web browser on the android phone and display whatever web page
- start the nfcDemoApp application in "poll" mode
- tap the phone to PN71xx antenna, the phone display should propose a beam
- press the phone screen to fire the beam, the web page URL information should be transmitted to PN71xx
- on the nfcDemoApp application side you should see the received message (URL)
See [here ](https://www.youtube.com/watch?v=XERC65Jzrr8) a video of the android beam feature (when using 2 android device).

Regards,
Jeremy.

aeyuxi, 2017-09-08:
@jgeslin. Thanks for your help. I have tried the Android Beam, but I consider that is not suitable for the application that I'm currently developing. I was hoping to be able to emulate the card that comes with the PN7120. However, it has been hard to find information. I was hoping to use the example in #50 along with the Android app that I'm currently developing.  

Is there a way to capture text (NDEF_FRIENDLY_TYPE_TEXT)  using the cell phone? How is it possible to emulate a tag?. Do you know where I can find an example of the code?  Any help and guidance will be very appreciated

jgeslin, 2017-09-08:
Hi,

To be used as implementation example, please find bellow attached the android HCE example application modified to expose NDEF text message to a remote NDEF reader.
When tapping the phone against PN7120 (run nfcDemoApp with "poll" option), the NDEF message will be read by PN7120 and display in the console.

[CardEmulation.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1288276/CardEmulation.zip)

Regards,
Jeremy.

aeyuxi, 2017-09-11:
Thanks @jgeslin. Unfortunately, I wasn't able to run the example. However, I think the source code of the link is this one ([android-card-emulation](https://github.com/googlesamples/android-CardEmulation)). Which it simulates a debit/credit card and sends the  

I have run the nfcDemoApp with the "poll" mode. However, I wasn't able to get any message nor the number that I was typed in the Android app. It only showed "Device Found"/"Device Lost". I was able to get something by only enabling the poll mode. However, it showed the URL, but not the numbers I was sending. I'm still asking myself if it's possible to emulate the tag and send just text. Hope you can guide me.

jgeslin, 2017-09-11:
The example I provided is the original android-card-emulation I modified myself to expose an NDEF message (Text message "Test" fixed in the code of the application).

> I'm still asking myself if it's possible to emulate the tag and send just text. Hope you can guide me.
I verified myself the behavior running the app on a Samsung GS5 (with android M) and PN7120 driven from the nfcDemoApp under Linux.

If you can't build the application from the zip file I can share the modification to the original application I've made.

aeyuxi, 2017-09-11:
@jgeslin. That would be very helpful. 

I have tried to open your the project you have provided. The first message that I have from Android Studio is this one. 

[image]

And when I try to compile it. I get the following error. 

[image]

I think I can run the app on my phone without compiling, since the apk is provided. I tried to run it and I saw the behavior already described. Hope you can help me with this issue.

jgeslin, 2017-09-11:
Here is the apk file : [Application-debug.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1293179/Application-debug.zip). The application "looks like" the original one since I didn't change the design. 

The first error you got when importing the project is expected, since you have not the same installation path of the SDK than I have. So the tool uses yours.
The second one relates to the test application which I didn't modified because useless. Please find here an updated package with the minimum required:

[CardEmulation.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1293203/CardEmulation.zip)

aeyuxi, 2017-09-11:
@jgeslin Thanks for your post. I got the same results previously mentioned. I keep getting "Device Found"/"Device Lost". I have to enable Android Beam to get something. However, I can't see anywhere the numbers that I have typed. Could you show me which is the result that you get when you approach your cell phone?. That would help me a lot.

jgeslin, 2017-09-12:
Sorry I missed to indicates that the nfcDemoApp must be updated to force looking to tag only (no more P2P devices, otherwise the phone is always seen as P2P).
For that you must update the demoApp/main.c to reflect the following discovery configuration (second parameter of the enableDiscovery API relates to enabling reader_only mode):
`                nfcManager_enableDiscovery(DEFAULT_NFA_TECH_MASK, 0x01, hce, 0);

Here is the log I got then when tapping the phone:

> Waiting for a Tag/Device...
> 
>         NFC Tag Found
> 
>                 Type :          'Type A'
>                 NFCID1 :        '01 02 03 04 '
>                 Record Found :
>                                 NDEF Content Max size :         '253 bytes'
>                                 NDEF Actual Content size :      '11 bytes'
>                                 ReadOnly :                                          'TRUE'
>                                 Type :                          'Text'
>                                 Text :                          'Test'
> 
> 
>                 11 bytes of NDEF data received :
>                 D1 01 07 54 02 65 6E 54 65 73 74
> 
>         NFC Tag Lost
> 
> Waiting for a Tag/Device...

Again, the account number is not used in this demonstration. The information pushed to the PN7120 from the phone is currently fixed in CardService.java as NDEF_MESSAGE variable.

lemimu, 2018-05-09:
Greetings @jgeslin, I saw this post and it was very helpful for me. I have one question, if I want to use the account number introduced by the user as NDEF_MESSAGE where  I have to change the code? Could you provide me some source code.
Thank you!

vtw3, 2018-07-03:
Hello jgeslin,

my partner and I are conducting research for a program we got into for the summer . Our project is centered around NFC. We have been reviewing documentation and have noticed that your source code follows the documentation provided by android clearly (thank you). Now we have an issue, we are not sure how to get the tag to be emulated through the source code you provided. do you have documentation you can provide us with that shows how you collected and emulated tags on your android device. It would be extremely helpful. Thank you in advance .

vtw3, 2018-07-03:
We are using Java on Android Studio to write an app to emulate a NFC tag. So far, we already developed a software on Windows machine which can read the tag and get the NFC tag UID. We want to use Host Card Emulation on Android in order to emulate a NFC tag that has consistent UID number to be read by the Windows program. However, we are having a problem because we are not sure how to emulate successfully.

jgeslin, 2018-07-03:
So what you are referring to is the Host-based card emulation feature of Android: https://developer.android.com/guide/topics/connectivity/nfc/hce.
Regarding your application, you can't base your implementation on the tag emulation UID because android HCE interface doesn't provide the capability to set it (this relates to NFC controller inside android phone and is differently managed according the manufacturer).
You should rather think of looking to the android HCE example application (android-card-emulation) which present an "ACCOUNT NUMBER" set in thanks to the application UI. Then from the windows program side you'll have to implement the appropriate scenario to select the App AID (F222222222) and you'll receive as response the "ACCOUNT NUMBER".
I hope that's help !

vtw3, 2018-07-03:
Thank you so much . Very helpful .

closed 2018-08-13 by jgeslin