NXPNFCLinux/linux_libnfc-nci #52: How can i read/write at block level (no NDEF) using this library?

FranciscoRaso, 2017-09-24:
Dear all

I'm using OM5577/PN7120S on Raspberry Pi, and different tag types (ISO15693 and ISO14443-2 / 3).
 
I'd like reading/writing sectors and blocks without NDEF format, but I can't find the way to do it by using this library.

Does anyone know of an example?

Thank you.

jgeslin, 2017-09-25:
Hi,

"Non NDEF" tag access is granted through 'nfcTag_transceive' API.
See attached simple example demonstrating read block access on non-NDEF MifareUL (Type 2 tag) and ISO15693 (Type 5 tag).

[raw_tag_access_example.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1329105/raw_tag_access_example.zip)

Regards,
Jeremy.

FranciscoRaso, 2017-09-25:
Hi Jeremy, 

Thank you very much for your answer.

This function (nfcTag_transceive), can it also be used for writing?

Regards,
Fran

jgeslin, 2017-09-26:
Indeed the nfcTag_transceive API allow raw exchange with the card, then you have to refer to ISO14443 or ISO15693 standards for the related protocol definition.

For instance writing MifareUL block can be done using the following command:
`unsigned char WriteCmd[] = {0xA2, /*block*/ 0x04, /* 4 bytes data to write */0xFF, 0xFF, 0xFF, 0xFF};`

and for ISO15693:
`unsigned char WriteCmd[] = {0x02, 0x21, /*block*/ 0x04, /* 4 bytes data to write */0xFF, 0xFF, 0xFF, 0xFF};`

FranciscoRaso, 2017-09-26:
Thanks for your answer Jeremy, 
I've tried writing in ISO15693, and reponse is 0103 (this code is supposed to indicate 'Option not supported')

Regards,
Fran

jgeslin, 2017-09-26:
'Option not supported' probably indicates the tag doesn't allow writing this block.

wirso, 2018-03-16:
@FranciscoRaso  
May I ask if you found any valuable informations bout R/W at block/raw level ?
If so, please contact me via y4m1.geek@gmail.com
Thank you in advance.
-Stanislav.

wirso, 2018-03-16:
By reading this I assume that this library doesn't support raw/block r/w for "Ntag213" type 2 tags or am I worng ?

jgeslin, 2018-03-19:
Raw read/write operation is supported by the libnfc-nci linux library.
For example with Ntag213 (relates to MIFARE Ultralight protocol) see https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/master/demoapp/main.c line 1435:

[image]

wirso, 2018-03-19:
Thank you for your reply @jgeslin .
The thing that I am troubled with is that with "nfcDemoApp poll" it tells me that it's a Type V Tag while the party that gave me the Tag assured that it was an Ntag213. I googled a little and found out that it's a Type 2 Tag and it's really confusing.
Output for "nfcDemoApp poll"
_____________________________________________________
Nfc Tag Found 
       Type :              'Type V'
       NFCID2 :         '75 00 70 01 00 26 02 E0 '
       NDEF Content  : NO, mode=1, tech=5

  Not a MIFARE card "
____________________________________________________
I will try what you mentioned earlier and if I will encounter any troubles I will post my issue.
Kind Regards,
-S.

jgeslin, 2018-03-19:
Please provide detailed log of the tag discovery for deeper analysis.
Furthermore, any chance you could scan the tag using NFC Tag info application on android (https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=fr) or iphone7/8 (https://itunes.apple.com/us/app/nfc-taginfo-by-nxp/id1246143596?mt=8)

wirso, 2018-03-19:
Sadly I can't find a way to send the output for "nfcDemoApp poll" to a file.
I tried to follow this "   https://askubuntu.com/a/731237   " but sadly any of the commands sent the output to my .txt file.
Would you please provide me a way (if known) to send the output to the .txt file ?
Kind Regards.
-S

jgeslin, 2018-03-19:
Simply start the application routing the stderr output to the file:
`nfcDemoApp poll 2>temp.txt`

Pay attention to enable the log in /etc/libnfc-nxp.conf file (NXPLOG_xxx_LOGLEVEL flags set to 0x03)

wirso, 2018-03-19:
[temp.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1824315/temp.txt)

I attached the temp.txt file with the mentioned earlier command.
Thank you for your kind support.
-S

jgeslin, 2018-03-19:
The trace "1:334:495 - NxpNciR:     len =  24 > 61051501010606FF010A000075007001002602E006808000
" confirm the NFC Controller detected card supporting ISO15693 protocol.
Furthermore, "1:340:524 - NxpNciX:     len =  13 > 00000A222B75007001002602E0
" and "1:352:375 - NxpNciR:     len =  17 > 00000E000B75007001002602E000002600
" show the card is answering to ISO15693 read command.

It confirms your card is definitely not Ntag213 (T2T) but ISO15693 tag (T5T).

wirso, 2018-03-19:
Thanks for your fast reply !
I will try to use the command for r/w on raw/block level with the ISO15693 type of tags.
I will post here if I encounter any problems.
Thanks again for you kind support.
Have a great day.
-S

robert1356, 2018-04-02:
MiFare tags cannot be read or written a byte at a time.  All memory accesses are block accesses of either 4 or 16 bytes.  If you really need byte level access, you'll need to read a block and access the bytes on the host side.

KevalSanghvi, 2018-04-03:
Thanks robert.

jgeslin, 2019-04-01:
Closed because considered answered and no comment since 1 year.

closed 2019-04-01 by jgeslin