NXPNFCLinux/linux_libnfc-nci #53: SNEP Client Register Callback Failed

titusece, 2017-10-13:
Hello,

I have interfaced the PN7150 into my i.MX6 board through i2c interface.

I have compiled like below.
./bootstrap
./configure --enable-pn7150
make

I am getting the following error while running on i.MX6 platform.

What could be the problem ?

> root@imx6:/RFID/linux_libnfc-nci-master# ./nfcDemoApp poll
> #########################################################################################
> ##                                       NFC demo                                      ##
> #########################################################################################
> ##                                 Poll mode activated                                 ##
> #########################################################################################
>                               ... press enter to quit ...
> 
> **SNEP Client Register Callback Failed**
> Leaving ...
> 
> 
> root@imx6:/RFID/linux_libnfc-nci-master#

Also changed the below file for i.MX6 i2c and GPIO pins.
./src/halimpl/pn54x/tml/i2c/phTmlNfc_alt.h

/* i.MX6 board */
 #define I2C_BUS         "/dev/i2c-2"
 #define I2C_ADDRESS     0x28
 #define PIN_INT         16   //GPIO_16 on i.MX6
 #define PIN_ENABLE      15   //GPIO_15 on i.MX6

2)
https://github.com/NXPNFCLinux/linux_NfcFactoryTestApp
Just did 'make' for this source and got the below output. (seems working, but don't know how to test)

> root@imx6:/linux_NfcFactoryTestApp# ./NfcFactoryTestApp
> 
> ----------------------------
> pn54x_dev_open : 10,62
> NFC Factory Test Application
> ---pn54x_dev_ioctl, cmd=1074063617, arg=0
> -------------------------
> pn544_disable power off
> pn54x_dev_ioctl, cmd=1074063617, arg=1
> pn544_enable power on
> RFID interrupt is raised!
> PN7150 NFC controller detected
> RFID interrupt is raised!
> Select the test to run:
>          1. Continuous RF ON
>          2. PRBS
>          3. Standby mode
> Your choice:

jgeslin, 2017-10-13:
Hi,

1. You simply missed configuration parameter to use "Alternative to pn5xx_i2c kernel driver" option:
> ./configure --enable-pn7150 **--enable-alt**

2. Already answered in this other issue entry: https://github.com/NXPNFCLinux/linux_NfcFactoryTestApp/issues/1
But NfcFactoryTestApp working it validates the phTmlNfc_alt.h settings with regards to your connections.

Regards,
Jeremy.

titusece, 2017-10-13:
Thanks Jeremy for your quick & kind support.

Still I am getting the same error.  Can you please help ?

```
./bootstrap
./configure --enable-pn7150 --enable-alt
make
./nfcDemoApp poll

#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
SNEP Client Register Callback Failed
Leaving ...

root@imx6:/RFID/linux_libnfc-nci-master# 
root@imx6:/RFID/linux_libnfc-nci-master# 
```

jgeslin, 2017-10-13:
I think you are missing the conf files (libnfcbrcm.conf and libnfc-nxp.conf). 
They should be present in /etc/ directory of the target.
They are basically installed at "make install" step but according to 'sysconfdir' path which must be set at configure step:
> ./configure --enable-pn7150 --enable-alt **--sysconfdir=/etc**

Or you can simply copy them by yourself from linux_libnfc-nci/conf/PN7150/.

titusece, 2017-10-13:
Thanks Jeremy for your help.
Now getting some different error.
Just I copied the configuration files into /etc/ directory.

```
./nfcDemoApp poll

#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
NfcService Init Failed
Leaving ...
```

jgeslin, 2017-10-13:
Activate debug trace by updating /etc/libnfc-nxp.conf file: change value of NXPLOG_xxx_LOGLEVEL parameters to 0x03 and copy the obtain log here.

titusece, 2017-10-13:
Hi Jeremy,
Thanks for your wonderful support!
Now I am able to run the nfc app in poll mode without any issues. (got the message 'waiting for a Tag/device')
Just did the below.

```
./bootstrap
./configure --enable-pn7150 --sysconfdir=/etc
make
make install
./nfcDemoApp poll
```
Thanks again!

Regards,
Titus S.

closed 2017-10-13 by titusece