NXPNFCLinux/linux_libnfc-nci #54: Use on embedded platform (ARM/nRF52) where Linux is not present

ryangrimm, 2017-10-18:
I'm working on a project using the PN7150 connected to a Nordic nRF52 microcontroller and am wondering how painful it would be to get this library to work without the Linux portions. Has anybody tried this?

jgeslin, 2017-10-18:
Hi,

Porting this library (derived from Android) to an MCU is to my mind not a good idea.

I would rather suggest to have a look to this: [AN11990](https://www.nxp.com/docs/en/application-note/AN11990.pdf). 
Related SW package is available from [PN7150 Software&Tools web page](https://www.nxp.com/products/identification-and-security/nfc/nfc-reader-ics/nfc-controller-solutions/high-performance-nfc-controller-supporting-all-nfc-forum-modes-with-integrated-firmware-and-nci-interface:PN7150B0HN?tab=Design_Tools_Tab), and is referenced as "NXP-NCI MCUXpresso example Project - SW4325".
I remind someone already ported this code to Nordik nRF52 MCU driving.

Regards,
Jeremy.

davidhelms, 2018-07-31:
@ryangrimm I'm looking at the same issue of running the PN7150 as an I2C peripheral off an nRF52. Did you complete this project and what was your preferred approach?

jgeslin, 2018-11-07:
Closing since no update for one year.

closed 2018-11-07 by jgeslin

ryangrimm, 2018-11-07:
I did manage to get it to work by making some changes but it was a very long time ago and I can't remember what my solution ended up being. Eventually I ended up scrapping it all together as I needed a solution that used interrupts to read data from i2c which proved to be too painful to implement. Instead I ended up reimplementing the portions that I needed for use in Mynewt which made my life so much easier.

il3ven, 2019-07-10:
> I remind someone already ported this code to Nordik nRF52 MCU driving.

Do you have a link ? I am searching for a library for nRF52.