NXPNFCLinux/linux_libnfc-nci #55: error comfiure with ubuntu 14.04

ljoli, 2017-10-23:
Hello,

I am using an UBUNTU 14.04 and I try to cross-compile  on x686 : linux_libnfc-nci.

But I have an error on configure :
`./configure --enable-pn7150 --enable-silent-rules --enable-i2c   --with-gnu-ld`
 
/..../
checking pkg-config is at least version 0.9.0... yes
configure: LLCP1.3 support is disabled (no secure P2P)
configure: Selected TML is I2C
configure: Selected chip type is PN7150
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands dd
config.status: executing libtool commands dd
rm: invalid option -- 'l'
Try 'rm ./-libtoolT' to remove the file '-libtoolT'.
Try 'rm --help' for more information.
mv: invalid option -- 'l'
Try 'mv --help' for more information.
rm: invalid option -- 'l'
Try 'rm ./-libtool' to remove the file '-libtool'.
Try 'rm --help' for more information.
chmod: invalid option -- 'l'
Try 'chmod --help' for more information.

is it a problem of bash version ?

jgeslin, 2017-10-23:
May be related to : --with_gnu-ld option
https://gcc.gnu.org/ml/gcc-help/2001-07/msg00051.html

ljoli, 2017-10-23:
humm no, even without this option ( --with-gnu-ld ), I have **the same error** : 

./configure --enable-pn7150 --enable-silent-rules --enable-i2c
/-----------------------------------------------/

 checking for pkg-config... /dk_2T/tmp-jethro-k200l/sysroots/x86_64-linux/usr/bin/pkg-config
| checking pkg-config is at least version 0.9.0...
| checking for pkg-config... /dk_2T/tmp-jethro-k200l/sysroots/x86_64-linux/usr/bin/pkg-config
| checking pkg-config is at least version 0.9.0... yes
| 
| yes
| configure: LLCP1.3 support is disabled (no secure P2P)
| 
| configure: LLCP1.3 support is disabled (no secure P2P)
| configure: Selected TML is I2C
| 
| configure: Selected TML is I2C
| configure: Selected chip type is PN7150
| 
| configure: Selected chip type is PN7150
| checking that generated files are newer than configure...
| checking that generated files are newer than configure... done
| 
| done
| configure: creating ./config.status
| 
| configure: creating ./config.status
| config.status: creating Makefile
| 
| config.status: creating Makefile
| config.status: creating config.h
| 
| config.status: creating config.h
| config.status: executing depfiles commands
| 
| config.status: executing depfiles commands
| config.status: executing libtool commands
| 
| config.status: executing libtool commands
| rm: invalid option -- 'l'
| Try 'rm --help' for more information.
| 
| rm: invalid option -- 'l'
| Try 'rm --help' for more information.
| mv: invalid option -- 'l'
| Try 'mv --help' for more information.
| 
| mv: invalid option -- 'l'
| Try 'mv --help' for more information.
| rm: invalid option -- 'l'
| Try 'rm --help' for more information.
| 
| rm: invalid option -- 'l'
| Try 'rm --help' for more information.
| chmod: invalid option -- 'l'
| Try 'chmod --help' for more information.
| 
| chmod: invalid option -- 'l'
| Try 'chmod --help' for more information.
| + rm -rf ./-libtoolT
| 
| + rm -rf ./-libtoolT
| + cp -v ./-libtool libtool
| 
| + cp -v ./-libtool libtool
| './-libtool' -> 'libtool'
| 
| './-libtool' -> 'libtool'
| + chmod +x libtool
| 
| + chmod +x libtool
| + chmod +x ./-libtool
| 
| + chmod +x ./-libtool
/.../

jgeslin, 2017-10-23:
Did you try as well without --enable-silent-rules ? I never tried this option.
What's your libtool version ?

ljoli, 2017-10-23:
I tried without --enable-silent-rules  but this is the same thing.

I use a libtool version according to YOCTO ( jethro version ).
./meta/recipes-devtools/libtool/libtool_2.4.6.bb

My aim is to cross-compile on x86 bord .

ljoli, 2017-10-24:
Which version of **libtool** is necessary fo build it ?

ljoli, 2017-10-24:
On Ubuntu 14.04 ( 32bits ), I have libtool 2.4.2, compilation is OK.
On Ubuntu 17.10 ( 32bits ), I have libtool 2.4.6, compilation is KO.

jgeslin, 2017-10-24:
On Ubuntu 16.04 LTS (64 bits), I have libtool 2.4.6 and compilation is OK.

ljoli, 2017-10-25:
Yes, for me on Ubuntu ( 64 bits ) compilation is OK.
But goal is to build **linux_libnfc-nci** in 32 bits architecture.

jgeslin, 2018-08-13:
No updates so considered closed.

closed 2018-08-13 by jgeslin