NXPNFCLinux/linux_libnfc-nci #56: Delay when APDU fail

ChristopheBelpaire, 2017-10-31:
Hello, 
I'm currently trying to get uid of VaultIC154 tag. 
In order to do it, I need to send an APDU to get the infos.
Everything seems fine, but sometimes when the apdu fails, the reader doesn't respond for a while (some seconds).
Is there a way to configure something to avoid this ? 
Even if I can't send the APDU, I would like to be able to reset the reader directly if it fail. I would like   to be able to read another tag directly.
Thanks in advance!

Here is my code :
```c
#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>
#include <ctype.h>
#include <string.h>
#include <pthread.h>
#include <ei.h>
#include "linux_nfc_api.h"

pthread_cond_t condition = PTHREAD_COND_INITIALIZER;
pthread_mutex_t mutex = PTHREAD_MUTEX_INITIALIZER;

nfcTagCallback_t g_TagCB;
nfc_tag_info_t g_tagInfos;

struct Response{
    unsigned char *apdu[255];
    int length;
}Response;

struct Tag{
    char type[255];
    int open;
    int openable;
    uint8_t uid[255];
    size_t uid_len;
}Tag;

void onTagArrival(nfc_tag_info_t *pTagInfo){
    g_tagInfos = *pTagInfo;
    pthread_cond_signal(&condition);
}

void onTagDeparture(void){
}

void print_unsigned_char(unsigned char* apdu, int length){
   int i = 0;
   for(i = 0; i < length; i++){
        printf("%02X ", apdu[i]);
   }
   printf("\n");
}

void print_tag_infos(struct Tag tag){
    printf("----------------\n");
    printf("  Type: %s      \n", tag.type);
    printf("  Openable: %d  \n", tag.openable);
    printf("  Open: %d      \n", tag.open);
    printf("  UUID: ");
    print_unsigned_char(tag.uid, tag.uid_len);
    printf("----------------\n");
}

int process(){
    struct Response response;
    struct Tag tag;

    pthread_cond_wait(&condition, &mutex);

    strcpy(tag.type, "NULL");
    tag.open = 0;
    tag.openable = 0;
    tag.uid_len = 0;
    memset(response.apdu, 0x00, 255);

    //VaultIC Tag
    unsigned char get_info[5] = {0x80, 0x01, 0x00, 0x00, 0x38};
    unsigned char* temp[255];

    response.length = nfcTag_transceive(g_tagInfos.handle, get_info, sizeof(get_info), response.apdu, 58 , 100);

    unsigned char* tag_infos = response.apdu;

    strcpy(tag.type, "VaultIC154");

    // Check opening support
    tag.openable = tag_infos[53];

    // Check opening state
    tag.open = tag_infos[54];

    // UUID
    int i;
    for (i = 32; i < 32+8; i++){
        tag.uid[i-32] = tag_infos[i];
    }
    tag.uid_len = 8;

    print_tag_infos(tag);

    return 1;
}

int main(int argc, char ** argv) {
    int res = -1;
    g_TagCB.onTagArrival = onTagArrival;
    g_TagCB.onTagDeparture = onTagDeparture;
    nfcManager_doInitialize();
    nfcManager_registerTagCallback(&g_TagCB);
    nfcManager_enableDiscovery(NFA_TECHNOLOGY_MASK_B, 0x01, 0, 0);
    while(1) {
        res = process();
    }
    nfcManager_doDeinitialize();
    return res;
}
```

jgeslin, 2017-11-07:
Hi,

Please provide logs of the issue (setting flags NXPLOG_xxx_LOGLEVEL to 0x03 in /etc/libnfc-nxp.conf file) for further analysis.

Regards,
Jeremy.

jgeslin, 2018-08-13:
No updates so considered closed.

closed 2018-08-13 by jgeslin