NXPNFCLinux/linux_libnfc-nci #58: How to disable ReaderMode and just keep P2P mode active.

Sourabh-Pandit, 2017-11-03:
Is is possible to disable Reader mode, and just keep the p2p mode active. I have this use case where in our device should always acts as P2P device.

It is observed that while interacting with some mobiles,  some times my device detects the mobile as Tag( Type 4B).. Which is not what my application accepts..
Is there a way to simply disable the Reader mode and keeping P2P only active ?

jgeslin, 2017-11-06:
Hi,

You can define the polling technology editing /etc/libnfc-brcm.conf file.
Set POLLING_TECH_MASK=0x40 for instance will make NFC Controller only supporting P2P active A (106kbps) for activation in Initiator mode, however it will still be visible as P2P target in all modes.
This way it shall no detect anymore any contactless card nor the phone in card mode.

Regards,
Jeremy.

Sourabh-Pandit, 2017-11-07:
there is a flag for P2P priority, what is that for ?

jgeslin, 2017-11-07:
Which flag are your refering to ?

Sourabh-Pandit, 2017-11-08:
"NFA_POLL_BAIL_OUT_MODE", which can be defined in conf file to 0x01, which will enable the a global variable, which in turn is being checked in  function nfa_dm_p2p_prio_logic.
My question was what this function will do if enable NFA_POLL_BAIL_OUT_MODE ? And why it is kept disabled.

jgeslin, 2017-11-08:
Does your question relates to linux_libnfc-nci stack or to nxpnfc_android_marshmallow one ?

Sourabh-Pandit, 2017-11-09:
Oh... My bad, it is indeed related to android :( 
I need to report the issue to nxpnfc_android_marshmallow one ?

jgeslin, 2017-11-09:
This issue was moved to NXPNFCLinux/nxpnfc_android_marshmallow#7

closed 2017-11-09 by jgeslin