NXPNFCLinux/linux_libnfc-nci #59: PN7150 stopped working

GokulTechno, 2017-11-10:
[nfc.log](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1461459/nfc.log)
I'm using a custom nfc board using pn7150 chip. Module stops responding. Please find the error log in the attachment.

jgeslin, 2017-11-10:
Hi,

It relates to the issue for which the fix https://github.com/NXPNFCLinux/linux_libnfc-nci/commit/9b774ef656c22c4a14018c8916691bd708c95de5#diff-fca6cf6d5c39d91a1ec38e30f4956fa8 was delivered.
Unfortunately, according to the log the sample seems to be corrupted. The only way to workaround the issue is by removing NXP_RF_CONF_BLK_1 definition from the /etc/libnfc-nxp.conf file.

Regards,
Jeremy.

GokulTechno, 2017-11-11:
Hi Jeremy,

I have updated the linux_libnfc-nci library with latest patch but couldn't able to fix the issue. Then removed NXP_RF_CONF_BLK_1 definition from the /etc/libnfc-nxp.conf file. Still couldn't able to fix the issue and receiving same error log. Finally I have replaced pn7150 chip with a new pn7150 chip and it worked fine in the same environment. What could have been issue with old pn7150 chip with above error log ?

jgeslin, 2017-11-13:
Hi,

As said, the IC gets corrupted, and there is no way to recover. With the latest patch corruption can no more occurs but it indeed cannot recover corrupted sample.
Basically, the corruption was due to concurrent access to internal EEPROM memory caused by a command present in previous libnfc-nxp.conf file. Under certain circumstance (supply voltage, clock stability, ...) it can lead to EEPROM corruption which cannot anymore be accessed.
Sorry for the inconvenience.

Regards,
Jeremy.

GokulTechno, 2017-11-13:
In my case it was due to supply voltage. Thanks for the info. 

Regards,
Gokulraj

closed 2018-11-07 by jgeslin