NXPNFCLinux/linux_libnfc-nci #60: segmentation fault nfcDemoApp when no nfc device

sabelka, 2017-11-16:
I'm trying to get the nfcDemoApp working with a mikroe-2540 USB dongle on a Debian Stretch amd64 system. So far "nfcDemoApp poll" seems to work when the device is attached. However, if I start the demo app without an nfc device it crashes with a SIGSEGV.

I've compiled the linux_libnfc-nci with the following options:

CXXFLAGS="-g -O0" CFLAGS="-g -O0" ./configure --enable-pn7150 --enable-lpcusbsio --sysconfdir=/etc

gdb reveals the following stack trace:

Thread 5 "nfcDemoApp" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff5041700 (LWP 22998)]
0x00007ffff774b844 in validHandle (dev=0x6)
    at src/halimpl/pn54x/tml/lpcusbsio/lpcusbsio/lpcusbsio.c:216
216             curDev = curDev->next;
(gdb) bt
#0  0x00007ffff774b844 in validHandle (dev=0x6)                                       
    at src/halimpl/pn54x/tml/lpcusbsio/lpcusbsio/lpcusbsio.c:216                      
#1  0x00007ffff774c938 in GPIO_SetVENValue (handle=0x6, newValue=0 '\000')            
    at src/halimpl/pn54x/tml/lpcusbsio/lpcusbsio/lpcusbsio.c:807                      
#2  0x00007ffff774b501 in phTmlNfc_i2c_reset (pDevHandle=0x6, level=0)                
    at src/halimpl/pn54x/tml/lpcusbsio/phTmlNfc_lpcusbsio.c:239                       
#3  0x00007ffff774a9b1 in phTmlNfc_CleanUp ()                                         
    at src/halimpl/pn54x/tml/lpcusbsio/phTmlNfc.c:599                                 
#4  0x00007ffff7749e0c in phTmlNfc_Init (pConfig=0x7ffff5040d00)                      
    at src/halimpl/pn54x/tml/lpcusbsio/phTmlNfc.c:171                                 
#5  0x00007ffff775e043 in phNxpNciHal_open (                                          
    p_cback=0x7ffff7744e6c <NfcAdaptation::HalDeviceContextCallback(unsigned char, unsigned char)>,                                                                         
    p_data_cback=0x7ffff7744eb4 <NfcAdaptation::HalDeviceContextDataCallback(unsigned short, unsigned char*)>) at src/halimpl/pn54x/hal/phNxpNciHal.c:474                   
#6  0x00007ffff7744e4c in NfcAdaptation::HalOpen (                                    
    p_hal_cback=0x7ffff771a900 <nfc_main_hal_cback>,                                  
    p_data_cback=0x7ffff771aa23 <nfc_main_hal_data_cback>)                            
    at src/libnfc-nci/adaptation/NfcAdaptation.cpp:391                                
#7  0x00007ffff771ab75 in NFC_Enable (                                                
    p_cback=0x7ffff76fd0db <nfa_dm_nfc_response_cback>)                               
    at src/libnfc-nci/nfc/nfc/nfc_main.c:770                                          
#8  0x00007ffff76fd7f1 in nfa_dm_enable (p_data=0x555555772b88)                       
    at src/libnfc-nci/nfa/dm/nfa_dm_act.c:506                                         
#9  0x00007ffff76ffb86 in nfa_dm_evt_hdlr (p_msg=0x555555772b88)                      
    at src/libnfc-nci/nfa/dm/nfa_dm_main.c:162                                        
#10 0x00007ffff76f6aab in nfa_sys_event (p_msg=0x555555772b88)                        
    at src/libnfc-nci/nfa/sys/nfa_sys_main.c:86                                       
#11 0x00007ffff7719cd1 in nfc_task (param=0) at src/libnfc-nci/nfc/nfc/nfc_task.c:486
#12 0x00007ffff770ced6 in gki_task_entry (params=140737347622296)
    at src/libnfc-nci/gki/ulinux/gki_ulinux.c:128
#13 0x00007ffff7488494 in start_thread (arg=0x7ffff5041700) at pthread_create.c:333
#14 0x00007ffff71caaff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97

The segmentation fault occurs because curDev is a NULL pointer

(gdb) p curDev
$1 = (LPCUSBSIO_I2C_Ctrl_t *) 0x0

jgeslin, 2017-11-17:
Fixed with commit https://github.com/NXPNFCLinux/linux_libnfc-nci/commit/ea5e408e131ace0ab7acd68a10269cf752552d04

sabelka, 2017-11-17:
Works perfectly - thanks!

closed 2017-11-17 by sabelka