NXPNFCLinux/linux_libnfc-nci #61: nfcDemoApp segfaults in libpthread

sabelka, 2017-12-08:
When I try to run the nfcDemoApp on a opensuse Tumbeweed PC it crashes with an segmentation fault.

I've compiled the linux_libnfc-nci with the following options:
CXXFLAGS="-g -O0" CFLAGS="-g -O0" ./configure --enable-pn7150 --enable-lpcusbsio --sysconfdir=/etc

Backtrace:
$ gdb /usr/local/sbin/nfcDemoApp
GNU gdb (GDB; openSUSE Tumbleweed) 8.0.1
(gdb) r poll
Starting program: /usr/local/sbin/nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
[New Thread 0x7ffff5ce2700 (LWP 16786)]
                              ... press enter to quit ...

[New Thread 0x7ffff54e1700 (LWP 16787)]
[New Thread 0x7ffff4ce0700 (LWP 16788)]
[New Thread 0x7fffeffff700 (LWP 16789)]

Thread 4 "nfcDemoApp" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff4ce0700 (LWP 16788)]
__lll_unlock_elision (lock=0x7ffff4cdfd48, private=0)
    at ../sysdeps/unix/sysv/linux/x86/elision-unlock.c:29
29      ../sysdeps/unix/sysv/linux/x86/elision-unlock.c: No such file or directory.
(gdb) bt
#0  __lll_unlock_elision (lock=0x7ffff4cdfd48, private=0)
    at ../sysdeps/unix/sysv/linux/x86/elision-unlock.c:29
#1  0x00007ffff7b4f28e in ThreadMutex::unlock (this=0x7ffff4cdfd40)
    at src/libnfc-nci/adaptation/NfcAdaptation.cpp:728
#2  0x00007ffff7b4f44c in AutoThreadMutex::~AutoThreadMutex (
    this=0x7ffff4cdfda0, __in_chrg=<optimized out>)
    at src/libnfc-nci/adaptation/NfcAdaptation.cpp:820
#3  0x00007ffff7b4ec53 in NfcAdaptation::Thread (arg=0)
    at src/libnfc-nci/adaptation/NfcAdaptation.cpp:287
#4  0x00007ffff7b16c16 in gki_task_entry (params=140737351857472)
    at src/libnfc-nci/gki/ulinux/gki_ulinux.c:128
#5  0x00007ffff725e558 in start_thread (arg=0x7ffff4ce0700)
    at pthread_create.c:465
#6  0x00007ffff6f956bf in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb)

jgeslin, 2017-12-12:
Hi,

In order to further debug the issue please:
- add "#define GKI_DEBUG TRUE" to src/libnfc-nci/include/gki_target.h file
- set ENABLE_API_TRACES to TRUE in src/halimpl/pn54x/log/phNxpLog.h
- rebuild
- reproduce the segmentation fault and share logs

Regards,
Jeremy.

sabelka, 2017-12-12:
Hi Jeremy,
I did as you said - this is the output now:

~~~~
(gdb) r poll
Starting program: /usr/local/sbin/nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
[New Thread 0x7ffff5cd0700 (LWP 13063)]
                              ... press enter to quit ...

1513095314:196:016 - NxpFunc:    nativeNfcManager_doInitialize: enter; NCI_VERSION=0x10
1513095314:196:724 - NxpFunc:    NfcAdaptation::Initialize: enter
1513095314:197:015 - NxpFunc:    NfcAdaptation::Initialize: Overriding NFA_EE_MAX_EE_SUPPORTED to use 1
1513095314:197:060 - NxpFunc:    Brcm: GKI_enable

1513095314:197:064 - NxpFunc:    Brcm: Leaving GKI_enable

1513095314:197:071 - NxpFunc:    Brcm: GKI_create_task func=0xf7b40486  id=4  name=NFCA_TASK  stack=0x0  stackSize=0

1513095314:197:077 - NxpFunc:    Brcm: GKI creating task 4, pCond/pMutex=0/0

[New Thread 0x7ffff54cf700 (LWP 13064)]
1513095314:197:339 - NxpFunc:    Brcm: Leaving GKI_create_task f7b40486 4 f54cf700 NFCA_TASK 0 0

1513095314:197:348 - NxpFunc:    Brcm: gki_task_entry task_id=4, thread_id=f54cf700/f54cf700, pCond/pMutex=0/0

1513095314:197:372 - NxpFunc:    Brcm: GKI creating task 0, pCond/pMutex=0/0

1513095314:197:367 - NxpFunc:    Brcm: GKI creating task 0, pCond/pMutex=0/0

1513095314:197:401 - NxpFunc:    NfcAdaptation::NFCA_TASK: enter
1513095314:197:405 - NxpFunc:    Brcm: GKI_run enter

1513095314:197:422 - NxpFunc:    Brcm: GKI_run, run_cond(f7dd31b8)=1 

[New Thread 0x7ffff4cce700 (LWP 13065)]
1513095314:197:496 - NxpFunc:    Brcm: Leaving GKI_create_task f7b40502 0 f4cce700 NFCA_THREAD 0 0

1513095314:197:494 - NxpFunc:    Brcm: Leaving GKI_create_task f7b40502 0 f4cce700 NFCA_THREAD 0 0

1513095314:197:518 - NxpFunc:    NfcAdaptation::Thread: enter
1513095314:197:526 - NxpFunc:    Brcm: GKI_create_task func=0xf7b150ea  id=3  name=NFC_TASK  stack=0x0  stackSize=0

1513095314:197:532 - NxpFunc:    Brcm: GKI creating task 3, pCond/pMutex=f4ccdd60/f4ccdd38

[New Thread 0x7fffeffff700 (LWP 13066)]
1513095314:197:813 - NxpFunc:    Brcm: gki_task_entry task_id=3, thread_id=effff700/effff700, pCond/pMutex=f4ccdd60/f4ccdd38

1513095314:197:838 - NxpFunc:    Brcm: Leaving GKI_create_task f7b150ea 3 effff700 NFC_TASK 0 0

1513095314:197:861 - NxpFunc:    Brcm: NFC_TASK started.

1513095314:197:887 - NxpFunc:    Brcm: GKI_get_taskid effff700 3 done

1513095314:197:891 - NxpFunc:    Brcm: GKI_wait 3 ffff 0

1513095314:197:896 - NxpFunc:    Brcm: GKI_wait task=3, pCond/pMutex = f4ccdd60/f4ccdd38

Thread 4 "nfcDemoApp" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff4cce700 (LWP 13065)]
__lll_unlock_elision (lock=0x7ffff4ccdd38, private=0)
    at ../sysdeps/unix/sysv/linux/x86/elision-unlock.c:29
29      ../sysdeps/unix/sysv/linux/x86/elision-unlock.c: No such file or directory.
(gdb) bt
#0  __lll_unlock_elision (lock=0x7ffff4ccdd38, private=0)
    at ../sysdeps/unix/sysv/linux/x86/elision-unlock.c:29
#1  0x00007ffff7b41084 in ThreadMutex::unlock (this=0x7ffff4ccdd30)
    at src/libnfc-nci/adaptation/NfcAdaptation.cpp:728
#2  0x00007ffff7b41242 in AutoThreadMutex::~AutoThreadMutex (
    this=0x7ffff4ccdd90, __in_chrg=<optimized out>)
    at src/libnfc-nci/adaptation/NfcAdaptation.cpp:820
#3  0x00007ffff7b405dd in NfcAdaptation::Thread (arg=0)
    at src/libnfc-nci/adaptation/NfcAdaptation.cpp:287
#4  0x00007ffff7b07e43 in gki_task_entry (params=140737351853376)
    at src/libnfc-nci/gki/ulinux/gki_ulinux.c:128
#5  0x00007ffff724e558 in start_thread (arg=0x7ffff4cce700)
    at pthread_create.c:465
#6  0x00007ffff6f856bf in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
~~~~

sabelka, 2017-12-12:
Hi Jeremy,
in NfcAdaptation.cpp around line 728 there is the following code:
~~~~
    {
        ThreadCondVar    CondVar;
        AutoThreadMutex  guard(CondVar);
        GKI_create_task ((TASKPTR)nfc_task, NFC_TASK, (INT8*)"NFC_TASK", 0, 0, (pthread_cond_t*)CondVar, (pthread_mutex_t*)CondVar);
        CondVar.wait();
    }
~~~~
In this block, the mutex contained in the CondVar is unlocked twice: The first time by CondVar.wait(), the second time by the AutoThreadMutex destructor. It is not clear to me why this should cause a segmentation fault, but nevertheless I tried to see what happens if i remove the unlock from the CondVar.wait() function:
~~~~
void ThreadCondVar::wait()
{
    pthread_cond_wait(&mCondVar, *this);
    // pthread_mutex_unlock(*this);
}
~~~~
With this modification I no longer get the segfault and the demo app seems to work. I'm not sure however, if my modification has any side effects.

-Rainer

jgeslin, 2017-12-12:
Hi,

Indeed I do not see the rationale for the unlock in the wait method since it has not previously locked (except by the constructor).
FYI, this piece of code is coming from the AOSP NFC libnfc-nci component code.

I'll give it a try in my side and will feedback.

Regards,
Jeremy.

jgeslin, 2017-12-13:
Hi,

Tested in my side and didn't observed any side effect.
Furthermore, reviewing the code I confirm the unlock inside the wait method is useless since already handled in AutoThreadMutex destructor.
Additionally, pthread documentation defines the following which may explain the issue you are facing with your distribution while it was not seen with other ones tested so far:

> Calling pthread_mutex_unlock() with a mutex that the calling thread does not hold will result in undefined behavior.

Based on this I would approve your fix of removing pthread_mutex_unlock from the wait method.

Regards,
Jeremy.

sabelka, 2017-12-15:
Hi Jeremy,
during more testing I found another segmentation fault, again a not-locked pthread mutex is being unlocked here. I guess the code here just tries to unlock the mutex (just in case it is locked) before it is being destroyed. I think it I never it locked at this point, so I removed the umlock, which fixes my segfault.

~~~~
diff --git a/src/halimpl/pn54x/utils/phNxpNciHal_utils.c b/src/halimpl/pn54x/utils/phNxpNciHal_utils.c
index 22c2e65..5af7b2e 100644
--- a/src/halimpl/pn54x/utils/phNxpNciHal_utils.c
+++ b/src/halimpl/pn54x/utils/phNxpNciHal_utils.c
@@ -369,7 +369,6 @@ void phNxpNciHal_cleanup_monitor(void)
     if (nxpncihal_monitor != NULL)
     {
         pthread_mutex_destroy(&nxpncihal_monitor->concurrency_mutex);
-        REENTRANCE_UNLOCK();
         pthread_mutex_destroy(&nxpncihal_monitor->reentrance_mutex);
         phNxpNciHal_releaseall_cb_data();
         listDestroy(&nxpncihal_monitor->sem_list);
~~~~

I think the reason I'm experiencing these problems is that I have glibc compiled with --enable-lock-elision and a processor which supports hardware lock elision (HLE). I don't see this segfaults on other systems which don't support HLE, nor in virtual machines, nor when I run the  program in valgrind.

sabelka, 2017-12-15:
Hi Jeremy,
I just created a pull request with a fix. In contrast to my original attempt to fix this by removing the pthread_mutex_unlock() from ThreadCondVar::wait() I decided to replace the AutoThreadMutex by a single lock(). The ThreadCondVar type is used in several other locations where the mutex is not explicitly unlocked after a wait(), so my previous attempt might have caused problems there.

jgeslin, 2018-08-13:
Latest release R2.4 now includes the fix.

closed 2018-11-07 by jgeslin