NXPNFCLinux/linux_libnfc-nci #63: _i2c_open() Failed: retval ffffffff

paolochiarlone, 2017-12-20:
Hello,
I have been trying to port the libnfc into the Digi CC6UL platform.
I am using the Raspberry Pi kit as it is the closest in terms of HW.

I created my own configuration (#4) in phTmlNfc_alt.h and checked that the GPIO
pins 11 and 76  are enabled in my HW and corresponding to the right pins (16, 18) as in
the RPi connector.

I also have I2C1 enabled in my device tree and corresponding to the same pins (2, 3) as in
the RPi connector.

Here is the configuration I use:
```
#elif (CONFIGURATION == 4)
/* OM557x on Digi CC6UL Starter */
 #define I2C_BUS         "/dev/i2c-1"
 #define I2C_ADDRESS     0x28
 #define PIN_INT         76
 #define PIN_ENABLE      11
```

The first issue I have is that the kit is not being sensed.
```
root@ccimx6ulstarter:~# i2cdetect 1
i2cdetect: WARNING! This program can confuse your I2C bus
Continue? [y/N] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
root@ccimx6ulstarter:~#
```
Which is weird, because by connecting the Sense HAT, with the same HW, I can see
several I2C devices listed correctly and I can also run the examples in the kit fine.

And when I launch the demoapp, with LOGLEVEL set to 0x03, this is what I see:
```
root@ccimx6ulstarter:~# ./nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

1513808858:881:668 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

1513808858:881:934 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

1513808858:882:018 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

1513808858:882:097 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

0:001:763 - NxpHal:     Entering phNxpNciHal_init_monitor
0:001:910 - NxpHal:     Returning with SUCCESS
0:002:818 - NxpTml:     phTmlNfc_i2c_open_and_configure

0:002:936 - NxpTml:     Opening port=/dev/pn544

0:003:674 - NxpTml:     _i2c_open() Failed: retval ffffffff
0:003:809 - NxpHal:     phTmlNfc_Init Failed
0:004:544 - NxpHal:     Failed to deallocate (list empty)
0:004:659 - NxpHal:     Node dump:
0:004:732 - NxpHal:     Failed to deallocate (list empty)
0:004:795 - NxpHal:     Node dump:
0:012:608 - NxpExtns:   find found NXPLOG_NCIHAL_LOGLEVEL=(0x3)

0:012:794 - NxpExtns:   find found NXPLOG_EXTNS_LOGLEVEL=(0x3)

0:012:873 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

0:012:945 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

0:013:014 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

0:013:082 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NfcService Init Failed
Leaving ...

root@ccimx6ulstarter:~# 
```
I am not an expert of this code and unfortunately I am trying to port it over quickly,
so I admit I have not debugged in details, but I was wondering if you can point me to
the possible cause for this?

Thanks in advance!
Paolo Chiarlone

jgeslin, 2017-12-21:
Hi,

Regarding i2c_detect report, I can see 2 reason why the NFC Controller is not seen here:
- VEN signal (PIN_ENABLE) must be high (3.3V) otherwise the NFC Controller is disabled and not answering. Insure the signal is high before running the i2c_detect command
- Because of standby mode (internal mode allowing reducing the overall power consumption) the NFC Controller may not acknowledge the first i2C request. It is requested in this case to resend the request a second time in the interval of less than 1 second. Launch 2 times in a raw the i2c_detect command to see the NFC Controller.

From the logs of the nfcDemoApp execution I can see that you are not using the Alternative configuration but instead the kernel driver approach. You may have missed the "enable-alt" option at the configure step. What is the exact command you run (`$ ./configure ...`) ?

Regards,
Jeremy.

paolochiarlone, 2017-12-21:
@jgeslin 
Jeremy, 1st of all, thank you so much for your quick reply!

And you did help indeed, now I created a script that writes 1 on the ENABLE GPIO and then run i2c_detect right after, and I see the NXP device at address ox28.
**Great, problem no.1 solved!!!**

In regards to the command I used to configure, I might be confused, so here what I used:
`./configure --enable-pn7150 --enable-i2c --host=arm-dey-linux-gnueabi --prefix=/usr`
Then I did:
`make clean; make`

Let me know your comments please. Thank you in advance!!!
Paolo

paolochiarlone, 2017-12-22:
@jgeslin Jeremy,
So I reconfigured my project as per your suggestion and now I can see it is reading my phTmlNfc_alt.h file. I recompiled and deployed.

So now the program start, have no "failed" messages, however, this is what happens:

```root@ccimx6ulstarter:~# ./nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

1513911315:256:002 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

1513911315:256:302 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

1513911315:256:405 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

1513911315:256:501 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

0:000:521 - NxpHal:     Entering phNxpNciHal_init_monitor
0:000:671 - NxpHal:     Returning with SUCCESS
0:000:821 - NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC

0:000:914 - NxpTml:     NFC - Assign IO pins

0:001:006 - NxpTml:     Pin /sys/class/gpio/gpio11

0:001:215 - NxpTml:     Create pin /sys/class/gpio/gpio11

0:102:146 - NxpTml:     Direction /sys/class/gpio/gpio11/direction

0:102:753 - NxpTml:     Pin 11 now an input

0:102:953 - NxpTml:     Edge /sys/class/gpio/gpio11/edge

0:103:281 - NxpTml:     Edge-string rising - 6

0:103:574 - NxpTml:     Pin 11 trigger on rising

0:103:718 - NxpTml:     Value /sys/class/gpio/gpio11/value

0:103:988 - NxpTml:     Pin /sys/class/gpio/gpio76

0:104:153 - NxpTml:     Create pin /sys/class/gpio/gpio76

0:205:002 - NxpTml:     Direction /sys/class/gpio/gpio76/direction

0:205:798 - NxpTml:     Pin 76 now an output

0:206:597 - NxpTml:     Pin 76 now off

0:206:793 - NxpTml:     NFCHW - open I2C bus - /dev/i2c-1

0:207:085 - NxpTml:     NFC - open I2C device - 0x28

0:409:213 - NxpTml:     PN54X - Tml Reader Thread Started................

0:409:649 - NxpTml:     PN54X - Read requested.....

0:409:818 - NxpTml:     PN54X - Invoking I2C Read.....

0:410:273 - NxpTml:     PN54X - Tml Writer Thread Started................

0:410:606 - NxpTml:     PN54X - Tml Writer Thread Running................

0:410:774 - NxpTml:     PN54X - Write requested.....

0:410:920 - NxpTml:     PN54X - Invoking I2C Write.....
[... 102 lines cut ...]
4:037:810 - NxpTml:     PN54X - Invoking I2C Write.....

4:039:076 - NxpNciX:     len =   4 > 20000101
4:039:271 - NxpTml:     PN54X - I2C Write successful.....

4:039:425 - NxpTml:     PN54X - Posting Fresh Write message.....

4:039:620 - NxpTml:     PN54X - Tml Writer Thread Running................

4:039:841 - NxpHal:     write successful status = 0x0
4:040:116 - NxpHal:     Response timer started
4:040:293 - NxpHal:     Waiting after ext cmd sent
5:040:962 - NxpHal:     hal_extns_write_rsp_timeout_cb - write timeout!!!
5:041:591 - NxpHal:     Response timer stopped
5:041:789 - NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
5:041:970 - NxpHal:     NCI_CORE_RESET: Failed
5:044:491 - NxpTml:     PN54X - Write request NOT enabled
```

After that, it is hanging here forever but not reading the card data.

Any clue of what is happening?

Thanks again!
Paolo

jgeslin, 2017-12-22:
From the log I can see that I2C write operation is acknowledged by the NFC Controller but no I2C read is fired afterwards to read the answer to the first NCI command. This is usually done when IRQ line is raised by the NFC Controller.
Can you check status of the IRQ line (using scope or voltage meter) during execution of the application ?

paolochiarlone, 2017-12-22:
@jgeslin Jeremy,
Thank you again for the reply.
OK I will check with the definition of the GPIO used for INT inside my DTS.

I think right now it is declared like this (but I am not positive it is correct):

```
&i2c2 {
        clock-frequency = <100000>;
        pinctrl-names = "default";
        pinctrl-0 = <&pinctrl_i2c2>;
        status = "disabled";
 
        pn544: pn544@28 {
                compatible = "nxp,pn544";
                reg = <0x28>;
                clock-frequency = <400000>;
                interrupt-parent = <&gpio6>;
                interrupt-gpios = <&gpio6 2 0>;
                enable-gpios = <&gpio5 22 0>;
        };
};
```

I will ran few tests and see what I can come up with. Stay tuned :)

paolochiarlone, 2018-01-05:
@jgeslin Jeremy,
Happy New Year!
I might need some further clarification...
I changed my dts and I am more confident it is correct, see below:

```
&i2c2 {
        status = "okay";

        /* NXP NFC7150 */
        pn544: pn544@28 {
                compatible = "nxp,pn544";
                reg = <0x28>;
                clock-frequency = <400000>;
                interrupt-parent = <&gpio1>;
                interrupt-gpios = <&gpio1 11 GPIO_ACTIVE_LOW>;
                enable-gpios = <&gpio3 12 GPIO_ACTIVE_HIGH>;
        };
};
```

However, I still have problems when I launch the app (slightly different error then above):

```
root@ccimx6ulstarter:~# ./nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...
                              
1515182314:054:911 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)
1515182314:055:179 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)
1515182314:055:258 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)
1515182314:055:334 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)
0:000:440 - NxpHal:     Entering phNxpNciHal_init_monitor
0:000:570 - NxpHal:     Returning with SUCCESS
0:000:690 - NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC
0:000:767 - NxpTml:     NFC - Assign IO pins
0:000:893 - NxpTml:     Pin /sys/class/gpio/gpio11
0:001:130 - NxpTml:     System already has pin /sys/class/gpio/gpio11
0:001:269 - NxpTml:     Direction /sys/class/gpio/gpio11/direction
0:001:536 - NxpTml:     Pin 11 now an input
0:001:646 - NxpTml:     Edge /sys/class/gpio/gpio11/edge
0:001:828 - NxpTml:     Edge-string rising - 6
0:001:922 - NxpTml:     Pin 11 trigger on rising
0:002:006 - NxpTml:     Value /sys/class/gpio/gpio11/value
0:002:177 - NxpTml:     Pin /sys/class/gpio/gpio76
0:002:331 - NxpTml:     System already has pin /sys/class/gpio/gpio76
0:002:420 - NxpTml:     Direction /sys/class/gpio/gpio76/direction
0:002:597 - NxpTml:     Pin 76 now an output
0:002:804 - NxpTml:     Pin 76 now off

0:003:040 - NxpTml:     NFCHW - open I2C bus - /dev/i2c-1
0:003:177 - NxpTml:     NFC - open I2C device - 0x28

0:204:227 - NxpTml:     PN54X - Tml Reader Thread Started................
0:204:452 - NxpTml:     PN54X - Read requested.....
0:204:528 - NxpTml:     PN54X - Invoking I2C Read.....
0:204:755 - NxpTml:     PN54X - Tml Writer Thread Started................
0:204:882 - NxpTml:     PN54X - Tml Writer Thread Running................
0:204:947 - NxpTml:     PN54X - Write requested.....
0:205:008 - NxpTml:     PN54X - Invoking I2C Write.....

0:205:199 - NxpHal:     thread started
0:205:448 - NxpTml:     _i2c_write() errno : 5
0:205:538 - NxpTml:     PN54X - Error in I2C Write.....
0:205:606 - NxpTml:     PN54X - Posting Fresh Write message.....
0:205:890 - NxpTml:     PN54X - Tml Writer Thread Running................
[... 67 lines cut ...]
3:831:890 - NxpTml:     PN54X - Posting Fresh Write message.....
3:832:089 - NxpTml:     PN54X - Tml Writer Thread Running................
3:832:318 - NxpHal:     write error status = 0x1ff
3:832:555 - NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
3:833:940 - NxpTml:     PN54X - Write requested.....
3:834:186 - NxpTml:     PN54X - Invoking I2C Write.....
3:835:482 - NxpNciX:     len =   4 > 20000101
3:835:673 - NxpTml:     PN54X - I2C Write successful.....
3:835:828 - NxpTml:     PN54X - Posting Fresh Write message.....
3:836:015 - NxpTml:     PN54X - Tml Writer Thread Running................
3:836:239 - NxpHal:     write successful status = 0x0
3:836:512 - NxpHal:     Response timer started
3:836:686 - NxpHal:     Waiting after ext cmd sent
4:837:324 - NxpHal:     hal_extns_write_rsp_timeout_cb - write timeout!!!
4:837:913 - NxpHal:     Response timer stopped
4:838:111 - NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
4:838:367 - NxpHal:     NCI_CORE_RESET: Failed
4:840:861 - NxpTml:     PN54X - Write request NOT enabled
```

So, basically my questions are:
1) what is _i2c_write() errno : 5 ?
2) is the IRQ line goes down when the module wants to communicate with the host or goes high?

Thanks again!

jgeslin, 2018-01-08:
Hi,

1. "_i2c_write() errno : 5" mean I2C write failed because I2C request is not acked by the NFC Controller. This is expected when the NFC Controller is in standby mode (the first request wakes-up, the second is properly received).

2. the IRQ is active HIGH, meaning the NFC Controller set the line high to indicates it has something to send (it must trigger a I2C read operation from the host).

You still got the same error: "hal_extns_write_rsp_timeout_cb - write timeout!!!"
It indicates no response to the NCI command ("20000101" meaning NCI CORE_RESET_CMD).
Again, you are mixing kernel driver and Alternative configurations. With option "--enable-alt" you are using the Alternative configuration (nxp5xx-i2c kernel driver not used) as observed in the log. In this case the definition "pn544: pn544@28" in the DTS file has no impact because it only related to the nxp5xx-i2c kernel driver.

Could you check the status of the IRQ line (with a scope or worse with a voltage meter) when the failing occurs ? It should be high but maybe not seen as is by the SW because of wrong GPIO configuration.

Regards,
Jeremy.

paolochiarlone, 2018-01-10:
@jgeslin Hello Jeremy,
Yeah I realized that I was confused between driver mode and user space mode.
So I recompiled the app with:

```
./configure --enable-alt --host=arm-dey-linux-gnueabi --prefix=/usr
make clean
make
```

Problem is that I still get the same error.
This is how I configure the GPIO pins from user space, and it seems correct to me.
GPIO11 is my IRQ (active HIGH 3.3V) and GPIO76 is my ENABLE (active HIGH, 3.3V)

```
#!/bin/sh
cd /sys/class/gpio/
echo 11 > export
echo in > gpio11/direction
echo rising > gpio11/edge
echo 0 > gpio11/active_low
echo 76 > export
echo out > gpio76/direction 
echo 1 > gpio76/value
echo 0 > gpio76/value
```

I can manually toggle GPIO76 and it is connected to pin 18 of the RPi connector.
I also put an LED on the IRQ line (pin 16 of the RPi connector) and I see the line going HIGH after I launch the app. Still, I get the same error:

```
root@ccimx6ulstarter:~# ./nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...
1515554925:360:052 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)
1515554925:360:491 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)
1515554925:360:602 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)
1515554925:360:699 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)
0:000:493 - NxpHal:     Entering phNxpNciHal_init_monitor
0:000:642 - NxpHal:     Returning with SUCCESS
0:000:792 - NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC
0:000:887 - NxpTml:     NFC - Assign IO pins
0:000:975 - NxpTml:     Pin /sys/class/gpio/gpio11
0:001:249 - NxpTml:     System already has pin /sys/class/gpio/gpio11
0:001:457 - NxpTml:     Direction /sys/class/gpio/gpio11/direction
0:001:752 - NxpTml:     Pin 11 now an input
0:001:890 - NxpTml:     Edge /sys/class/gpio/gpio11/edge
0:002:121 - NxpTml:     Edge-string rising - 6
0:002:241 - NxpTml:     Pin 11 trigger on rising
0:002:352 - NxpTml:     Value /sys/class/gpio/gpio11/value
0:002:615 - NxpTml:     Pin /sys/class/gpio/gpio76
0:002:834 - NxpTml:     System already has pin /sys/class/gpio/gpio76
0:002:958 - NxpTml:     Direction /sys/class/gpio/gpio76/direction
0:003:190 - NxpTml:     Pin 76 now an output
0:003:450 - NxpTml:     Pin 76 now off
0:003:552 - NxpTml:     NFCHW - open I2C bus - /dev/i2c-1
0:003:712 - NxpTml:     NFC - open I2C device - 0x28
0:205:864 - NxpTml:     PN54X - Tml Reader Thread Started................
0:206:229 - NxpTml:     PN54X - Read requested.....
0:206:396 - NxpTml:     PN54X - Invoking I2C Read.....
0:206:843 - NxpTml:     PN54X - Tml Writer Thread Started................
0:207:175 - NxpTml:     PN54X - Tml Writer Thread Running................
0:207:345 - NxpTml:     PN54X - Write requested.....
0:207:496 - NxpTml:     PN54X - Invoking I2C Write.....
0:207:843 - NxpHal:     thread started
0:209:609 - NxpNciX:     len =   4 > 20000101
[... 58 lines cut ...]
3:832:934 - NxpTml:     PN54X - Write requested.....
3:833:134 - NxpTml:     PN54X - Invoking I2C Write.....
3:833:656 - NxpTml:     _i2c_write() errno : 5
3:833:845 - NxpTml:     PN54X - Error in I2C Write.....
3:834:000 - NxpTml:     PN54X - Posting Fresh Write message.....
3:834:200 - NxpTml:     PN54X - Tml Writer Thread Running................
3:834:418 - NxpHal:     write error status = 0x1ff
3:834:663 - NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
3:836:020 - NxpTml:     PN54X - Write requested.....
3:836:224 - NxpTml:     PN54X - Invoking I2C Write.....
3:837:533 - NxpNciX:     len =   4 > 20000101
3:837:721 - NxpTml:     PN54X - I2C Write successful.....
3:837:879 - NxpTml:     PN54X - Posting Fresh Write message.....
3:838:068 - NxpTml:     PN54X - Tml Writer Thread Running................
3:838:287 - NxpHal:     write successful status = 0x0
3:838:571 - NxpHal:     Response timer started
3:838:745 - NxpHal:     Waiting after ext cmd sent
4:839:588 - NxpHal:     hal_extns_write_rsp_timeout_cb - write timeout!!!
4:840:185 - NxpHal:     Response timer stopped
4:840:388 - NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
4:840:567 - NxpHal:     NCI_CORE_RESET: Failed
4:843:082 - NxpTml:     PN54X - Write request NOT enabled
```

Do you think the problem is still a misconfiguration of PIN11?

jgeslin, 2018-01-11:
One thing you can already try out (to check i2C bus is well managed) is to simply left wait4interrupt function (inside phTmlNfc_i2c.c) empty. This will trigger i2c read request in loop allowing to handle PN7150 responses but clearly not efficient in term of CPU usage (I2C continuous polling).

Then I would then advise you to instrument (just add printf) this piece of code (wait4interrupt  function) handling the detection of IRQ line high level to fire read request to figure out why it is not working.

paolochiarlone, 2018-01-12:
@jgeslin 
How can I thank you enough?
So, after I found out my board has a bad GPIO pin, I was refreshing the whole tree, recompile, load
and apparently I am able to have it running!!!
THANK YOU!

Only caveat, I still have some minor errors displayed, and wonder what do you think of these?
I scanned a few different NFC card (the 1st is the one in the kit) and this is the result:

```
root@ccimx6ulstarter:~# ./nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	NFC Tag Found

		Type : 		'Type A - Mifare Ul'
		NFCID1 :    	'04 AA 57 D2 9C 39 80 '
		Record Found :
				NDEF Content Max size : 	'868 bytes'
				NDEF Actual Content size : 	'29 bytes'
				ReadOnly :                      'FALSE'
				Type : 				'URI'
				URI : 				'http://www.nxp.com/demoboard/OM5578'

		29 bytes of NDEF data received :
		D1 01 19 55 01 6E 78 70 2E 63 6F 6D 2F 64 65 6D 6F 62 6F 61 72 64 2F 4F 4D 35 35 37 38 

	NFC Tag Lost

Waiting for a Tag/Device...

	NFC Tag Found

		Type : 		'Type A - Mifare Ul'
		NFCID1 :    	'04 AA 57 D2 9C 39 80 '
		NDEF Content : NO, mode=1, tech=9

	MIFARE UL card
		Mifare Read command: 30 01 

		RAW Tag transceive failed
	NFC Tag Lost

Waiting for a Tag/Device...

	NFC Tag Found

		Type : 		'Type A - Mifare Ul'
		NFCID1 :    	'04 45 6E 82 97 3C 80 '
		Record Found :
				NDEF Content Max size : 	'182 bytes'
				NDEF Actual Content size : 	'182 bytes'
				ReadOnly : 					    'TRUE'
				Read NDEF Content Failed
	NFC Tag Lost

Waiting for a Tag/Device...
```
What are:
- NFC Tag Lost
- RAW Tag transceive failed
- Read NDEF Content Failed

Is there something still not quite right?
Thanks again, Paolo

_PS: this is with wait4interrupt enabled/uncommented, as I noticed there was no real difference
when the function was emptied_

jgeslin, 2018-01-12:
"NFC Tag Lost" is the message triggered when the application detect card removal.
I assume there is some RF errors when trying to read the tag content. To better understand what is going on I would need the NCI exchanges. Just enable the realted trace setting NXPLOG_NCIX_LOGLEVEL and NXPLOG_NCIR_LOGLEVEL flags to 0x03 in libnfc-nxp.conf file.

paolochiarlone, 2018-01-13:
OK, so I did as you asked and I noticed that unless I remove the card, the debug prints keep coming,
like if it was stuck in a loop, and when I remove the card, then the dump stops and the error appears...

Here is the debug log for 1 card read:
```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

0:207:272 - NxpNciX:     len =   4 > 20000101
0:234:894 - NxpNciR:     len =   6 > 400003001101
0:236:788 - NxpNciX:     len =   3 > 200100
0:243:005 - NxpNciR:     len =  28 > 40011900031E030008000102038081828302D002FF020004881001A0
0:245:130 - NxpNciX:     len =   6 > 20030301A005
0:248:099 - NxpNciR:     len =   9 > 4003060001A0050100
0:253:159 - NxpNciX:     len =   4 > 20000101
0:257:396 - NxpNciR:     len =   6 > 400003001101
0:259:119 - NxpNciX:     len =   3 > 200100
0:265:010 - NxpNciR:     len =  28 > 40011900031E030008000102038081828302D002FF020004881001A0
0:266:778 - NxpNciX:     len =   3 > 2F0200
0:269:503 - NxpNciR:     len =   8 > 4F02050000009F97
0:272:178 - NxpNciX:     len =  10 > 20030703A002A003A004
0:276:218 - NxpNciR:     len =  17 > 40030E0003A0020101A0030108A0040101
0:278:178 - NxpNciX:     len =   8 > 20020501A0030108
0:279:896 - NxpNciR:     len =   5 > 4002020000
0:284:085 - NxpNciX:     len =  26 > 20021702A0EA08FFFFFFFFFFFFFFFFA0EB08FFFFFFFFFFFFFFFF
0:286:099 - NxpNciR:     len =   5 > 4002020000
0:287:909 - NxpNciX:     len =   8 > 20020501A0440100
0:289:665 - NxpNciR:     len =   5 > 4002020000
0:291:762 - NxpNciX:     len =  10 > 20020701A00E03066400
0:293:487 - NxpNciR:     len =   5 > 4002020000
0:316:019 - NxpNciX:     len = 166 > 2002A313A00D0604359001F401A00D06064401900300A00D060630B0011000A00D0606420200FFFFA00D03063F04A00D0620428800FFFFA00D0422442200A00D06222D50340C00A00D063242F800FFFFA00D06342D24370C00A00D06343386800070A00D0434442200A00D06422D15450D00A00D0446442200A00D06462D05590E00A00D0644428800FFFFA00D06562D059F0C00A00D0654428800FFFFA00D060A3380860070
0:359:673 - NxpNciR:     len =   5 > 4002020000
0:363:628 - NxpNciX:     len =  24 > 20021501A01D115733141700AA850080552A040063000000
0:365:410 - NxpNciR:     len =   5 > 4002020000
0:367:709 - NxpNciX:     len =  12 > 20020902A05E0101A0400100
0:369:655 - NxpNciR:     len =   5 > 4002020000
0:371:231 - NxpNciX:     len =   6 > 20030301A085
0:374:154 - NxpNciR:     len =  12 > 4003090001A0850450082800
0:375:379 - NxpNciX:     len =   4 > 2F000101
0:376:870 - NxpNciR:     len =   4 > 4F000100
0:383:692 - NxpNciX:     len =  46 > 20022B0D2801002101003001083101033304010203045401065001025B010060010E80010181010182010E180101
0:405:028 - NxpNciR:     len =   5 > 4002020000
0:407:011 - NxpNciX:     len =  10 > 21010700010103000105
0:411:431 - NxpNciR:     len =   4 > 41010100
0:414:637 - NxpNciX:     len =  16 > 21000D04040302050303800180030201
0:430:691 - NxpNciR:     len =   4 > 41000100
0:431:963 - NxpNciX:     len =   4 > 22000101
0:436:391 - NxpNciR:     len =   5 > 4200020001
0:438:392 - NxpNciR:     len =   8 > 6200050101010100
0:440:324 - NxpNciX:     len =   9 > 200406030101020101
0:442:235 - NxpNciR:     len =   7 > 40040400FF0103
0:443:852 - NxpNciX:     len =   5 > 0300028103
0:445:627 - NxpNciR:     len =   6 > 600603010301
0:447:237 - NxpNciR:     len =   5 > 0300028180
0:449:012 - NxpNciX:     len =   8 > 03000581010302C0
0:450:954 - NxpNciR:     len =   6 > 600603010301
0:452:640 - NxpNciR:     len =   5 > 0300028180
0:455:266 - NxpNciX:     len =  14 > 03000B8101011EDA00001ADA0000
0:462:884 - NxpNciR:     len =   6 > 600603010301
0:464:493 - NxpNciR:     len =   5 > 0300028180
0:465:934 - NxpNciX:     len =   5 > 2201020101
0:470:509 - NxpNciR:     len =   4 > 42010100
0:475:255 - NxpNciR:     len =   8 > 6200050202018000
0:623:285 - NxpNciX:     len =   6 > 030003810204
0:625:188 - NxpNciR:     len =   6 > 600603010301
0:626:955 - NxpNciR:     len =   6 > 030003818000
[... 114 lines cut ...]
6:388:079 - NxpNciR:     len =   6 > 600603010001
6:392:497 - NxpNciR:     len =  20 > 00001104AA5771D29C3980F7480000E1106D0000
6:519:970 - NxpNciX:     len =   5 > 0000023000
6:522:614 - NxpNciR:     len =   6 > 600603010001
6:527:418 - NxpNciR:     len =  20 > 00001104AA5771D29C3980F7480000E1106D0000
6:655:000 - NxpNciX:     len =   5 > 0000023000
6:657:761 - NxpNciR:     len =   6 > 600603010001
6:662:560 - NxpNciR:     len =  20 > 00001104AA5771D29C3980F7480000E1106D0000
6:789:930 - NxpNciX:     len =   5 > 0000023000
6:792:316 - NxpNciR:     len =   6 > 600603010001
6:797:243 - NxpNciR:     len =  20 > 00001104AA5771D29C3980F7480000E1106D0000
6:924:856 - NxpNciX:     len =   5 > 0000023000
6:927:640 - NxpNciR:     len =   6 > 600603010001
6:932:380 - NxpNciR:     len =  20 > 00001104AA5771D29C3980F7480000E1106D0000
7:059:764 - NxpNciX:     len =   5 > 0000023000
7:062:347 - NxpNciR:     len =   6 > 600603010001
7:212:556 - NxpNciX:     len =   4 > 21060103
7:214:783 - NxpNciR:     len =   4 > 41060100
7:217:371 - NxpNciR:     len =   5 > 6106020300
	NFC Tag Lost

Waiting for a Tag/Device...
```

Please let me know and thanks again so much for your incredible support!

jgeslin, 2018-01-15:
The  behavior is exactly as expected.
The stack sends presence check command ("0000023000") in a loop to detect when the tag is being removed. It avoid restarting the discovery loop and then rediscovering the same tag again and again.
"NFC Tag Lost" is not an error, it only indicates to the user that the tag was removed and that the application is now ready to discover a new device.

paolochiarlone, 2018-02-12:
@jgeslin I just wanted to thank you for the incredible support!

closed 2018-02-12 by paolochiarlone