NXPNFCLinux/linux_libnfc-nci #66: Maximum size of NDEF Message is 1024?

sabelka, 2018-01-09:
In the function nativeNdef_createMime in src/service/interface/nativeNdef.cpp the maximum size of a message is limited to 1024 bytes.

Is there a reason for this limit?

If not, I'd suggest setting the limit to the size of the output buffer like this:
```
--- a/src/service/interface/nativeNdef.cpp
+++ b/src/service/interface/nativeNdef.cpp
@@ -294,7 +294,7 @@ INT32 nativeNdef_createMime(char *mimeType, UINT8 *mimeData, UINT32 mimeDataLeng
         return 0;
     }
 
-    status = NDEF_MsgAddRec(outNdefBuff, 1024, &current_size, NDEF_TNF_MEDIA, (UINT8 *)mimeType, mimeTypeLength, NULL, 0,
+    status = NDEF_MsgAddRec(outNdefBuff, outBufferLen, &current_size, NDEF_TNF_MEDIA, (UINT8 *)mimeType, mimeTypeLength, NULL, 0,
                                     (UINT8*)mimeData, (UINT32)mimeDataLength);
 
     if (status != NFA_STATUS_OK )
```

jgeslin, 2018-01-10:
Thank you for the remark.
We will consider it for the next release.

Jeremy.

jgeslin, 2018-08-13:
Latest release R2.4 now includes the fix.

closed 2018-11-07 by jgeslin