NXPNFCLinux/linux_libnfc-nci #68: About Card Detection / Discovery Mode at low field

JuanAngelSG, 2018-01-19:
Hi, 

I would like to add one question regarding the behavour of the reader with card detection. I am using an application based on mRamiroGonzalez application (but with some modifications). See #34. 

I have observed that for a certain distance, the programs runs smoothly up to line:
main() - nfcManager_enableDiscovery(DEFAULT_NFA_TECH_MASK, 0x01, 0, 0);
and, although the card responses everything correctly, the reader does not send the APDUs specified inside the while(1) loop. At least they cannot be seen on the oscilloscope or the spy tool.

[image]

PPS sets the communication to 424 kbps (in both directions) After the "B2" block (Presence check?), the reader sends a RESET (after around 470 ms, the complete block takes 500 ms) and starts from WUPA again.

[image]

This behaviour disappears if the card is moved towards the reader antenna (so for shorter distances this issue does not exist). Next picture shows this "normal" behaviour.

[image]

Why could this be? How does the reader board detects that the card is present? Does it require a minimum load? Does this "low load" overrule any possible data reception? Is maybe a sensitivity issue, where the reader cannot decode the response from the card?

I also attach the logfile with some comments on it.
[log.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1646242/log.txt)

I looked at the documentation, but found nothing that I could use to find this out.

Best Regards,
Juan

jgeslin, 2018-01-22:
Hi,

From the logs, and based on your explanation of the scenario, the issue seems to be related to the reader performance (in distance) towards the card used. We can see in the scope traces that the card load is smaller at longer distance, this may not be sufficient according to the reader sensitivity.

Here is 2 suggestions you can try to improve things:
- Prevent baudrate switch to 424 by setting PI_BIT_RATE NCI parameter (0x21) to 0x00 (however this is the default value set inside libnfc-nxp.conf file)
- If it is PN7150, increase reader sensitivity (through CLIF_ANA_RX_REG settings in reader mode) as described in https://www.nxp.com/docs/en/application-note/AN11755.pdf (this relates to settings "A0, 0D, 04, 34, 44, 22, 00," for type A and "" for type B inside libnfc-nxp.conf file.

Regards,
Jeremy.

JuanAngelSG, 2018-01-22:
Hi Jeremy,

Many thanks for the information. I understand, that when the load is small, there might be some sensitivity issues.

To hint number one... I´ll consider it, but I wanted to have a higher bitrate than 106 kbps.

To hint number two... I am using the PN7120 (NFC Controller Board, Raspberry Shield), I have seen the AN11564 document, and I know the register is also present in that IC, but I am not quite sure how to modify it (Register not present in libnfc-nxp.conf). Would this be possible as well?

BR,
Juan