NXPNFCLinux/linux_libnfc-nci #69: NxpTml:     _i2c_write() errno : b

edurne1893, 2018-01-29:
Hello, I'm trying to make my NFC controller work in a custom platform running Linux 4.6.
For that, I use configuration option:

./configure --enable-pn7150 --enable-alt --host=arm-linux-gnueabihf

I get the following output (with debug information enabled)
./nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

93:283:527 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

93:283:662 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

93:283:709 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

93:283:755 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

0:000:216 - NxpHal:     Entering phNxpNciHal_init_monitor
0:000:280 - NxpHal:     Returning with SUCCESS
0:000:355 - NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC

0:000:402 - NxpTml:     NFC - Assign IO pins

0:000:445 - NxpTml:     Pin /sys/class/gpio/gpio904

0:000:541 - NxpTml:     Create pin /sys/class/gpio/gpio904

0:100:873 - NxpTml:     Direction /sys/class/gpio/gpio904/direction

0:101:095 - NxpTml:     Pin 904 now an input

0:101:164 - NxpTml:     Edge /sys/class/gpio/gpio904/edge

0:101:270 - NxpTml:     Edge-string rising - 6

0:101:493 - NxpTml:     Pin 904 trigger on rising

0:101:567 - NxpTml:     Value /sys/class/gpio/gpio904/value

0:101:675 - NxpTml:     Pin /sys/class/gpio/gpio905

0:101:754 - NxpTml:     Create pin /sys/class/gpio/gpio905

0:202:006 - NxpTml:     Direction /sys/class/gpio/gpio905/direction

0:202:179 - NxpTml:     Pin 905 now an output

0:202:307 - NxpTml:     Pin 905 now off

0:202:361 - NxpTml:     NFCHW - open I2C bus - /dev/i2c-0

0:202:434 - NxpTml:     NFC - open I2C device - 0x28

0:402:636 - NxpTml:     PN54X - Tml Reader Thread Started................

0:402:837 - NxpHal:     thread started
0:402:885 - NxpTml:     PN54X - Tml Writer Thread Started................

0:402:951 - NxpTml:     PN54X - Read requested.....

0:403:018 - NxpTml:     PN54X - Tml Writer Thread Running................

0:403:103 - NxpTml:     PN54X - Invoking I2C Read.....

0:403:181 - NxpTml:     PN54X - Write requested.....

0:403:303 - NxpTml:     PN54X - Invoking I2C Write.....

0:403:394 - NxpTml:     _i2c_write() errno : b
0:403:451 - NxpTml:     _i2c_write() errno : b
0:403:502 - NxpTml:     _i2c_write() errno : b
0:403:551 - NxpTml:     _i2c_write() errno : b
0:403:600 - NxpTml:     _i2c_write() errno : b
0:403:649 - NxpTml:     _i2c_write() errno : b
0:403:698 - NxpTml:     _i2c_write() errno : b
0:403:746 - NxpTml:     _i2c_write() errno : b
0:403:795 - NxpTml:     _i2c_write() errno : b
0:403:844 - NxpTml:     _i2c_write() errno : b
0:403:893 - NxpTml:     _i2c_write() errno : b
0:403:942 - NxpTml:     _i2c_write() errno : b
0:403:990 - NxpTml:     _i2c_write() errno : b
0:404:040 - NxpTml:     _i2c_write() errno : b
0:404:089 - NxpTml:     _i2c_write() errno : b
0:404:138 - NxpTml:     _i2c_write() errno : b
0:404:188 - NxpTml:     _i2c_write() errno : b
0:404:236 - NxpTml:     _i2c_write() errno : b
0:404:286 - NxpTml:     _i2c_write() errno : b
0:404:334 - NxpTml:     _i2c_write() errno : b
0:404:383 - NxpTml:     _i2c_write() errno : b
0:404:432 - NxpTml:     _i2c_write() errno : b
0:404:480 - NxpTml:     _i2c_write() errno : b
0:404:529 - NxpTml:     _i2c_write() errno : b
0:404:578 - NxpTml:     _i2c_write() errno : b
0:404:627 - NxpTml:     _i2c_write() errno : b
0:404:675 - NxpTml:     _i2c_write() errno : b
0:404:724 - NxpTml:     _i2c_write() errno : b
0:404:772 - NxpTml:     _i2c_write() errno : b
0:404:821 - NxpTml:     _i2c_write() errno : b
0:404:870 - NxpTml:     _i2c_write() errno : b
0:404:919 - NxpTml:     _i2c_write() errno : b
0:404:968 - NxpTml:     _i2c_write() errno : b
0:405:016 - NxpTml:     _i2c_write() errno : b
0:405:065 - NxpTml:     _i2c_write() errno : b
0:405:113 - NxpTml:     _i2c_write() errno : b
0:405:162 - NxpTml:     _i2c_write() errno : b
0:405:211 - NxpTml:     _i2c_write() errno : b
0:405:260 - NxpTml:     _i2c_write() errno : b
0:405:308 - NxpTml:     _i2c_write() errno : b
0:405:356 - NxpTml:     _i2c_write() errno : b
0:405:405 - NxpTml:     _i2c_write() errno : b
0:405:453 - NxpTml:     _i2c_write() errno : b
0:405:502 - NxpTml:     _i2c_write() errno : b
0:405:551 - NxpTml:     _i2c_write() errno : b
0:405:599 - NxpTml:     _i2c_write() errno : b
0:405:648 - NxpTml:     _i2c_write() errno : b
0:405:696 - NxpTml:     _i2c_write() errno : b
0:405:745 - NxpTml:     _i2c_write() errno : b
0:405:794 - NxpTml:     _i2c_write() errno : b
0:405:843 - NxpTml:     _i2c_write() errno : b
0:405:891 - NxpTml:     _i2c_write() errno : b
0:405:940 - NxpTml:     _i2c_write() errno : b
0:405:989 - NxpTml:     _i2c_write() errno : b

And it continues like that forever.
My dts definition for i2c is just the following. 
&i2c0 {
	clock-frequency = <400000>;
	status = "okay";
};

Thank you in advance for your response.

jgeslin, 2018-01-29:
Hi,

Errno 11 relates to "resource temporarily unavailable" or "No more processes".
You seems to have issue accessing the I2C ressources. 
Did you check that you have appropriate rights on /dev/i2c-0 ?
For instance, can you run "i2cdetect -y 0" ?

Regards,
Jeremy.

edurne1893, 2018-01-29:
Than you for your help, Jeremy. 
I think rights are OK. 
I can run i2cdetect. I obtain the following:
# i2cdetect -y 0
i2cdetect: warning: can't use SMBus quick write command, will skip some addresses
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:
10:
20:
30: -- -- -- -- -- -- -- --
40:
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60:
70:

Best regards,
Edurne

jgeslin, 2018-01-29:
Then, check you can see PN7150:
- set VEN pin to high: 
`echo "1" > /sys/class/gpio/gpio905/value`
- check i2c devices presence: 
`i2cdetect -y 0` (you must run the command twice in less than 1 second to get around PN7150 standby mode)

edurne1893, 2018-01-29:
Same result as before with i2cdetect, even running it twice. So I'm not sure if I'm able to see PN7150.
echo "0" > /sys/class/gpio/gpio905/value
cat /sys/class/gpio/gpio905/value
0
echo "1" > /sys/class/gpio/gpio905/value
cat /sys/class/gpio/gpio905/value
1
i2cdetect -y 0
i2cdetect: warning: can't use SMBus quick write command, will skip some addresses
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:
10:
20:
30: -- -- -- -- -- -- -- --
40:
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60:
70:
i2cdetect -y 0
i2cdetect: warning: can't use SMBus quick write command, will skip some addresses
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:
10:
20:
30: -- -- -- -- -- -- -- --
40:
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60:
70:

jgeslin, 2018-01-29:
Then to my mind the next step is to check VEN and I2C signals using a scope (or I2C sniffer) when running this scenario (direct access to gpio and i2c).

edurne1893, 2018-01-30:
After changing my pins (VEN is GPIO5 and INT is GPIO4) and I2C bus (I2C-2), I have made some progress. Now I think I can see the the PN7150:
root@imx6qvab820:/sys/class/gpio/gpio5# echo 0 > value
root@imx6qvab820:/sys/class/gpio/gpio5# i2cdetect -y 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- UU -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: 20 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- 64 -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
root@imx6qvab820:/sys/class/gpio/gpio5# echo 1 > value
root@imx6qvab820:/sys/class/gpio/gpio5# i2cdetect -y 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 UU 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: -- -- -- -- -- -- -- -- 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 -- -- 74 -- -- --

But now I have different errors:
root@imx6qvab820:/sys/class/gpio/gpio5# nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

1517258817:038:194 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

1517258817:038:274 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

1517258817:038:307 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

1517258817:038:336 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

0:000:165 - NxpHal:     Entering phNxpNciHal_init_monitor
0:000:209 - NxpHal:     Returning with SUCCESS
0:000:260 - NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC

0:000:292 - NxpTml:     NFC - Assign IO pins

0:000:323 - NxpTml:     Pin /sys/class/gpio/gpio4

0:000:382 - NxpTml:     Create pin /sys/class/gpio/gpio4

0:100:673 - NxpTml:     Direction /sys/class/gpio/gpio4/direction

0:100:824 - NxpTml:     Pin 4 now an input

0:100:868 - NxpTml:     Edge /sys/class/gpio/gpio4/edge

0:100:943 - NxpTml:     Edge-string rising - 6

0:101:029 - NxpTml:     Pin 4 trigger on rising

0:101:073 - NxpTml:     Value /sys/class/gpio/gpio4/value

0:101:147 - NxpTml:     Pin /sys/class/gpio/gpio5

0:101:211 - NxpTml:     System already has pin /sys/class/gpio/gpio5

0:101:249 - NxpTml:     Direction /sys/class/gpio/gpio5/direction

0:101:317 - NxpTml:     Pin 5 now an output

0:101:394 - NxpTml:     Pin 5 now off

0:101:427 - NxpTml:     NFCHW - open I2C bus - /dev/i2c-2

0:101:475 - NxpTml:     NFC - open I2C device - 0x28

0:301:636 - NxpTml:     PN54X - Tml Reader Thread Started................

0:301:729 - NxpTml:     PN54X - Tml Writer Thread Started................

0:301:808 - NxpTml:     PN54X - Tml Writer Thread Running................

0:301:832 - NxpTml:     0:301:817 - NxpHal:     thread started0:301:847 - PN54X - Read requested.....

NxpTml:     PN54X - Write requested.....

0:301:975 - 0:301:959 - NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     PN54X - Invoking I2C Read.....

0:302:251 - NxpTml:     _i2c_write() errno : 5
0:302:293 - NxpTml:     PN54X - Error in I2C Write.....

0:302:322 - NxpTml:     PN54X - Posting Fresh Write message.....

0:302:409 - NxpTml:     PN54X - Tml Writer Thread Running................

0:302:431 - NxpHal:     write error status = 0x1ff
0:302:461 - NxpTml:     _i2c_read() [hdr] errno : 5
0:302:550 - NxpTml:     0:302:519 - NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
PN54X - Error in I2C Read.....

0:302:632 - NxpTml:     PN54X - Read requested.....

0:302:678 - NxpTml:     PN54X - Invoking I2C Read.....

0:303:646 - NxpTml:     PN54X - Write requested.....

0:303:678 - NxpTml:     PN54X - Invoking I2C Write.....

0:395:902 - NxpTml:     _i2c_read() [hdr] errno : 6e
0:395:938 - NxpTml:     PN54X - Error in I2C Read.....

0:395:966 - NxpTml:     PN54X - Read requested.....

0:395:992 - NxpTml:     PN54X - Invoking I2C Read.....

0:495:904 - NxpTml:     _i2c_write() errno : 6e
0:495:949 - NxpTml:     PN54X - Error in I2C Write.....

0:495:977 - NxpTml:     PN54X - Posting Fresh Write message.....

0:496:009 - NxpTml:     PN54X - Tml Writer Thread Running................

0:496:024 - NxpHal:     write error status = 0x1ff
0:496:095 - NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
0:497:161 - NxpTml:     PN54X - Write requested.....

0:497:192 - NxpTml:     PN54X - Invoking I2C Write.....

0:595:897 - NxpTml:     _i2c_read() [hdr] errno : 6e
0:595:941 - NxpTml:     PN54X - Error in I2C Read.....

0:595:968 - NxpTml:     PN54X - Read requested.....

0:595:993 - NxpTml:     PN54X - Invoking I2C Read.....

0:695:903 - NxpTml:     _i2c_write() errno : 6e
0:695:947 - NxpTml:     PN54X - Error in I2C Write.....

0:695:975 - NxpTml:     PN54X - Posting Fresh Write message.....

0:696:007 - NxpTml:     PN54X - Tml Writer Thread Running................

And always getting error 6e.

edurne1893, 2018-01-30:
I gety this after six __i2c_write() errno_ : 6e errors:

1:201:261 - NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode (max count = 0x6)
1:301:067 - NxpTml:     _i2c_read() [hdr] errno : 6e
1:301:107 - NxpTml:     PN54X - Error in I2C Read.....

1:301:133 - NxpTml:     PN54X - Read requested.....

1:301:157 - NxpTml:     PN54X - Invoking I2C Read.....

1:401:069 - NxpTml:     _i2c_read() [hdr] errno : 6e
1:401:108 - NxpTml:     PN54X - Error in I2C Read.....

1:401:133 - NxpTml:     PN54X - Read requested.....

1:401:158 - NxpTml:     PN54X - Invoking I2C Read.....

1:401:359 - NxpHal:     PN54X Reset - SUCCESS

1:401:401 - NxpTml:     1:401:416 - NxpHal:     phNxpNciHal_write failed for hal ext
1:401:454 - NxpHal:     Force FW Download, NFCC not coming out from Standby
_i2c_read() [hdr] errno : 5

jgeslin, 2018-01-30:
I see weird thing: running i2cdetect it looks you have a lot of devices conected on the bus (almost all address are acknowledged). This looks nearly impossible !

Additionally, i2c_read requests should only be fired when IRQ line is high (connected to GPIO4 in your case), can you check the level of this line when running the scenario ?

Would you have the capability to spy the I2C bus (a scope can be sufficient) ?

edurne1893, 2018-01-30:
Yes, I have also noticed that. It's strange. 
But the point is that...I have been analyzing the different signals, and after rebooting the system, everything works perfectly. 
I suppose one of the lines or maybe the bus wasn't initialized. 

THANK YOU so much for you wonderful support Jeremy.

closed 2018-08-13 by jgeslin

lucar91, 2021-05-05:
Hello,  
I use a home made card which I connect to PN7150 controller via I2C. Some time ago @jgeslin solved my problem "No device loop" with a [patch](https://www.google.com/url?q=https://github.com/NXPNFCLinux/linux_libnfc-nci/files/3973575/Patch-NoDevice_loop.txt&source=gmail&ust=1620284696802000&usg=AFQjCNE2Rr5nRvbvGO_wKUy3EAz9iBwfhw). 

Now I have encountered another problem: when the cable is connected, the application works and the logs are in log_withCable.txt, when the cable is NOT connected I have the same error as edurne1893 (log_noCable.txt). 

This happens when I call nfcManager_doInitialize function. Do you know what could be the problem?

 
i2cdetect -y 0   (with and without cable):

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  
```

[log_noCable.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/6425590/log_noCable.txt)
[log_withCable.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/6425591/log_withCable.txt)

lucar91, 2021-05-07:
The problem is probably due to the absence of the pull-up resistors on the I2C bus on the NFC connector side. I'm trying to apply this solution:

```
echo 133 > /sys/class/gpio/export                #export the reset pin (133 in my case)
echo out > /sys/class/gpio/gpio133/direction  #change direction
echo "1" > /sys/class/gpio/gpio133/value       #set level pin to high
i2cdetect -y 0                                #check if there are connected controllers.
```

If controller are not connected: 

```
echo 127 > /sys/class/gpio/export                   #export  pin NFC_I2C_DISABLE (127 in my case)
echo out > /sys/class/gpio/gpio127/direction  #set as output
echo 1 > /sys/class/gpio/gpio127/value          #disabling bus
```

these commands isolate the section of the I2C on the NFC connector side.

And it works: nfcManager_doInitialize does not enter the loop.

Thanks anyway. If anyone has more efficient solutions they are accepted :)