NXPNFCLinux/linux_libnfc-nci #70: STATUS_INVALID_PARAM issue

tomicedani, 2018-01-29:
Hi there!

I've setup a OM5578/PN7150S on a Raspberry Pi and have been playing around with low power mode configurations.

So far I haven't had any issues in getting the device to work, or into low power mode though I've come across an issue I can't seem to solve.

This happened all of a sudden to my device, as I was using it with no issues for a couple of days. I purchased a new OM5578 which fixed the issue but again after a few days this comes up again.

As a note when I first saw the error I was powering the device externally, and not through the Pi's power supply (4.5V system and 3.3V for IO). That being said, I still see the issue running on the Pi power supply too.

The problem is when I run the demo app I get the following error. Both `/etc/libnfc-nxp.conf `and `/etc/libnfc-brcm.conf` are the default when I see this.

```0:000:579 - NxpHal:     Entering phNxpNciHal_init_monitor
0:001:211 - NxpHal:     Returning with SUCCESS
0:204:367 - NxpHal:     thread started
0:205:316 - NxpHal:     write error status = 0x1ff
0:205:681 - NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
0:207:743 - NxpNciX:     len =   4 > 20000101
0:208:145 - NxpHal:     write successful status = 0x0
0:208:555 - NxpHal:     Response timer started
0:208:820 - NxpHal:     Waiting after ext cmd sent
0:211:117 - NxpNciR:     len =   6 > 400003001101
0:211:503 - NxpHal:     read successful status = 0x0
0:211:839 - NxpHal:     Response timer stopped
0:212:062 - NxpHal:     Checking response
0:212:906 - NxpNciX:     len =   3 > 200100
0:213:276 - NxpHal:     write successful status = 0x0
0:213:625 - NxpHal:     Response timer started
0:213:878 - NxpHal:     Waiting after ext cmd sent
0:217:240 - NxpNciR:     len =  28 > 40011900031E030008000102038081828302D002FF020004881001A0
0:217:772 - NxpHal:     read successful status = 0x0
0:218:046 - NxpHal:     NxpNci> FW Version: 10.1.a0
0:218:585 - NxpHal:     Response timer stopped
0:218:857 - NxpHal:     Checking response
0:220:124 - NxpNciX:     len =   6 > 20030301A005
0:220:586 - NxpHal:     write successful status = 0x0
0:220:946 - NxpHal:     Response timer started
0:221:201 - NxpHal:     Waiting after ext cmd sent
0:222:841 - NxpNciR:     len =   9 > 4003060001A0050100
0:223:215 - NxpHal:     read successful status = 0x0
0:223:482 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
0:223:786 - NxpHal:     Response timer stopped
0:224:039 - NxpHal:     Checking response
0:224:273 - NxpHal:     i2c_fragmentation_status existing
0:225:657 - NxpHal:     FW version for FW file = 0x0
0:225:972 - NxpHal:     FW version from device = 0x1001a0
0:227:799 - NxpNciX:     len =   4 > 20000101
0:228:165 - NxpHal:     write successful status = 0x0
0:231:183 - NxpNciR:     len =   6 > 400003001101
0:231:579 - NxpHal:     read successful status = 0x0
0:232:781 - NxpNciX:     len =   3 > 200100
0:233:239 - NxpHal:     write successful status = 0x0
0:236:751 - NxpNciR:     len =  28 > 40011900031E030008000102038081828302D002FF020004881001A0
0:237:149 - NxpHal:     read successful status = 0x0
0:237:593 - NxpHal:     NxpNci> FW Version: 10.1.a0
0:238:752 - NxpNciX:     len =   3 > 2F0200
0:239:207 - NxpHal:     write successful status = 0x0
0:239:557 - NxpHal:     Response timer started
0:239:816 - NxpHal:     Waiting after ext cmd sent
0:241:354 - NxpNciR:     len =   8 > 4F02050000009F97
0:241:719 - NxpHal:     read successful status = 0x0
0:242:065 - NxpHal:     Response timer stopped
0:242:526 - NxpHal:     Checking response
0:244:221 - NxpNciX:     len =  10 > 20030703A002A003A004
0:244:599 - NxpHal:     write successful status = 0x0
0:244:893 - NxpHal:     Response timer started
0:245:171 - NxpHal:     Waiting after ext cmd sent
0:247:554 - NxpNciR:     len =  17 > 40030E0003A0020101A0030108A0040101
0:247:960 - NxpHal:     read successful status = 0x0
0:248:231 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
0:248:594 - NxpHal:     Response timer stopped
0:248:852 - NxpHal:     Checking response
0:250:447 - NxpNciX:     len =   8 > 20020501A0030108
0:250:943 - NxpHal:     write successful status = 0x0
0:251:296 - NxpHal:     Response timer started
0:251:552 - NxpHal:     Waiting after ext cmd sent
0:252:787 - NxpNciR:     len =   5 > 4002020000
0:253:153 - NxpHal:     read successful status = 0x0
0:253:424 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
0:253:731 - NxpHal:     Response timer stopped
0:253:982 - NxpHal:     Checking response
0:254:237 - NxpHal:     phNxpNciHal_check_factory_reset file not present = /data/nfc/nfaStorage.bin1
0:257:295 - NxpNciX:     len =  26 > 20021702A0EA08FFFFFFFFFFFFFFFFA0EB08FFFFFFFFFFFFFFFF
0:257:816 - NxpHal:     write successful status = 0x0
0:258:333 - NxpHal:     Response timer started
0:258:702 - NxpHal:     Waiting after ext cmd sent
0:259:987 - NxpNciR:     len =   5 > 4002020000
0:260:367 - NxpHal:     read successful status = 0x0
0:260:761 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
0:261:082 - NxpHal:     Response timer stopped
0:261:315 - NxpHal:     Checking response
0:262:651 - NxpNciX:     len =   8 > 20020501A0440100
0:263:018 - NxpHal:     write successful status = 0x0
0:263:329 - NxpHal:     Response timer started
0:263:585 - NxpHal:     Waiting after ext cmd sent
0:264:841 - NxpNciR:     len =   5 > 4002020000
0:265:215 - NxpHal:     read successful status = 0x0
0:265:489 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
0:265:791 - NxpHal:     Response timer stopped
0:266:043 - NxpHal:     Checking response
0:266:460 - NxpHal:     Performing TVDD Settings
0:268:188 - NxpNciX:     len =  10 > 20020701A00E03066400
0:268:652 - NxpHal:     write successful status = 0x0
0:269:007 - NxpHal:     Response timer started
0:269:265 - NxpHal:     Waiting after ext cmd sent
0:270:559 - NxpNciR:     len =   5 > 4002020000
0:270:939 - NxpHal:     read successful status = 0x0
0:271:205 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
0:271:509 - NxpHal:     Response timer stopped
0:271:763 - NxpHal:     Checking response
0:271:998 - NxpHal:     Performing RF Settings BLK 1
0:288:149 - NxpNciX:     len = 166 > 2002A313A00D0604359001F401A00D06064401900300A00D060630B0011000A00D0606420200FFFFA00D03063F04A00D0620428800FFFFA00D0422442200A00D06222D50340C00A00D063242F800FFFFA00D06342D24370C00A00D06343386800070A00D0434442200A00D06422D15450D00A00D0446442200A00D06462D05590E00A00D0644428800FFFFA00D06562D059F0C00A00D0654428800FFFFA00D060A3380860070
0:288:575 - NxpHal:     write successful status = 0x0
0:288:925 - NxpHal:     Response timer started
0:289:178 - NxpHal:     Waiting after ext cmd sent
0:295:033 - NxpNciR:     len =  39 > 4002240911A00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00D
0:295:378 - NxpHal:     read successful status = 0x0
0:295:607 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_INVALID_PARAM
0:295:868 - NxpHal:     Response timer stopped
0:296:081 - NxpHal:     Checking response
0:497:066 - NxpHal:     phNxpNciHal_CheckValidFwVersion current_major_no = 0x0
0:797:968 - NxpHal:     Calling Seq handler for FW Download 
```

jgeslin, 2018-01-29:
Hi,

Can you confirm that when the issue occurs first time you were using a previous libnfc-nxp conf file (basically prior to commit https://github.com/NXPNFCLinux/linux_libnfc-nci/commit/9b774ef656c22c4a14018c8916691bd708c95de5#diff-fca6cf6d5c39d91a1ec38e30f4956fa8 aiming to prevent this issue) ?

Regards,
Jeremy.

tomicedani, 2018-01-29:
I have always used the latest version of that config file, i.e. with 

```
NXP_CORE_CONF_EXTN={20, 02, 09, 02,
    A0, 5E, 01, 01,
    A0, 40, 01, 00
}

NXP_I2C_FRAGMENTATION_ENABLED=0x00

NXP_CORE_CONF={ 20, 02, 2B, 0D,
    28, 01, 00,
    21, 01, 00,
    30, 01, 08,
    31, 01, 03,
    33, 04, 01, 02, 03, 04,
    54, 01, 06,
    50, 01, 02,
    5B, 01, 00,
    60, 01, 0E,
    80, 01, 01,
    81, 01, 01,
    82, 01, 0E,
    18, 01, 01
}

NXP_RF_CONF_BLK_1={ 20, 02, A3, 13,
    A0, 0D, 06, 04, 35, 90, 01, F4, 01,
    A0, 0D, 06, 06, 44, 01, 90, 03, 00,
    A0, 0D, 06, 06, 30, B0, 01, 10, 00,
    A0, 0D, 06, 06, 42, 02, 00, FF, FF,
    A0, 0D, 03, 06, 3F, 04,
    A0, 0D, 06, 20, 42, 88, 00, FF, FF,
    A0, 0D, 04, 22, 44, 22, 00,
    A0, 0D, 06, 22, 2D, 50, 34, 0C, 00,
    A0, 0D, 06, 32, 42, F8, 00, FF, FF,
    A0, 0D, 06, 34, 2D, 24, 37, 0C, 00,
    A0, 0D, 06, 34, 33, 86, 80, 00, 70,
    A0, 0D, 04, 34, 44, 22, 00,
    A0, 0D, 06, 42, 2D, 15, 45, 0D, 00,
    A0, 0D, 04, 46, 44, 22, 00,
    A0, 0D, 06, 46, 2D, 05, 59, 0E, 00,
    A0, 0D, 06, 44, 42, 88, 00, FF, FF,
    A0, 0D, 06, 56, 2D, 05, 9F, 0C, 00,
    A0, 0D, 06, 54, 42, 88, 00, FF, FF,
    A0, 0D, 06, 0A, 33, 80, 86, 00, 70
}
```

At some point I did modify `NXP_CORE_CONF_EXTN` for low power mode as so:

```
NXP_CORE_CONF_EXTN={20, 02, 11, 04,
    A0, 5E, 01, 01,
    A0, 40, 01, 01,
    A0, 41, 01 ,04,
    A0, 43, 01, 00
}
```

jgeslin, 2018-01-29:
Try commenting out NXP_RF_CONF_BLK_1 definition in libnfc-nxp.conf file.

tomicedani, 2018-01-29:
Goes into a loop:

```

0:686:215 - NxpHal:     write error status = 0x1ff
0:686:381 - NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
0:688:074 - NxpHal:     write error status = 0x1ff
0:688:315 - NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
0:690:358 - NxpNciX:     len =   4 > 20000100
0:690:573 - NxpHal:     write successful status = 0x0
0:690:735 - NxpHal:     Response timer started
0:690:826 - NxpHal:     Waiting after ext cmd sent
0:693:444 - NxpNciR:     len =   6 > 400003001100
0:693:668 - NxpHal:     read successful status = 0x0
0:693:826 - NxpHal:     Response timer stopped
0:693:915 - NxpHal:     Checking response
0:694:913 - NxpNciX:     len =   3 > 200100
0:695:125 - NxpHal:     write successful status = 0x0
0:695:383 - NxpHal:     Response timer started
0:695:471 - NxpHal:     Waiting after ext cmd sent
0:698:901 - NxpNciR:     len =  28 > 40011900031E030008000102038081828302D002FF020004881001A0
0:699:137 - NxpHal:     read successful status = 0x0
0:699:240 - NxpHal:     NxpNci> FW Version: 10.1.a0
0:699:378 - NxpHal:     Response timer stopped
0:699:464 - NxpHal:     Checking response
0:700:232 - NxpNciX:     len =   3 > 2F0200
0:700:473 - NxpHal:     write successful status = 0x0
0:700:632 - NxpHal:     Response timer started
0:700:719 - NxpHal:     Waiting after ext cmd sent
0:702:335 - NxpNciR:     len =   8 > 4F02050000009F97
0:702:630 - NxpHal:     read successful status = 0x0
0:702:801 - NxpHal:     Response timer stopped
0:702:889 - NxpHal:     Checking response
0:704:557 - NxpNciX:     len =  10 > 20030703A002A003A004
0:704:871 - NxpHal:     write successful status = 0x0
0:705:176 - NxpHal:     Response timer started
0:705:271 - NxpHal:     Waiting after ext cmd sent
0:707:748 - NxpNciR:     len =  17 > 40030E0003A0020101A0030108A0040101
0:707:980 - NxpHal:     read successful status = 0x0
0:708:082 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
0:708:223 - NxpHal:     Response timer stopped
0:708:310 - NxpHal:     Checking response
0:709:619 - NxpNciX:     len =   8 > 20020501A0030108
0:710:137 - NxpHal:     write successful status = 0x0
0:710:311 - NxpHal:     Response timer started
0:710:398 - NxpHal:     Waiting after ext cmd sent
0:711:793 - NxpNciR:     len =   5 > 4002020000
0:712:013 - NxpHal:     read successful status = 0x0
0:712:114 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
0:712:252 - NxpHal:     Response timer stopped
0:712:338 - NxpHal:     Checking response
0:712:442 - NxpHal:     phNxpNciHal_check_factory_reset file not present = /data/nfc/nfaStorage.bin1
0:715:345 - NxpNciX:     len =  26 > 20021702A0EA08FFFFFFFFFFFFFFFFA0EB08FFFFFFFFFFFFFFFF
0:715:580 - NxpHal:     write successful status = 0x0
0:715:740 - NxpHal:     Response timer started
0:715:832 - NxpHal:     Waiting after ext cmd sent
0:717:407 - NxpNciR:     len =   5 > 4002020000
0:717:629 - NxpHal:     read successful status = 0x0
0:717:729 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
0:717:864 - NxpHal:     Response timer stopped
0:717:950 - NxpHal:     Checking response
0:719:257 - NxpNciX:     len =   8 > 20020501A0440100
0:719:594 - NxpHal:     write successful status = 0x0
0:719:783 - NxpHal:     Response timer started
0:719:871 - NxpHal:     Waiting after ext cmd sent
0:721:543 - NxpNciR:     len =   5 > 4002020000
0:721:773 - NxpHal:     read successful status = 0x0
0:721:876 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
0:722:012 - NxpHal:     Response timer stopped
0:722:101 - NxpHal:     Checking response
[... 89 lines cut ...]
0:765:228 - NxpNciX:     len =   4 > 2F150100
0:765:446 - NxpHal:     write successful status = 0x0
0:765:746 - NxpHal:     Response timer started
0:765:836 - NxpHal:     Waiting after ext cmd sent
0:767:584 - NxpNciR:     len =   4 > 4F150100
0:767:819 - NxpHal:     read successful status = 0x0
0:767:981 - NxpHal:     Response timer stopped
0:768:074 - NxpHal:     Checking response
0:768:148 - NxpHal:     Sending discovery as raw packet!!
0:771:225 - NxpNciX:     len =  28 > 2103190C000101010201030105018001810182018301850106017001
0:771:460 - NxpHal:     write successful status = 0x0
0:771:624 - NxpHal:     Response timer started
0:771:712 - NxpHal:     Waiting after ext cmd sent
0:773:872 - NxpNciR:     len =   4 > 41030100
0:774:091 - NxpHal:     read successful status = 0x0
0:774:253 - NxpHal:     Response timer stopped
0:774:339 - NxpHal:     Checking response
0:774:415 - NxpHal:     Sending last command for Recovery 
0:774:489 - NxpHal:     Invoking data callback!!
0:821:768 - NxpNciR:     len =   9 > 600006A00000000000
0:821:991 - NxpHal:     read successful status = 0x0
0:822:093 - NxpHal:     CORE_RESET_NTF received!
0:822:163 - NxpHal:     phNxpNciHal_emergency_recovery: abort()
1:022:534 - NxpHal:     PN54X Reset - SUCCESS
```

tomicedani, 2018-01-29:
To clarify, the the app has no response when presented with a card still when removing that section.

From looking at the antenna on a scope, there to be no polls happening.

jgeslin, 2018-01-30:
Indeed, the discovery loop cannot be started when RF settings are corrupted (which is what is causing the initial SET_CONFIG_RSP error ("4002240911A00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00DA00D").
We discovered this infrequent possibility with previous libnfc-nxp.conf file which was causing EEPROM memory corruption. Unfortunately this memory corruption cannot be recovered, whatever the libnfc-nxp config file used afterwards.
I would be really interested to know the exact condition in which I could replicate your issue.

jgeslin, 2018-01-30:
Indeed, the discovery loop cannot be started when RF settings are corrupted (as indicated by the SET_CONFIG_RSP error in your previous log).
Weird thing is that we discovered really uncommon RF settings corruption when running previous version of libnfc-nxp.conf file. This EEPROM memory corruption cannot be recovered, even using new libnfc-nxp.conf file.
I would be interesting in replicating the scenario leading to this situation in your side. 
Can you please contact me [here](jeremy.geslin@nxp.com) ?

tomicedani, 2018-01-30:
Hey jgeslin, the contact link does not work for me.

closed 2018-11-07 by jgeslin