NXPNFCLinux/linux_libnfc-nci #72: Time for initialization

prattom, 2018-02-16:
I am using PN7150 and currently it takes about 1 second to initialise it to reach the "waiting for tag" state. Is there any way to minimise this time period?

jgeslin, 2018-02-21:
Typically it should be in the range of 100ms.
Could you please provides logs of the startup (enabled setting NXPLOG_xxx_LOGLEVEL flags to 0x03 within /etc/libnfc-nxp.conf file) using the latest version of the library (at least after commit https://github.com/NXPNFCLinux/linux_libnfc-nci/commit/320ffa40ec7458046c56d161ba253b9b2201768d) ?

prattom, 2018-02-22:
Here is my log
1519305080:131:626 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

1519305080:132:073 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

1519305080:132:225 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

1519305080:132:362 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

0:000:372 - NxpHal:     Entering phNxpNciHal_init_monitor
0:000:601 - NxpHal:     Returning with SUCCESS
0:000:780 - NxpTml:     phTmlNfc_i2c_open_and_configure

0:000:919 - NxpTml:     Opening port=/dev/pn544

0:001:110 - NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
0:211:345 - NxpTml:     phTmlNfc_i2c_reset(), VEN level 0
0:421:337 - NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
0:531:871 - NxpTml:     PN54X - Tml Reader Thread Started................

0:532:144 - NxpTml:     PN54X - Read requested.....

0:532:278 - NxpTml:     PN54X - Invoking I2C Read.....

0:532:499 - NxpTml:     PN54X - Tml Writer Thread Started................

0:532:693 - NxpTml:     PN54X - Tml Writer Thread Running................

0:532:821 - NxpTml:     PN54X - Write requested.....

0:532:946 - NxpTml:     PN54X - Invoking I2C Write.....

0:533:205 - NxpHal:     thread started
0:535:129 - NxpNciX:     len =   4 > 20000101
0:535:315 - NxpTml:     PN54X - I2C Write successful.....

0:535:490 - NxpTml:     PN54X - Posting Fresh Write message.....

0:535:816 - NxpTml:     PN54X - Tml Writer Thread Running................

0:536:063 - NxpHal:     write successful status = 0x0
0:536:322 - NxpHal:     Response timer started
0:536:484 - NxpHal:     Waiting after ext cmd sent
0:563:252 - NxpTml:     PN54X - I2C Read successful.....

0:563:493 - NxpNciR:     len =   6 > 400003001101
0:563:637 - NxpTml:     PN54X - Posting read message.....

0:564:000 - NxpHal:     read successful status = 0x0
0:564:285 - NxpHal:     Response timer stopped
0:564:452 - NxpHal:     Checking response
0:564:638 - NxpTml:     PN54X - Read requested.....

0:564:786 - NxpTml:     PN54X - Invoking I2C Read.....

0:564:971 - NxpTml:     PN54X - Write requested.....

0:565:120 - NxpTml:     PN54X - Invoking I2C Write.....

0:566:926 - NxpNciX:     len =   3 > 200100
0:567:097 - NxpTml:     PN54X - I2C Write successful.....

0:567:227 - NxpTml:     PN54X - Posting Fresh Write message.....

0:567:381 - NxpTml:     PN54X - Tml Writer Thread Running................

0:567:637 - NxpHal:     write successful status = 0x0
0:567:890 - NxpHal:     Response timer started
0:568:064 - NxpHal:     Waiting after ext cmd sent
0:573:901 - NxpTml:     PN54X - I2C Read successful.....
[... 806 lines cut ...]
0:535:236 - NxpHal:     > Polling Loop Started
0:535:409 - NxpTml:     PN54X - Read requested.....

0:535:554 - NxpTml:     PN54X - Invoking I2C Read.....

0:535:727 - NxpTml:     PN54X - Write requested.....

0:535:875 - NxpTml:     PN54X - Invoking I2C Write.....

0:538:115 - NxpNciX:     len =   6 > 210303010001
0:538:299 - NxpTml:     PN54X - I2C Write successful.....

0:538:423 - NxpTml:     PN54X - Posting Fresh Write message.....

0:538:719 - NxpTml:     PN54X - Tml Writer Thread Running................

0:538:895 - NxpHal:     write successful status = 0x0
0:542:167 - NxpTml:     PN54X - I2C Read successful.....

0:542:376 - NxpNciR:     len =   4 > 41030100
0:542:513 - NxpTml:     PN54X - Posting read message.....

0:542:688 - NxpHal:     read successful status = 0x0
0:543:012 - NxpTml:     PN54X - Read requested.....
I am using latest version of library. I want only NFC technology Type A reader mode for passive tags.

jgeslin, 2018-02-23:
Attached patch allows to save time during the initialization.
[patch_reset-time.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1751011/patch_reset-time.txt)

You can also set NFA_MAX_EE_SUPPORTED to 0x00 in /etc/libnfc-brcm.conf to avoid useless initialization sequence.

Additionally you can update the pn5xx_i2c kernel driver, reducing the msleep(100) calls by msleep(5) which is enough to insure proper communication.

prattom, 2018-02-24:
Thanks for reply. Your suggestion has reduced initialization time to about 0.35 seconds. Also tag reading has also improved a lot.

closed 2018-02-24 by prattom