NXPNFCLinux/linux_libnfc-nci #73: regarding initialization step

prattom, 2018-03-09:
There are 2 steps in InitMode:
1) nfcManager_doInitialize();
2) enableDiscovery
I want to know if RF is turned on in enableDiscovery mode or it turns on in nfcManager_doInitialize step itself. Is it possible to control RF turning on/off.
In my step I want to do the following:
1) Initialize NFC. RF is off
2) When tag comes (detected by proximity sensor) then turn RF on and detect tag.

jgeslin, 2018-03-09:
Hi,

RF is off until nfcManager_enableDiscovery is called. 
So you should be able to implement your scenario, turning on RF only when tag comes proximate.

Regards,
Jeremy.

prattom, 2018-03-09:
Thanks for clearing doubt Jeremy.
Regards
Pratyush

closed 2018-03-09 by prattom