NXPNFCLinux/linux_libnfc-nci #74: Missing /dev/pn544 - > _i2c_open() Failed : retval ffffffff

wirso, 2018-03-15:
Hello. 
I'm trying to use this library to read some NFCs but 
I have a problem that's quite like these ones : 
https://github.com/NXPNFCLinux/linux_libnfc-nci/issues/53
and
https://github.com/NXPNFCLinux/linux_libnfc-nci/issues/63

This is the output for "sudo i2cdetect -y 1 "
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- 28 -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
So there shouldn't be a problem with my RPI reading/detecting the Module.
---------------------------------------------------------------------------------------------------------
This is the output for "sudo lsmod | grep i2c"
i2c_dev               6386  0
i2c_bcm2708      5988  0
-------------------------------------------------------------------------------------------------------
I tried everything that i found on the links Mentioned earlier (--enable-alt or --enable-i2c etc)
The thing i didn't try is building a Device Tree (.dts) because I'm not that good with Low Level Access but 
now i would be ready to try yet I'm not sure if I should ( To put it simply, I don't know how to create/compile or w/e a Device Tree file.)

This is the output for "nfcDemoApp poll" : 
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpTml:     phTmlNfc_i2c_open_and_configure

NxpTml:     Opening port=/dev/pn544

NxpTml:     _i2c_open() Failed: retval ffffffff
NxpHal:     phTmlNfc_Init Failed
NxpHal:     Failed to deallocate (list empty)
NxpHal:     Node dump:
NxpHal:     Failed to deallocate (list empty)
NxpHal:     Node dump:
NxpExtns:   find found NXPLOG_NCIHAL_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_EXTNS_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NfcService Init Failed
Leaving ...
__________________________________________________________________________________________________________
I checked and found that /dev/pn544 directory is missing. 
I guess that this is my problem and I can't comprehend how to solve it.
__________________________________________________________________________________________________________
To sum it up a little : 
following the steps from 
https://www.nxp.com/docs/en/application-note/AN11697.pdf
I cloned, and did every step on chapter 4 (following the examples of ./configure from Issue 53 and 63)
Did the steps from 
http://ozzmaker.com/i2c/
Tried to add through "sudo adduser pi i2c" the Pi user (Actually i don't know why i did this but I thought maybe it will solve my problem)
Set my GPIO Pin (for me it was GPIO24) to 1/High since i found that the Module while in Standby won't read a signal (unless you send 2 signals in >1sec or put the GPIO pin to 1/High.
Yet nothing helped.
---------------------------------------------------------------------------------------------------------
My Devices : 
Raspberry Pi B+ (2014 i think)
OM5578 Raspberry Kit with PN7150 Chip.
---------------------------------------------------------------------------------------------------------
I'm sorry if my post is not that well written.
Hope to hear from you soon.
Thanks in advance.
-Stanislav

wirso, 2018-03-15:
Thank you for your Fast Reply ! 
I noticed that when i try to change something in "./configure" and after "make" or "make install" i get some errors (Since I was pretty confused and lost, I reinstalled the library).
------------------------------------------------------------------------------------
Make error -> 
Makefile:959: recipe for target 'all' failed
make: *** [all] Error 2
-------------------------------------------------------------------------------------
Make Install Error ->
Makefile:2266: recipe for targe 'src/halimpl/pn54x/tml/i2c/phTmlNfc_i2c.lo' failed
make: *** [src/halimpl/pn54x/tml/i2c/phTmlNfc_i2c.lo] Error 1.
-------------------------------------------------------------------------------------------
My "sudo nfcDemoApp poll" hasn't changed from the earlier version. 
Regards.
-S

jgeslin, 2018-03-15:
Please provide the application logs while running this configuration. That the only way to figure out what's going on.

Regards,
Jeremy.

wirso, 2018-03-16:
Hello.
A friend of mine tried something with the phTmlNfc_i2c.h, he commented a define (something like "#define *") and after that the Make & Make Install hadn't given an error with phTmlNfc_i2c.lo 
There was no file  named "phTmlNfc_i2c.lo" and now there is.
But my output from nfcDemoApp poll now is quite different.
Output for "sudo nfcDemoApp poll"
-----------------------------------------------------------------------
#########################################################################################
NFC demo

#########################################################################################
Poll mode activated

#########################################################################################
... press enter to quit ...

NxpExtns: find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns: find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns: find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns: find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpHal: Entering phNxpNciHal_init_monitor
NxpHal: Returning with SUCCESS
NxpTml: phTmlNfc_i2c_open_and_configure Alternative NFC

NxpTml: NFC / Assign IO pins

NxpTml: Pin /sys/class/gpio/gpio23

NxpTml: System already has pin /sys/class/gpio/gpio23

NxpTml: Direction /sys/class/gpio/gpio23/direction

NxpTml: Pin 23 now an input

NxpTml:  Edge /sys/class/gpio/gpio23/edge

NxpTml: Edge-string rising - 6

NxpTml: Pin 23 trigger on rising

NxpTml: Value /sys/class/gpio/gpio23/value

NxpTml: Pin /sys/class/gpio/gpio24

NxpTml: System already has pin /sys/class/gpio/gpio24

NxpTml: Direction /sys/class/gpio/gpio24/direction

NxpTml: Pin 24 now an output

NxpTml: Pin 24 now off

NxpTml: NFCHW - open I2C bus - /dev/i2c-1

NxpTml: Could not open I2C bus '/dev/i2c-1' (No such file or directory)
NxpHal: phTmlNfc_Init Failed
NxpHal: Failed to deallocate (list empty)
NxpHal: Node dump:
NxpHal: Failed to deallocate (list empty)
NxpHal: Node dump:
NxpExtns: find found NXPLOG_NCIHAL_LOGLEVEL=(0x3)

NxpExtns: find found NXPLOG_EXTNS_LOGLEVEL=(0x3)

NxpExtns: find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns: find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns: find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns: find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NfcService Init Failed
Leaving ...
-------------------------------------------------------------------------

Hope to hear from you soon.
I will try to find the solution if I can, will update if anything changes.
PS: You were right indeed, when i ran "nfcDemoApp poll" there were some errors telling me that the app didn't have permission to enter "xxxx" file while running "sudo nfcDemoApp" didn't give me any of such.
Regards,
-S.

jgeslin, 2018-03-16:
"NxpTml: Could not open I2C bus '/dev/i2c-1' "

Can you confirm /dev/i2c-1 is present and with proper access rights ?
$ ls -als /dev/

Regards,
Jeremy.

wirso, 2018-03-16:
Thanks for your FAST Reply. 
I checked and I got no directory named "i2c-1"
-S.

jgeslin, 2018-03-16:
You must enable it via raspi-config

Regards,
Jeremy.

wirso, 2018-03-16:
Well, this is quite interesting. 
I did enable it and now I do have the i2c-1 dir. 
The problem is that while running "sudo nfcDemoApp" the first 50-100 lines or so, are normal and after that I guess that the RPi confuses itself because it shows some symbols instead of chars. 
Not only while running nfcDemoApp but even after that. 
I have to reboot to turn the symbols into chars.

wirso, 2018-03-16:
Tried only "nfcDemoApp poll" 
I found out that it just repeats the following lines : 
_______________________________________________________
NxpTml: Error with interrupt-detect pin (0)
NxpTml: _i2c_rea() [hdr] errno : 5
NxpTml PN54X - Error in I2C Read.....

NxpTml: PN54X - Read requested ....

NxpTml:  PN54X - Invoking I2C Read ...

NxpTml:  Error with interrupt-detect pin (0)
______________________________________________________
These following lines just continue repeating until i cancel the app.
-S

wirso, 2018-03-16:
Found out that the RPi now doesn't read my module.
Output for "sudo i2cdetect -y 1 "
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
-S

wirso, 2018-03-16:
Tried again and now it gives me a little different error 
Output for "nfcDemoApp poll"
_________________________________________________
NxpTml: Error with interrupt-detect pin (0)
NxpTml: _i2c_read() [hdr] errno : 5
NxpTml PN54X - Error in I2C Read.....

NxpTml: PN54X - Read requested ....

NxpTml:  PN54X - Invoking I2C Read ...

NxpTml:  Error with interrupt-detect pin (0)
NxpTml:  _i2c_write() errno : 5
NxpTml:  PN54X - Error in I2C Write...

NxpTml:  PN54X - Posting Fresh Write message...

NxpTml:  PN54X - Tml Writer Thread Running.................

NxpHal: write error status = 0x1ff
NxpTml:  _i2c_read() [hdr] errno : 5
NxpTml:  PN54X - Error in I2C Read....

NxpTml:  PN54X - Read requested ....
____________________________________________________-
I'm really sorry for disturbing you, I'm really confused and continued Updating my latter comments because I wasn't that concentrated.
Hope to hear from you soon.
-S

wirso, 2018-03-16:
Actually I deactivated the debug trace by setting NXPLOG_xxx_LOGLEVEL from 0x03 to 0x00 and now I don't have any problems. It works fine.
Thank you for your kind support. 
Good luck and have a nice day.
-Stanislav.

closed 2018-03-16 by wirso