NXPNFCLinux/linux_libnfc-nci #75: NDEF Write Tag Failed.

wirso, 2018-03-21:
Hello Again. 
By following #74  and #52 I successfully read and wrote RAW/Block data on my Type V Tag, yet I can't manage to read/write NDEF format.
PS : I don't have any problems with reading and writing NDEF format on any other tags (Mifare UL etc)
This is the output for "nfcDemoApp poll" : 
[temp.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1824315/temp.txt)
This is the output for "nfcDemoApp write --type=Text -l en -r "Test" "
[temp.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/1833752/temp.txt)
Kind Regards,
-S.

jgeslin, 2018-03-21:
From what I can see from the log , the tag seems to not comply to expectation.
Indeed, it doesn't provides VICC memory size in answer to the "Get system information" command as defined per ISO15693-3 spec.
This is preventing the library to further look for NDEF message (when application is started in "poll" mode) and preventing NDEF writing (when application is started in "write" mode), also the library still allow raw operation with the tag.

You could eventually bypass this limitation by disabling this check (commenting out lines 1587 to 1592 of src/libnfc-nci/nfc/tags/rw_i93.c file) but I am not 100% confident in the further operations (you may have to fake the "block_size" to avoid Null operations).

wirso, 2018-03-30:
Sorry for the late reply.
Thanks for your information and hard work for guiding me through this.
Hope you'll have a good day.
Thanks again, 
-Stanislav

closed 2018-03-30 by wirso

lucar91, 2021-01-20:
Hi,

I am using 2 PN7150 readers to write on tags. I have two situations:
Reader 1: write tag correctly.
Reader 2: error formatting ndef.

This is the code that I use when I want to write in tag:

    int tag_is_written=false;//bool, false if tag is not written
    res  = ndef_createMime("audio/ogg",tag_content,NFC_TAG_DATA_MAX_CONTENT_SIZE,NDEF_msg,MAX_NFC_LEN);
    if(res <= 0x00){
        //!TODO creare notify_that_volume_is_to_modified per comunicare la non riuscita della scrittura
        log_this(NFC_ACTIONS_LOG_NAME," ! ! ! Failed to build TEXT NDEF Message! ! !",0);
        tag_is_written = false;
    }else{
        NDEF_msg_len = res;
        if( nfcTag_formatTag( g_tagInfos.handle) != 0x00){
            dbg_fprintf(2,stdout,"\e[39m\e[41m[%d] TAG NOT FORMATTED  \e[0m\n");
            tag_is_written=false;
        }else{
            if( nfcTag_writeNdef( g_tagInfos.handle, NDEF_msg, NDEF_msg_len)  != 0x00){
                //!TODO creare notify per comunicare la non riuscita della scrittura con un general issue?
                dbg_fprintf(2,stdout,"\e[39m\e[41m[%d] TAG NOT WRITTEN  \e[0m\n");
                tag_is_written = false;
            }else{
                dbg_fprintf(3,stdout,"Tag written successfully.\n");
                tag_is_written=true;//bool, false if tag is not written
            }
        }
    }

and I have the "TAG NOT FORMATTED" error.
Can you tell if it's a hardware problem via the log files? Maybe reader 2 is damaged?

I have modified the configuration files for my applicatoion:

[libnfc-nci.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5841416/libnfc-nci.txt)
[libnfc-nxp-init.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5841420/libnfc-nxp-init.txt)
[libnfc-nxp-pn547.conf.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5841421/libnfc-nxp-pn547.conf.txt)
[libnfc-nxp-pn548.conf.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5841422/libnfc-nxp-pn548.conf.txt)

Log files:

[log_NXPLOG_DEBUG_LOGLEVEL_enabled.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5841380/log_NXPLOG_DEBUG_LOGLEVEL_enabled.txt)
[log_NXPLOG_ERROR_LOGLEVEL_enabled.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5841381/log_NXPLOG_ERROR_LOGLEVEL_enabled.txt)

Thanks a lot in advance :)

jgeslin, 2021-01-20:
Hi, 

The provided log files are not enough to analyze the behavior.
Please enable full logs (by setting all NXPLOG_XXX_LOGLEVEL flags to 0x03) and provide trace for working reader and non-working reader for comparison.

Regards,
Jeremy.

lucar91, 2021-01-20:
[log_ALL_reader1.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5842335/log_ALL_reader1.txt)
[log_ALL_reader2.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5842336/log_ALL_reader2.txt)

Thanks

jgeslin, 2021-01-21:
Trace of the failing reader shows a NAck response by the card on the MIFARE write command to page 33, while all previous MIFARE write commands (from pages 3 to page 32 succeed).
With the working reader, the MIFARE write command to page 33 is succeed.

Can you check the failing reader is always failing at the same step of MIFARE commands sequence ?

MIFARE command to page 33:
`NxpNciX:     len =   9 > 000006A221D3A91634`
is answered with NAck ('05' while Ack would be '0A'):
`NxpNciR:     len =   5 > 0000020500`

Regards,
Jeremy.

lucar91, 2021-01-22:
I don't understand. 
`NxpTml:     PN54X - I2C Read successful.....`
`NxpNciR:     len =   5 > 0000020500`
`NxpTml:     PN54X - Posting read message.....`

 5 > 0000020500 is for read Ack in this case? Correctly?

I sent you new logs:
Working reader
 
[log_working_reader.log](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5854546/log_working_reader.log)

Not working reader, 3 different writings

[log_reader_not_working.log](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5854547/log_reader_not_working.log)
[log_reader_not_working2.log](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5854548/log_reader_not_working2.log)
[log_reader_not_working3.log](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/5854549/log_reader_not_working3.log)

In all 3 writings with not working reader, I have the following error (error not present with the reader working):
`NxpHal:     write error status = 0x1ff`
`NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry`
`NxpTml:     PN54X - Write requested....`

can this have to do with my problem?

Thanks

jgeslin, 2021-01-22:
"0000020500" is a NAck answer to a write command from the card, an Ack is "0000020A00". Answer to a read command is including the bytes read like "0000110300FE0080617564696F2F6F6767116F00".

From the 3 traces, I can see the issue occurs at different steps of the formatting operation, with different error reasons. It most probably indicates issue in the RF performances of the reader which doesn't allow stable communication with the card.
You should verify the tuning of this reader, especially comparing with the working one.

Regards,
Jeremy.

lucar91, 2021-01-22:
I will try as soon as possible. Thanks a lot in the meantime :)