NXPNFCLinux/linux_libnfc-nci #76: Could benefit from more installation instruction

benrochon, 2018-04-03:
Hey, could it be possible to add a little bit more detail on the install procedure? 
Thanks.

jgeslin, 2018-04-03:
Did you have a look to [AN11697](https://www.nxp.com/docs/en/application-note/AN11697.pdf) listed as documentation ?

benrochon, 2018-04-03:
I did but I get `NfcService Init Failed` when I start `./nfcDemoApp poll`.

```
./bootstrap
./configure --enable-pn7150
make
make install
```

jgeslin, 2018-04-04:
Please provide logs of the issue by enabling it through /etc/libnfc-nxp.conf file (setting NXPLOG_xxx_LOGLEVEL flags to 0x03).

benrochon, 2018-04-04:
Looks like that:

```
1522860707:936:462 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

1522860707:936:772 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

1522860707:936:863 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

1522860707:936:941 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

0:000:479 - NxpHal:     Entering phNxpNciHal_init_monitor
0:000:619 - NxpHal:     Returning with SUCCESS
0:000:764 - NxpTml:     phTmlNfc_i2c_open_and_configure

0:000:847 - NxpTml:     Opening port=/dev/pn544

0:000:998 - NxpTml:     _i2c_open() Failed: retval ffffffff
0:001:129 - NxpHal:     phTmlNfc_Init Failed
0:001:270 - NxpHal:     Failed to deallocate (list empty)
0:001:361 - NxpHal:     Node dump:
0:001:441 - NxpHal:     Failed to deallocate (list empty)
0:001:513 - NxpHal:     Node dump:
0:008:397 - NxpExtns:   find found NXPLOG_NCIHAL_LOGLEVEL=(0x3)

0:013:841 - NxpExtns:   find found NXPLOG_EXTNS_LOGLEVEL=(0x3)

0:015:402 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

0:016:788 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

0:017:107 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

0:017:410 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NfcService Init Failed

```

jgeslin, 2018-04-05:
Can you confirm you have the pn5xx_i2c kernel driver installed (should pop-up as /dev/pn544 device node)?
FYI, installation of pn5xx_i2c kernel module is described in AN11697.

However it exists a altenative solution to the use of this kernel driver. It uses i2c-dev driver and gpio class of the platform.

What is your HW platform ? How are you connecting the OM5578 demo kit ?

benrochon, 2018-04-05:
Oh I did not installed the driver. I thought _linux_libnfc-nci_ handled it. 
The instructions refers to drivers/misc folder which would be in the kernel source. I'd like to avoid this option if possible. :)
I have a Raspberry Pi 2 B with the OM5578 interface, It's currently running the latest version of raspbian.

Thanks,

jgeslin, 2018-04-05:
If you are using RPi I would recommend to use the "Alternative to kernel driver" solution.
To do that, simply use "--enable-alt" option at configure step:
```
./bootstrap
./configure --enable-pn7150 --enable-alt --sysconfdir=/etc
make
make install
```

Make sure that i2c-dev is enabled in your kernel (this can be done via raspi-config)

benrochon, 2018-04-05:
Nice, it works! 

Thanks for your help!

benrochon, 2018-04-05:
I think it could be helpful to have these RPi-specific instructions written somewhere in the doc (or in the readme file?)

closed 2018-04-06 by benrochon