NXPNFCLinux/linux_libnfc-nci #77: NFCDemoApp not working with i.MX6 Board

md-jamal, 2018-04-06:
Hi Guys,

I have a customized i.MX6 Board , I first tested PN7120 on Raspberry Pi and it worked out of the box.

When I switched to my i.MX6 Board, it is not working.

Here are my changes:

1. Device Tree:

pn547: pn547@28 {
                compatible = "nxp,pn547";
                reg=<0x28>;
                clock-frequency = <400000>;
                interrupt-gpios = <&gpio3 19 0>;
                enable-gpios = <&gpio2 30 0>;

        };
2. Added pn5xx_i2c driver as a Module and when loading the module here are the logs and it created /dev/pn544

pn54x_dev_init
pn54x_probe
pn544 1-0028: FIRM GPIO <OPTIONAL> error getting from OF node
pn544 1-0028: CLKREQ GPIO <OPTIONAL> error getting from OF node
pn54x_get_pdata: could not get nxp,pn54x-pvdd, rc=-19
pn54x_get_pdata: could not get nxp,pn54x-vbat, rc=-19
pn54x_get_pdata: could not get nxp,pn54x-pmuvcc, rc=-19
pn54x_get_pdata: could not get nxp,pn54x-sevdd, rc=-19
pn54x_probe: request irq_gpio 83
pn54x_probe: request ven_gpio 62
pn54x_probe : requesting IRQ 243

3. When I executed ./nfcDemoApp poll, it failed with the following logs.

#########################################################################################
##    pn54x_dev_open : 10,58
                                 pn54x_dev_ioctl, cmd=1074063617, arg=1
  NFC demo                       pn544_enable power on
               ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

1523001242:920:847 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

1523001242:920:946 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

1523001242:920:986 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

1523001242:921:024 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

0:000:211 - NxpHal:     Entering phNxpNciHal_init_monitor
0:000:268 - NxpHal:     Returning with SUCCESS
0:000:332 - NxpTml:     phTmlNfc_i2c_open_and_configure

0:000:382 - NxpTml:     Opening port=/dev/pn544

0:002:874 - NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
0:217:033 - pn54x_dev_ioctl, cmd=1074063617, arg=0
NxpTml:     phTmlNfc_i2c_reset(),pn544_disable power off
 VEN level 0
0:427:036 - pn54x_dev_ioctl, cmd=1074063617, arg=1
NxpTml:     phTmlNfc_i2c_reset(),pn544_enable power on
 VEN level 1
0:537:093 - pn54x_dev_write : i2c_master_send returned -5
NxpTml:     PN54X - Tml Reader Thpn54x_dev_write : i2c_master_send returned -5
read Started................

0pn54x_dev_write : i2c_master_send returned -5
[... 224 lines cut ...]
0:916:664 - NxpTml:     PN54X - Invoking I2C Read.....

0:923:346 - NxpTml:     _i2c_read() [hdr] errno : 5
0:9pn54x_dev_ioctl, cmd=1074063617, arg=1
23:396 - NxpTml:     PN54X - Erropn544_enable power on
r in I2C Read.....

0:923:423 - NxpTml:     PN54X - Read requested.....

0:923:448 - NxpTml:     PN54X - Invoking I2C Read.....

0:930:143 - NxpTml:     _i2c_read() [hdr] errno : 5
0:930:182 - NxpTml:     PN54X - Error in I2C Read.....

1:017:016 - NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
1:263:552 - NxpTml:     PN54X - Read requested.....

1:274:747 - NxpTml:     PN54X - Invoking I2C Read.....

1:376:993 - NxpHal:     PN54X Reset - SUCCESS

1:377:091 - NxpHal:     phNxpNciHal_write failed for hal ext
1:377:122 - NxpHal:     Force FW Download, NFCC not coming out from Standby

I have seen a similar issue in github, what is the next step for me to resolve this issue.

md-jamal, 2018-04-06:
I also tried switching to Alt Mode and after enabling logs it is printing these in loop

13:746:530 - NxpTml:     PN54X - Error in I2C Read.....

13:746:557 - NxpTml:     PN54X - Read requested.....

13:746:593 - NxpTml:     PN54X - Invoking I2C Read.....

13:746:625 - NxpTml:     Error with interrupt-detect pin (0)
13:746:905 - NxpTml:     _i2c_read() [hdr] errno : 5
13:746:939 - NxpTml:     PN54X - Error in I2C Read.....

jgeslin, 2018-04-06:
The first thing to check is the level of PN7150 VEN signal (HW reset function). It should be driven high to allow proper PN7150 function. If not going high during linux libnfc-nci library initialization (when starting the demo application) it probably mean the DTS definition does not match the physical connection.

While using the Alternative to kernel driver solution (alt mode), you must insure the gpios are not already used. This is probably the case if the pn5xx_i2c driver is loaded. So before lanuching the demo app (with libnfc-nci library configured in alt mode), unload the pn5xx_i2c  driver.

md-jamal, 2018-04-06:
What do you mean by driving high in Driver Mode.. Are there any cons of using alt mode..

jgeslin, 2018-04-06:
VEN must be driven high to make PN7150 active (a low pulse triggers PN7150 reset) independently of the mode used (kernel driver or alt). This is managed by the libnfc-nci stack through the /dev/pn544 (in case of pn5xx_i2c driver use) or through /sys/class/gpio (in case of alt mode).

md-jamal, 2018-04-06:
You mean in the device tree, we need to set the default value to 1.

jgeslin, 2018-04-06:
> You mean in the device tree, we need to set the default value to 1.

No, the libnfc-nci linux is usually managing the GPIO via the pn5xx_i2c driver.
What I meant is that you should check the signal level of VEN. If it is not high it point to an issue with the driver or rather about the DTS definition.

jgeslin, 2019-04-01:
Closed because considered answered and no comment since 1 year.

closed 2019-04-01 by jgeslin