NXPNFCLinux/linux_libnfc-nci #80: Sending multiple NDEF over P2P

orange-kao, 2018-05-15:
Hello.

I'm evaluating PN7150 and looking for possibilities to send file (about 1MB) over NFC between two PN7150 (by using OM5578/PN7150RPI development board). And it seems that it makes more sense to use P2P mode instead of HCE mode. So I've started to use "nfcDemoApp poll" and "nfcDemoApp push --type=mime -m xyz/123 -d 6162636465" to evaluate this possibility.

Due to the size of the file (1MB), I'm planning to split the file to multiple NDEF messages. Which would be around 1KB each (does that make sense?)

So I expect the software to keep the P2P connection until the data transfer (will be multiple NDEF) has finished. And it seems that the nfcDemoApp will cut the P2P connection after receiving the first NDEF.

Detail as below and thanks for the support.

## Reproduce steps
1. Get two pieces of Raspberry Pi (unit A and B), and install OM5578/PN7150RPI on it.
2. Put two pieces of OM5578/PN7150RPI facing each other (less than 1cm between two antenna)
3. Issue "nfcDemoApp poll" on unit A
4. Issue "nfcDemoApp push --type=mime -m xyz/123 -d 6162636465" on unit B

## Expected results
Unit A receives NDEF from unit B, print the NDEF, but keep the P2P connection.
Both units does NOT print "Device Lost" until nfcDemoApp been terminated or two antenna has been separated.

## Actual result
Unit A print "Device Lost" after receiving NDEF. It seems that unit A cut the connection after receiving NDEF.

However, "Device Lost" message was NOT printed on unit B. (why?)

Message from unit A
```
Waiting for a Tag/Device...

    Device Found

        NDEF Message Received : 

        TNF Media

            Type : xyz/123
            Data : abcde

        15 bytes of NDEF data received :
        D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

    Device Lost

Waiting for a Tag/Device...
```

Message from unit B

```
Waiting for a Tag/Device...

    Device Found
        Push successful
```

## Additional information
I can't explain why unit A print "Device Lost", but unit B doesn't.

Thus I tried to modify the nfcDemoApp by changing the WaitDeviceArrival() function, hoping to be able to send multiple messages before P2P connection has been cut. In WaitDeviceArrival(), I tried to call SnepPush() repeatedly (repeating 5 times using the same data in this test). Code snippet as below. Unit A is able to receive the message 5 times, and this is great. However, unit B indicates there are push failures.

```
else if(eDevType_P2P == g_Dev_Type)/*P2P Detected*/
{
    framework_UnlockMutex(g_devLock);
    printf("\tDevice Found\n");

    if(2 == mode)
    {
        SnepPush(msgToSend, len);
        SnepPush(msgToSend, len);
        SnepPush(msgToSend, len);
        SnepPush(msgToSend, len);
        SnepPush(msgToSend, len);
    }

    framework_LockMutex(g_SnepClientLock);

    if(eSnepClientState_READY == g_SnepClientState)
    {
        g_SnepClientState = eSnepClientState_WAIT_OFF;
        framework_WaitMutex(g_SnepClientLock, 0);
    }

    framework_UnlockMutex(g_SnepClientLock);
    framework_LockMutex(g_devLock);

}
```

Message from unit A
```
Waiting for a Tag/Device...

    Device Found

        NDEF Message Received : 

        TNF Media

            Type : xyz/123
            Data : abcde

        15 bytes of NDEF data received :
        D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

    Device Lost

Waiting for a Tag/Device...

    Device Found

        NDEF Message Received : 

        TNF Media

            Type : xyz/123
            Data : abcde

        15 bytes of NDEF data received :
        D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

    Device Lost

Waiting for a Tag/Device...

    Device Found

        NDEF Message Received : 

        TNF Media

            Type : xyz/123
            Data : abcde

        15 bytes of NDEF data received :
        D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

    Device Lost

Waiting for a Tag/Device...

    Device Found

        NDEF Message Received : 

        TNF Media

            Type : xyz/123
            Data : abcde

        15 bytes of NDEF data received :
        D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

    Device Lost

Waiting for a Tag/Device...

    Device Found

        NDEF Message Received : 

        TNF Media

            Type : xyz/123
            Data : abcde

        15 bytes of NDEF data received :
        D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

    Device Lost

Waiting for a Tag/Device...
```

Message from unit B

```
Waiting for a Tag/Device...

    Device Found
        Push successful
        Push Failed
        Push Failed
        Push Failed
        Push Failed
```

jgeslin, 2018-05-18:
Hi,

> Due to the size of the file (1MB), I'm planning to split the file to multiple NDEF messages. Which would be around 1KB each (does that make sense?)

The SNEP protocol defines frame size on 4 bytes, so 1M length could fit into one packet (which will anyway will be splitted when transmitted over RF / I2C)

> Unit A print "Device Lost" after receiving NDEF. It seems that unit A cut the connection after receiving NDEF.

This is linked to the SNEP service implementation in libnfc-nci library closing the SNEP connection after pushing NDEF message.
Attached patch fixes this limitation: [patch_SnepMultiplePut.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2017124/patch_SnepMultiplePut.txt)

Regards,
Jeremy.

jgeslin, 2018-08-13:
Latest release R2.4 now includes the fix.

orange-kao, 2019-01-11:
Hello.

I tried the latest commit 751f7bfa9f3f5b43e8a179d0a80dabd6bc1dac17 and this issue seems still there.
In this test, two devices have antenna facing each other and never separate physically.
I want to send multiple NDEF via P2P so I modify demoapp/main.c as below.

```
diff --git a/demoapp/main.c b/demoapp/main.c
index e98ada2..45c735c 100644
--- a/demoapp/main.c
+++ b/demoapp/main.c
@@ -1406,6 +1406,10 @@ int WaitDeviceArrival(int mode, unsigned char* msgToSend, unsigned int len)
                 if(2 == mode)
                 {
                     SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
                 }
                 
                 framework_LockMutex(g_SnepClientLock);
```

### Actual results

Device A prints

```
$ nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

	Device Lost

Waiting for a Tag/Device...
```

Device B prints
```
$ nfcDemoApp push --type=mime -m xyz/123 -d 6162636465
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Push mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found
		Push successful
[... 101 lines cut ...]
```

After these modification, sometimes I still experiences push failure after several (e.g., 8) push success and cannot recover without separating two peers. It prints the following message if debug flag `NXPLOG_GLOBAL_LOGLEVEL=0x03` has been set.

Please advice. Thank you.

```
NxpFunc:    nfaConnectionCallback: event= 14
NxpFunc:    nfaConnectionCallback: NFA_LLCP_DEACTIVATED_EVT
NxpFunc:    nfaConnectionCallback: event= 14
NxpFunc:    nfaConnectionCallback: NFA_LLCP_DEACTIVATED_EVT
NxpFunc:    nfaConnectionCallback: event= 14
NxpFunc:    nfaConnectionCallback: NFA_LLCP_DEACTIVATED_EVT
NxpFunc:    nfaConnectionCallback: event= 14
NxpFunc:    nfaConnectionCallback: NFA_LLCP_DEACTIVATED_EVT
NxpFunc:    nfaConnectionCallback: event= 14
NxpFunc:    nfaConnectionCallback: NFA_LLCP_DEACTIVATED_EVT
NxpFunc:    nfaConnectionCallback: event= 14
NxpFunc:    nfaConnectionCallback: NFA_LLCP_DEACTIVATED_EVT
NxpFunc:    nfaConnectionCallback: event= 14
NxpFunc:    nfaConnectionCallback: NFA_LLCP_DEACTIVATED_EVT
NxpFunc:    nfaConnectionCallback: event= 14
NxpFunc:    nfaConnectionCallback: NFA_LLCP_DEACTIVATED_EVT
NxpFunc:    nfaConnectionCallback: event= 14
```

jgeslin, 2019-01-11:
Hi,

On same commit https://github.com/NXPNFCLinux/linux_libnfc-nci/commit/751f7bfa9f3f5b43e8a179d0a80dabd6bc1dac17 I am unable to replicate the problem.
In my side I receive properly the messages:
```
Waiting for a Tag/Device...

	Device Found
		Push successful
		Push successful
		Push successful
		Push successful
		Push successful
```

And sometimes (depending of the position of the antennas), it fails after several exchanges or even at the first:
```
Waiting for a Tag/Device...

	Device Found
		Push successful
		Push successful
		Push Failed
	Device Lost
```

Could enable full logs from both side and share them for deeper analysis ?

orange-kao, 2019-01-14:
Hi Jeremy,

Hardware setup, build log and debug log as below.
Thank you.

### Hardware setup
OM5578 on Raspberry Pi, tied with rubber bands.
The hardware never moves during the test.

[image]

### Build log
- [01-compile-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2753748/01-compile-device-a.txt)
- [01-compile-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2753749/01-compile-device-b.txt)

Note: it's build with `openssldir=/usr ./configure --prefix=/usr --enable-alt --enable-llcp1_3`

### First run (~10 seconds)
Full log with 0xff debug flags:
- [02-first-run-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2753750/02-first-run-device-a.txt)
- [02-first-run-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2753751/02-first-run-device-b.txt)

The following section is a copy of `02-first-run-device-a.txt` but debug messages has been removed
```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...
Waiting for a Tag/Device...
	Device Found
		NDEF Message Received : 
		TNF Media
			Type : xyz/123
			Data : abcde
		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 
	Device Lost
Waiting for a Tag/Device...
	Device Found
		NDEF Message Received : 
		TNF Media
			Type : xyz/123
			Data : abcde
		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 
Waiting for a Tag/Device...
^C
```

The following section is a copy of `02-first-run-device-b.txt` but debug messages has been removed
```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Push mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...
Waiting for a Tag/Device...
	Device Found
		Push successful
		Push Failed
	Device Lost
	Device Found
		Push Failed
		Push Failed
		Push Failed
		Push Failed
^C
[... 114 lines cut ...]
```

The following section is a copy of `04-run-60sec-device-b.txt` but debug messages has been removed
```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Push mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...
Waiting for a Tag/Device...
	Device Found
		Push successful
	Device Lost
Waiting for a Tag/Device...
	Device Found
		Push Failed
		Push Failed
		Push Failed
		Push Failed
		Push Failed
	Device Lost
Waiting for a Tag/Device...
Leaving ...
```

jgeslin, 2019-01-14:
Can you confirm you don’t see the problem if not using LLCP1.3 ?

[om5578_tied_face_to_face]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F33737253%2F51092773-6e7fc280-17ef-11e9-9e36-f9ce7173a3b5.jpg&data=02%7C01%7Cjeremy.geslin%40nxp.com%7C0a63a41ff4f34435135e08d679c071d1%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C636830264779125240&sdata=Hom7Sv41bRd87dmFKcSjB%2Bba%2FUy0n9o8Lkbej%2FrVLn0%3D&reserved=0>

orange-kao, 2019-01-15:
The issue is still reproducible, even if LLCP was disabled at compile time.
It's build with `./configure --prefix=/usr --enable-alt`

I test it 5 times with 60 seconds each. Debug log as attached.
Thank you.

## Actual result
With development boards tied together with rubber bands and never move during the test, the result as below.
- Push failure on 2nd and later push on device B
- In the best case, there are only 3 messages has been received in a minute, according to the messages on device A.
- In the worst case, there is no messages has been received in a minute, according to the messages on device A.

## Build log
- [09-nollcp-compile-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757753/09-nollcp-compile-device-a.txt)
- [09-nollcp-compile-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757755/09-nollcp-compile-device-b.txt)

## Full log for running 60 seconds
These are full log with 0xff debug flags
- [10-nollcp-run-60sec-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757756/10-nollcp-run-60sec-device-a.txt)
- [10-nollcp-run-60sec-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757757/10-nollcp-run-60sec-device-b.txt)
- [11-nollcp-run-60sec-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757758/11-nollcp-run-60sec-device-a.txt)
- [11-nollcp-run-60sec-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757759/11-nollcp-run-60sec-device-b.txt)
- [12-nollcp-run-60sec-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757760/12-nollcp-run-60sec-device-a.txt)
- [12-nollcp-run-60sec-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757761/12-nollcp-run-60sec-device-b.txt)
- [13-nollcp-run-60sec-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757762/13-nollcp-run-60sec-device-a.txt)
- [13-nollcp-run-60sec-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757763/13-nollcp-run-60sec-device-b.txt)
- [14-nollcp-run-60sec-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757764/14-nollcp-run-60sec-device-a.txt)
- [14-nollcp-run-60sec-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2757765/14-nollcp-run-60sec-device-b.txt)

## Copy of full log, with debug message removed
This is a copy of above logs `1?-run-60sec-device-?.txt` but debug messages has been removed with [gist libnfcnci-issue80-filter-debug-message](https://gist.github.com/orange-kao/6ffc2b0ae8fc0bebc469fe3bb150a018) and cleaned up manually.

### 10-nollcp-run-60sec-device-a.txt
```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...
Waiting for a Tag/Device...
	Device Found
		NDEF Message Received : 
		TNF Media
			Type : xyz/123
			Data : abcde
		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 
	Device Lost
Waiting for a Tag/Device...
	Device Found
	Device Lost
Waiting for a Tag/Device...
Leaving ...
```

### 10-nollcp-run-60sec-device-b.txt
```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Push mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...
Waiting for a Tag/Device...
	Device Found
		Push successful
		Push Failed
	Device Lost
Waiting for a Tag/Device...
[... 207 lines cut ...]
	Device Found
		Push successful
		Push Failed
	Device Lost
Waiting for a Tag/Device...
	Device Found
		Push Failed
		Push Failed
	Device Lost
Waiting for a Tag/Device...
	Device Found
		Push Failed
		Push Failed
	Device Lost
Waiting for a Tag/Device...
	Device Found
		Push Failed
		Push Failed
		Push Failed
		Push Failed
		Push Failed
	Device Lost
Waiting for a Tag/Device...
Leaving ...
```

orange-kao, 2019-01-21:
Hi Jeremy,

Thank you for your help so far, you must be very busy.
Does there any help you need? Have you been able to reproduce the issue?
Thank you.

Best,
Orange

jgeslin, 2019-01-22:
Hi,

Indeed, I am unable to reproduce your issue in my side. I can anyway observe different behavior in the LLCP protocol handling, but I'm missing some traces from your side.

Please use the latest version (master branch) of the library and enable all logs (only one shot from both device should be sufficient to analyze):
- in libnfc-nxp-init.conf:
```
NXPLOG_GLOBAL_LOGLEVEL=0x03
NXPLOG_EXTNS_LOGLEVEL=0x03
NXPLOG_NCIHAL_LOGLEVEL=0x03
NXPLOG_NCIX_LOGLEVEL=0x03
NXPLOG_NCIR_LOGLEVEL=0x03
NXPLOG_FWDNLD_LOGLEVEL=0x03
NXPLOG_TML_LOGLEVEL=0x03
```
- in libnfc-nci.conf
```
APPL_TRACE_LEVEL=0xFF
PROTOCOL_TRACE_LEVEL=0xFF
```

To simplify the export of the traces in a log file, just run the application with such parameters:
`$ nfcDemoApp poll 2>log-device-a.txt`

Regards,
Jeremy.

orange-kao, 2019-01-22:
Hello,

The following test result is based on commit 751f7bfa9f3f5b43e8a179d0a80dabd6bc1dac17 and debug flags enabled.
Build with `./configure --prefix=/usr --enable-alt`
Running for 60 seconds.
- On device A, there is only two successful NDEF message received in 60 seconds.
- On device B, only one "Push successful" message.

Thank you.

## Debug log (stderr)
- [15-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2782724/15-device-a.txt)
- [15-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2782725/15-device-b.txt)

## Console output (stdout)
Device A
```
pi@orange-pi1:~/linux_libnfc-nci $ nfcDemoApp poll 2> 15-device-a.txt
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

	Device Lost

Waiting for a Tag/Device...

	Device Found

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

	Device Lost

Waiting for a Tag/Device...

Leaving ...
```

Device B
```
pi@orange-pi2:~/linux_libnfc-nci $ nfcDemoApp push --type=mime -m xyz/123 -d 6162636465 2> 15-device-b.txt
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Push mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found
		Push successful
		Push Failed
	Device Lost

Waiting for a Tag/Device...

	Device Found
		Push Failed
		Push Failed
		Push Failed
		Push Failed
		Push Failed
	Device Lost

Waiting for a Tag/Device...

Leaving ...
```

## git diff
```
diff --git a/conf/libnfc-nci.conf b/conf/libnfc-nci.conf
index 1b87d7a..845114b 100644
--- a/conf/libnfc-nci.conf
+++ b/conf/libnfc-nci.conf
@@ -2,8 +2,8 @@
 
 ###############################################################################
 # Log levels for libnfc-nci. Suggested value for debugging is 0xFF.
-APPL_TRACE_LEVEL=0x00
-PROTOCOL_TRACE_LEVEL=0x00
+APPL_TRACE_LEVEL=0xFF
+PROTOCOL_TRACE_LEVEL=0xFF
 
 ###############################################################################
 # Specify HOST to listen for a selected protocol
diff --git a/conf/libnfc-nxp-init.conf b/conf/libnfc-nxp-init.conf
index 9079dda..5bc32fb 100644
--- a/conf/libnfc-nxp-init.conf
+++ b/conf/libnfc-nxp-init.conf
@@ -19,13 +19,13 @@
 #  NXPLOG_WARN_LOGLEVEL       0x02
 #  NXPLOG_ERROR_LOGLEVEL      0x01
 #  NXPLOG_SILENT_LOGLEVEL     0x00
-NXPLOG_GLOBAL_LOGLEVEL=0x00
-NXPLOG_EXTNS_LOGLEVEL=0x00
-NXPLOG_NCIHAL_LOGLEVEL=0x00
-NXPLOG_NCIX_LOGLEVEL=0x00
-NXPLOG_NCIR_LOGLEVEL=0x00
-NXPLOG_FWDNLD_LOGLEVEL=0x00
-NXPLOG_TML_LOGLEVEL=0x00
+NXPLOG_GLOBAL_LOGLEVEL=0x03
+NXPLOG_EXTNS_LOGLEVEL=0x03
+NXPLOG_NCIHAL_LOGLEVEL=0x03
+NXPLOG_NCIX_LOGLEVEL=0x03
+NXPLOG_NCIR_LOGLEVEL=0x03
+NXPLOG_FWDNLD_LOGLEVEL=0x03
+NXPLOG_TML_LOGLEVEL=0x03
 
 ###############################################################################
 # NXP HW Device Node information, when pn5xx_i2c kernel driver configuration is used
diff --git a/demoapp/main.c b/demoapp/main.c
index e98ada2..45c735c 100644
--- a/demoapp/main.c
+++ b/demoapp/main.c
@@ -1406,6 +1406,10 @@ int WaitDeviceArrival(int mode, unsigned char* msgToSend, unsigned int len)
                 if(2 == mode)
                 {
                     SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
                 }
                 
                 framework_LockMutex(g_SnepClientLock);
```

jgeslin, 2019-01-22:
Hi,

Can you please do a new trace (only device 'a' is useful at this point) enabling one more trace ? 
In src\libnfc-nci\include\bt_trace.h:
`define BT_TRACE_VERBOSE    TRUE`

Regards,
Jeremy.

orange-kao, 2019-01-22:
Hi Jeremy,

Debug log as attached. Thank you.

## Debug log (stderr)
- [16-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2784859/16-device-a.txt)
- [16-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2784860/16-device-b.txt)

## Console output (stdout)
Device A
```
pi@orange-pi1:~/linux_libnfc-nci $ nfcDemoApp poll 2> 16-device-a.txt
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

	Device Lost

Waiting for a Tag/Device...

	Device Found

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

	Device Lost

Waiting for a Tag/Device...

Leaving ...
```

Device B
```
pi@orange-pi2:~/linux_libnfc-nci $ nfcDemoApp push --type=mime -m xyz/123 -d 6162636465 2> 16-device-b.txt
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Push mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found
		Push successful
		Push Failed
		Push Failed
		Push Failed
	Device Lost

Waiting for a Tag/Device...

	Device Found
		Push Failed
		Push Failed
		Push Failed
		Push Failed
		Push Failed
	Device Lost

Waiting for a Tag/Device...

Leaving ...
```

## git diff
```
diff --git a/conf/libnfc-nci.conf b/conf/libnfc-nci.conf
index 1b87d7a..845114b 100644
--- a/conf/libnfc-nci.conf
+++ b/conf/libnfc-nci.conf
@@ -2,8 +2,8 @@
 
 ###############################################################################
 # Log levels for libnfc-nci. Suggested value for debugging is 0xFF.
-APPL_TRACE_LEVEL=0x00
-PROTOCOL_TRACE_LEVEL=0x00
+APPL_TRACE_LEVEL=0xFF
+PROTOCOL_TRACE_LEVEL=0xFF
 
 ###############################################################################
 # Specify HOST to listen for a selected protocol
diff --git a/conf/libnfc-nxp-init.conf b/conf/libnfc-nxp-init.conf
index 9079dda..5bc32fb 100644
--- a/conf/libnfc-nxp-init.conf
+++ b/conf/libnfc-nxp-init.conf
@@ -19,13 +19,13 @@
 #  NXPLOG_WARN_LOGLEVEL       0x02
 #  NXPLOG_ERROR_LOGLEVEL      0x01
 #  NXPLOG_SILENT_LOGLEVEL     0x00
-NXPLOG_GLOBAL_LOGLEVEL=0x00
-NXPLOG_EXTNS_LOGLEVEL=0x00
-NXPLOG_NCIHAL_LOGLEVEL=0x00
-NXPLOG_NCIX_LOGLEVEL=0x00
-NXPLOG_NCIR_LOGLEVEL=0x00
-NXPLOG_FWDNLD_LOGLEVEL=0x00
-NXPLOG_TML_LOGLEVEL=0x00
+NXPLOG_GLOBAL_LOGLEVEL=0x03
+NXPLOG_EXTNS_LOGLEVEL=0x03
+NXPLOG_NCIHAL_LOGLEVEL=0x03
+NXPLOG_NCIX_LOGLEVEL=0x03
+NXPLOG_NCIR_LOGLEVEL=0x03
+NXPLOG_FWDNLD_LOGLEVEL=0x03
+NXPLOG_TML_LOGLEVEL=0x03
 
 ###############################################################################
 # NXP HW Device Node information, when pn5xx_i2c kernel driver configuration is used
diff --git a/demoapp/main.c b/demoapp/main.c
index e98ada2..45c735c 100644
--- a/demoapp/main.c
+++ b/demoapp/main.c
@@ -1406,6 +1406,10 @@ int WaitDeviceArrival(int mode, unsigned char* msgToSend, unsigned int len)
                 if(2 == mode)
                 {
                     SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
+                    SnepPush(msgToSend, len);
                 }
                 
                 framework_LockMutex(g_SnepClientLock);
diff --git a/src/libnfc-nci/include/bt_trace.h b/src/libnfc-nci/include/bt_trace.h
index d9c58dd..93551ea 100644
--- a/src/libnfc-nci/include/bt_trace.h
+++ b/src/libnfc-nci/include/bt_trace.h
@@ -231,7 +231,7 @@ EXPORT_API extern void BTTRC_StackTrace6(tBTTRC_LAYER_ID layer_id,
 
 /* Enables or disables verbose trace information. */
 #ifndef BT_TRACE_VERBOSE
-#define BT_TRACE_VERBOSE    FALSE
+#define BT_TRACE_VERBOSE    TRUE
 #endif
 
 /* Enables or disables all trace messages. */
```

## Build with
```
./bootstrap
./configure --prefix=/usr --enable-alt
make
sudo make install
sudo cp conf/* /etc/
```

jgeslin, 2019-01-23:
I'm still unable to reproduce the issue, although using the same setup.
Please run again with attached updated files containing more traces: [llcp.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2787346/llcp.zip)

orange-kao, 2019-01-23:
Hi Jeremy,

Debug log as attached. Two rounds with 60 seconds each.
Thank you.

## Debug log (stderr)
- [17-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2787469/17-device-a.txt)
- [17-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2787470/17-device-b.txt)
- [18-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2787471/18-device-a.txt)
- [18-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2787472/18-device-b.txt)

## Console output (stdout)
Device A (round 17)
```
pi@orange-pi1:~/linux_libnfc-nci $ nfcDemoApp poll 2> 17-device-a.txt
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

	Device Lost

Waiting for a Tag/Device...

	Device Found
	Device Lost

Waiting for a Tag/Device...

	Device Found
	Device Lost

Waiting for a Tag/Device...

	Device Found

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

	Device Lost

Waiting for a Tag/Device...

	Device Found
	Device Lost

Waiting for a Tag/Device...
[... 411 lines cut ...]
     /* nothing to send */
     *p_next_pdu_length = 0;
+    LLCP_TRACE_DEBUG0 ("llcp_link_get_next_pdu end, returning NULL");
     return NULL;
 }
 
@@ -1985,7 +1996,7 @@ static BT_HDR *llcp_link_build_next_pdu (BT_HDR *p_pdu)
     UINT8  *p, ptype;
     UINT16  next_pdu_length, pdu_hdr;
 
-    LLCP_TRACE_DEBUG0 ("llcp_link_build_next_pdu ()");
+    LLCP_TRACE_DEBUG1 ("llcp_link_build_next_pdu (0x%x)", p_pdu);
 
     /* add any pending SNL PDU into sig_xmit_q for transmitting */
     llcp_sdp_check_send_snl ();
```

## build with
```
./bootstrap
./configure --prefix=/usr --enable-alt
make
sudo make install
sudo cp conf/* /etc/
```

jgeslin, 2019-01-24:
Additional traces enabled in attached package: [SNEP.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2791204/SNEP.zip)
Also add a potential fix ... let see.
Please run it and provide device-a.txt log.

orange-kao, 2019-01-24:
Hi Jeremy,

Debug log for running 60 seconds as attached.
Thank you.

## Debug log (stderr)
- [19-device-a.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2791533/19-device-a.txt)
- [19-device-b.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2791534/19-device-b.txt)

## Console log (stdout)
Device A
```
pi@orange-pi1:~/linux_libnfc-nci $ nfcDemoApp poll 2> 19-device-a.txt
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

		NDEF Message Received : 

		TNF Media

			Type : xyz/123
			Data : abcde

		15 bytes of NDEF data received :
		D2 07 05 78 79 7A 2F 31 32 33 61 62 63 64 65 

	Device Lost

Waiting for a Tag/Device...

	Device Found
	Device Lost

Waiting for a Tag/Device...

Leaving ...
```

Device B
```
pi@orange-pi2:~/linux_libnfc-nci $ nfcDemoApp push --type=mime -m xyz/123 -d 6162636465 2> 19-device-b.txt
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Push mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found
		Push successful
		Push successful
		Push successful
		Push successful
		Push successful
	Device Lost

Waiting for a Tag/Device...

	Device Found
		Push Failed
		Push Failed
		Push Failed
		Push Failed
		Push Failed
	Device Lost

Waiting for a Tag/Device...

Leaving ...
```

jgeslin, 2019-01-24:
It looks the first issue is gone with the fix !
You are indeed able to successfully push 5 times the NDEF message:
```
	Device Found
		Push successful
		Push successful
		Push successful
		Push successful
		Push successful
```
Then LLCP symmetry frames are exchange to detect peer removal, triggering device departure notification and restart of the discovery:
```
	Device Lost

Waiting for a Tag/Device...
```
The device is rediscovered again:
```
	Device Found
```
But the first push fails because of RF transmission error:
```
		Push Failed
```
In this condition you should not try to push again, but just wait for peer removal and re-discovery.

To improve a bit the communication stability, you could add a spacer (about half centimeter to 1 centimeter in between the 2 antennas.

jgeslin, 2019-01-25:
The push fails because of a transmission error detection. However the symmetry is still running between the 2 devices which prevents restarting the communication.
When the push fails, I would suggest to restart the discovery:
```
                nfcManager_disableDiscovery();
                nfcManager_enableDiscovery(DEFAULT_NFA_TECH_MASK, 0x00, 0, 1);  
```

orange-kao, 2019-01-26:
Hi Jeremy,

Thank you. Restarting the discovery works, git diff as below. I've run the test for 90 minutes and it did not stuck there anymore.

The first few lines of message from device B looks like

```
Waiting for a Tag/Device...

	Device Found
		Push success (count 1)
		Push Failed
	Device Lost

Waiting for a Tag/Device...

	Device Found
		Push success (count 50)
		Push Failed
	Device Lost

Waiting for a Tag/Device...

	Device Found
		Push success (count 7)
		Push Failed
	Device Lost

Waiting for a Tag/Device...

	Device Found

		Push Failed
	Device Lost
```

However, I noticed that the number of successive SNEP push varies. Based on the message above, it is 1, 50, 7 and 0 between device found/lost messages. The number of SNEP successive push could range from zero (no success push) to 2840 successive push. I have converted the messages from device B into an array (of 859 elements from 90 minutes test), [21-successive-push-array.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2798363/21-successive-push-array.txt)

And I did some statistics based on the array. As below.
- 0 successful push: 10.71%
- 1~5 successive push: 28.64%
- 6~10 successive push: 20.02%
- 11~20 successive push: 21.30%
- 21~40 successive push: 12.46%
- 41~100 successive push: 4.66%
- 101~1000 successive push: 1.16%
- 1001~2840 successive push: 1.05%

It seems that the communication between the two devices is not stable. Does there any way to improve this?

Thank you.

## Debug log (stderr)
The text files are 1.2GB and 1.7GB respectively. It's been compressed with XZ to reduce the size to 2.0MB and 2.3MB respectively, and compressed again with ZIP to be uploaded to GitHub.
- [21-device-a.txt.xz.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2798426/21-device-a.txt.xz.zip)
- [21-device-b.txt.xz.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2798428/21-device-b.txt.xz.zip)

## git diff demoapp/main.c
```
diff --git a/demoapp/main.c b/demoapp/main.c
index e98ada2..5a70261 100644
--- a/demoapp/main.c
+++ b/demoapp/main.c
@@ -695,11 +695,11 @@ int SnepPush(unsigned char* msgToPush, unsigned int len)
         
         if(0x00 != res)
         {
-            printf("\t\tPush Failed\n");
+/*            printf("\t\tPush Failed\n");*/
         }
         else
         {
-            printf("\t\tPush successful\n");
+/*            printf("\t\tPush successful\n");*/
         }
     }
     else
@@ -1158,6 +1158,7 @@ int WaitDeviceArrival(int mode, unsigned char* msgToSend, unsigned int len)
     short unsigned int HCEResponseLen = 0x00;
     int tag_count=0;
     int num_tags = 0;
+    int push_success_counter = 0;
     
     nfc_tag_info_t TagInfo;
     
@@ -1405,7 +1406,19 @@ int WaitDeviceArrival(int mode, unsigned char* msgToSend, unsigned int len)
                 
                 if(2 == mode)
                 {
-                    SnepPush(msgToSend, len);
+                    push_success_counter = 0;
+                    while(1){
+                        if (SnepPush(msgToSend, len) != 0){
+                            printf("\n");
+                            printf("\t\tPush Failed\n");
+                            break;
+                        }
+                        push_success_counter++;
+                        printf("\r\t\tPush success (count %d)", push_success_counter);
+                        fflush(stdout);
+                    }
+                    nfcManager_disableDiscovery();
+                    nfcManager_enableDiscovery(DEFAULT_NFA_TECH_MASK, 0x00, 0, 1);
                 }
                 
                 framework_LockMutex(g_SnepClientLock);

```

jgeslin, 2019-01-28:
Hi,

Please try the attached fix which could prevent rediscovering the perr device in case of erro which could be recovered: [llcp_fix.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2802529/llcp_fix.zip)

Regards,
Jeremy.

orange-kao, 2019-01-28:
Hi Jeremy.

It is much more reliable. Thank you.

I run the test twice, with 120 seconds each. Round 22 is very stable with 925 successive push with no push failure. However, round 23 have multiple push failure, and successive SNEP push range from 6 to 547.

It seems that the stability issue is still there, as the successive push could as low as 6. Does there any way to make it better?

I have modified the code as [22-extra-debug-message.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2803411/22-extra-debug-message.zip) to include more debug message and make the stdout of device A more readable. Console log and debug log as below.

Thank you.

## Round 22 - 925 successive push in 120 seconds.
- [22-device-a.txt.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2803430/22-device-a.txt.zip)
- [22-device-b.txt.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2803431/22-device-b.txt.zip)

### Device A stdout
```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found
		NDEF Message Received (count 926)    
	Device Lost

Leaving ...
```

### Device B stdout
```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Push mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found
		Push success (count 925)
		Push Failed (return 3)
	Device Lost

Waiting for a Tag/Device...

Leaving ...
```

## Round 23 - 41+6+182+114+547 push in 120 seconds
- [23-device-a.txt.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2803432/23-device-a.txt.zip)
- [23-device-b.txt.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2803433/23-device-b.txt.zip)

### Device A stdout
```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found
		NDEF Message Received (count 41)    	Device Lost

Waiting for a Tag/Device...

	Device Found
		NDEF Message Received (count 6)    	Device Lost

Waiting for a Tag/Device...

	Device Found
		NDEF Message Received (count 182)    	Device Lost

Waiting for a Tag/Device...

	Device Found
		NDEF Message Received (count 114)    	Device Lost

Waiting for a Tag/Device...

	Device Found
		NDEF Message Received (count 547)    
	Device Lost

Leaving ...
```

### Device B stdout
```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Push mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	Device Found
		Push success (count 41)
		Push Failed (return 3)
	Device Lost

Waiting for a Tag/Device...

	Device Found
		Push success (count 6)
		Push Failed (return 3)
	Device Lost

Waiting for a Tag/Device...

	Device Found
		Push success (count 182)
		Push Failed (return 3)
	Device Lost

Waiting for a Tag/Device...

	Device Found
		Push success (count 114)
		Push Failed (return 3)
	Device Lost

Waiting for a Tag/Device...

	Device Found
		Push success (count 547)
		Push Failed (return 3)
	Device Lost

Waiting for a Tag/Device...

Leaving ...
```

orange-kao, 2019-01-31:
Hi.

I found a way to improve the P2P reliability by using passive mode instead of active mode.

I've run the test for 120 seconds under different mode (NFC-A/NFC-F and active/passive) by modifying `POLLING_TECH_MASK` and `P2P_LISTEN_TECH_MASK` in `libnfc-nci.conf`. I try different distances between the antennas as well. In all the cases, passive mode is more stable and have a better performance than active mode. Test results as tables below.

Based on my observation, I have a few questions.
1. Why passive mode always perform better than active mode?
2. During the test (not included in the test result below), I noticed that if the antenna gets too close (7.7mm), the stability will be reduced sometimes. Why?
3. Does there any way to adjust transmission power to get a better performance when two antenna gets very close or very far away?
4. Does there any way to read the signal strength, signal quality or other RF parameters from PN7150 for troubleshooting purpose?

Thank you.

## 7.7mm between two antennas
| Mode | Number of<br/>success SNEP push<br/>message on device B | Number of<br/>device lost/found<br/>message on device B | Comment |
| :--- | ---: | ---: | --- |
| A passive | 1997 | 1 | good performance, very stable |
| F passive | 2235 | 1 | good performance, very stable |
| A active | 1605 | 50 | good performance, not stable |
| F active | 1898 | 22 | good performance, not stable |

## 30mm between two antennas
| Mode | Number of<br/>success SNEP push<br/>message on device B | Number of<br/>device lost/found<br/>message on device B | Comment |
| :--- | ---: | ---: | --- |
| A passive | 1776 | 7 | good performance, stable |
| F passive | 1850 | 5 | good performance, stable |
| A active | 12 | 272 | poor performance, very unstable |
| F active | 30 | 219 | poor performance, very unstable |

## 50mm between two antennas
| Mode | Number of<br/>success SNEP push<br/>message on device B | Number of<br/>device lost/found<br/>message on device B | Comment |
| :--- | ---: | ---: | --- |
| A passive | 2266 | 1 | good performance, very stable |
| F passive | 2223 | 1 | good performance, very stable |
| A active | 0 | 695 | unusable |
| F active | 0 | 81 | unusable |

## 75mm between two antennas
| Mode | Number of<br/>success SNEP push<br/>message on device B | Number of<br/>device lost/found<br/>message on device B | Comment |
| :--- | ---: | ---: | --- |
| A passive | 1945 | 1 | good performance, very stable |
| F passive | 1952 | 1 | good performance, very stable |
| A active | 0 | 0 | unusable |
| F active | 0 | 0 | unusable |

## Configuration

### A passive
```
POLLING_TECH_MASK=0x01
P2P_LISTEN_TECH_MASK=0x01
```

### F passive
```
POLLING_TECH_MASK=0x04
P2P_LISTEN_TECH_MASK=0x04
```

### A active
```
POLLING_TECH_MASK=0x40
P2P_LISTEN_TECH_MASK=0x40
```

### F active
```
POLLING_TECH_MASK=0x80
P2P_LISTEN_TECH_MASK=0x80
```

jgeslin, 2019-01-31:
Hi,

Indeed I was willing to suggest you to use passive mode instead of active mode.

1. The demokit has been optimized for passive communication (because it is same settings than for reader and card emulation modes) rather than for active mode. Usually P2P communication don't need to be stable across more than a couple of seconds since usual use case implies limited data exchange (for bigger data exchange NFC is usually used to perform connection handover to another RF technology like BT or WIFI).

2. The 2 demo kits having the exact same antenna, positioning them too close increase coupling effect inducing de-tuning of the antenna. This explain lower performance with lower distance. You could improve it by dis-aligning the antennas one against the other.

3. and 4. You can refer to https://www.nxp.com/docs/en/application-note/AN11755.pdf chapter 5.3 for guidelines to improve communication.

Regards,
Jeremy.

jgeslin, 2019-04-01:
Closed because considered answered and no comment since almost 1 year.

closed 2019-04-01 by jgeslin