NXPNFCLinux/linux_libnfc-nci #82: nfcDemoApp cannot push (P2P) message with 3-byte MIME and 1016-byte data

orange-kao, 2018-05-30:
## Reproduce steps
1. Get two pieces of Raspberry Pi (unit A and B), and install OM5578/PN7150RPI on it.
2. Put two pieces of OM5578/PN7150RPI facing each other (less than 1cm between two antenna)
3. Issue "nfcDemoApp poll" on unit A
4. Issue command "nfcDemoApp push --type=mime -m x/x -d ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff" on unit B
(There are 2032 "f", which is 1016 bytes of 0xff in step 4)

# Expected results
Unit A receives NDEF from unit B.

# Actual result
Unit A print
```
Waiting for a Tag/Device...

	Device Found
```

Unit B print
```
Failed to build MIME NDEF Message
Waiting for a Tag/Device...

	Device Found
		Push Failed
```

# Additional information
Reproducibility: 100%

There is an error message "Failed to build MIME NDEF Message" on unit B.
It seems that ndef_createMime() will return non-zero if (1+1+4+(size of MIME) + (size of data)) >= 1025.
A 4-byte MIME and 1015-byte data can also reproduce the issue.

- 3-byte MIME and 1015-byte data: unreproducible (1+1+4+3+1015 == 1024)
- 3-byte MIME and 1016-byte data: reproducible   (1+1+4+3+1016 == 1025)
- 4-byte MIME and 1014-byte data: unreproducible (1+1+4+4+1014 == 1024)
- 4-byte MIME and 1015-byte data: reproducible   (1+1+4+4+1015 == 1025)

jgeslin, 2018-06-04:
Hi,

Indeed, there is a fixed limitation in the libnfc-nci stack @1024 bytes.
Please try the attached patch : [patch_LargeNdefMime.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/2067540/patch_LargeNdefMime.txt)

Regards,
Jeremy.

jgeslin, 2018-08-13:
Latest release R2.4 now includes the fix.

closed 2018-11-07 by jgeslin