NXPNFCLinux/linux_libnfc-nci #83: nfcDemoApp poll: errno : 6

XianglongLu, 2018-06-04:
Hi sir.

I set up the nfc environment on my embedded linux (kernel 4.4.103) product by using the "chapter 3.2 Alternative to pn5xx_i2c kernel driver" way in AN11697 file.  But I run "nfcDemoApp poll" fail.
the debug log is as follow:
# nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

0:000:176 - NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC
0:000:306 - NxpTml:     NFC - Assign IO pins
0:000:659 - NxpTml:     Pin /sys/class/gpio/gpio1066
0:001:137 - NxpTml:     System already has pin /sys/class/gpio/gpio1066
0:001:652 - NxpTml:     Direction /sys/class/gpio/gpio1066/direction
0:002:270 - NxpTml:     Pin 1066 now an input
0:002:553 - NxpTml:     Edge /sys/class/gpio/gpio1066/edge
0:003:036 - NxpTml:     Edge-string rising - 6
0:003:486 - NxpTml:     Pin 1066 trigger on rising
0:003:796 - NxpTml:     Value /sys/class/gpio/gpio1066/value
0:004:302 - NxpTml:     Pin /sys/class/gpio/gpio1067
0:004:709 - NxpTml:     System already has pin /sys/class/gpio/gpio1067
0:005:263 - NxpTml:     Direction /sys/class/gpio/gpio1067/direction
0:005:828 - NxpTml:     Pin 1067 now an output
0:006:204 - NxpTml:     Pin 1067 now off
0:006:498 - NxpTml:     NFCHW - open I2C bus - /dev/i2c-1
0:006:986 - NxpTml:     NFC - open I2C device - 0x28
0:207:888 - NxpTml:     PN54X - Tml Reader Thread Started................
0:208:087 - NxpTml:     PN54X - Read requested.....
0:208:493 - NxpTml:     PN54X - Invoking I2C Read.....
0:208:037 - NxpTml:     PN54X - Tml Writer Thread Started................
0:209:395 - NxpTml:     _i2c_read() [hdr] errno : 6
0:209:901 - NxpTml:     PN54X - Error in I2C Read.....
0:210:338 - NxpTml:     PN54X - Read requested.....
0:210:747 - NxpTml:     PN54X - Invoking I2C Read.....
0:209:506 - NxpTml:     PN54X - Tml Writer Thread Running................
0:211:760 - NxpTml:     PN54X - Write requested.....
0:212:180 - NxpTml:     PN54X - Invoking I2C Write.....
0:211:574 - NxpTml:     _i2c_read() [hdr] errno : 6
0:213:025 - NxpTml:     PN54X - Error in I2C Read.....
0:213:457 - NxpTml:     PN54X - Read requested.....
0:213:864 - NxpTml:     PN54X - Invoking I2C Read.....
0:213:008 - NxpTml:     _i2c_write() errno : 6
0:214:684 - NxpTml:     PN54X - Error in I2C Write.....
0:215:106 - NxpTml:     PN54X - Posting Fresh Write message.....
0:214:868 - NxpTml:     _i2c_read() [hdr] errno : 6
0:216:012 - NxpTml:     PN54X - Error in I2C Read.....

at the same time I cannot find my i2c device with the following log. (default i2c add is 0x28)
In "chapter 3.2 Alternative to pn5xx_i2c kernel driver" of AN11697 file, it did not ask me to create the i2c device node. It means that the nfcDemoApp would create it when running? 
How to debug this problem? could you help me ?

# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

jgeslin, 2018-06-04:
Hi,

Indeed, "errno: 6" indicates no device at address 0x28.
Let's first verify the connection without the libnfc-nci stack.

From my mind this is probably due to the level of VEN signal: you must insure it is high. Please check it with a scope or voltage meter. You should be able to set it with such command:
`$ echo 1 > /sys/class/gpio/gpio1067/value`
If not it could mean that the gpio number assigned to VEN signal is wrong.

Then, using i2cdetect tool to check the presence of PN7150, you must run it 2 times in less than 1 second because PN7150 integrates a sleep mode preventing it answering to the first request (first request wake-up the PN7150, but it falls back in sleep mode after 1 second of inactivity).

Regards,
Jeremy.

XianglongLu, 2018-06-04:
@jgeslin  Hi sir.  thank you for your help. now it works. but after running for a wait, the linux system would dead. the only thing i can do is push the reboot hw button to reboot.
the log is as follow；
# nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

Carmachine: Update media resource.
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
NFC Tag Found

Type : 'Type A - Mifare Ul'
NFCID1 :    '04 8A 4A D2 9C 39 80 '
Record Found :
NDEF Content Max size : '868 bytes'
NDEF Actual Content size : '29 bytes'
ReadOnly :                      'FALSE'
Type : 'URI'
URI : 'http://www.nxp.com/demoboard/OM5578'

29 bytes of NDEF data received :
D1 01 19 55 01 6E 78 70 2E 63 6F 6D 2F 64 65 6D 6F 62 6F 61 72 64 2F 4F 4D 35 35 37 38 

NFC Tag Lost

Waiting for a Tag/Device...

NFC Tag Found

Type : 'Type A'
NFCID1 :    'CB A0 B0 B0 '
NDEF Content : NO, mode=1, tech=1

Not a MIFARE card
[  126.087326] mmc1: Timeout waiting for hardware interrupt.
[  136.116202] NMI watchdog: BUG: soft lockup - CPU#0 stuck for 22s! [kworker/0:1:103]
[  136.116880] Modules linked in:
[  136.117164] 
[  136.117313] CPU: 0 PID: 103 Comm: kworker/0:1 Not tainted 4.4.103 #5
[  136.117874] Hardware name: Rockchip RK3399 Excavator Board (Linux Opensource) (DT)
[  136.118552] Workqueue: events kernfs_notify_workfn
[  136.118987] task: ffffffc0ee10c100 task.stack: ffffffc0ee0c4000
[  136.119508] PC is at run_rebalance_domains+0x44/0x1ac
[  136.119957] LR is at __do_softirq+0x170/0x310
[  136.120345] pc : [<ffffff80080dae94>] lr : [<ffffff80080a0b14>] pstate: 60000145
[  136.120988] sp : ffffffc0f7edfe40
[  136.121282] x29: ffffffc0f7edfe40 x28: ffffff800921f000 
[  136.121766] x27: ffffff80090d5000 x26: 0000000000000000 
[  136.122249] x25: 0000000000000100 x24: ffffff800909b000 
[  136.122732] x23: ffffff80090d7000 x22: ffffffc0f7ee6080 
[  136.123215] x21: 0000000000000020 x20: ffffff80090a5080 
[  136.123698] x19: ffffff80090d5138 x18: 0000000000000001 
[  136.124180] x17: 0000007f9a7bb920 x16: ffffff8008116514 
[  136.124663] x15: 0000000000000000 x14: 0000000000000000 
[  136.125146] x13: 0000000000000000 x12: ffffff8008b98588 
[  136.125628] x11: 0000000000000004 x10: 0000000000000920 
[  136.126111] x9 : ffffffc0ee0c7d00 x8 : ffffffc0ee10ca80 
[  136.126593] x7 : ffffffc0ee10c100 x6 : ffffff800909fb80 
[  136.127076] x5 : 00000040eee41000 x4 : ffffff800925dd00 
[  136.127559] x3 : 00000040eee41000 x2 : 00000040eee41000 
[  136.128042] x1 : ffffff80090a5000 x0 : 0000000000000000 
[  136.128525] 
[... 57 lines cut ...]
[  136.165758] fbc0  ******** ******** ******** ******** ******** ******** ******** ********
[  136.166511] fbe0  ******** ******** ******** ******** ******** ******** ******** ********
[  136.167264] 
[  136.167264] X7: 0xffffffc0ee10c080:
[  136.167699] c080  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  136.168451] c0a0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  136.169203] c0c0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  136.169954] c0e0  a0800c72 c80c3b6c 01440081 00000000 00000007 22000100 eedd9ac8 ffffffd2
[  136.170706] c100  00000002 00000000 ffffffff ffffffff 00010100 00000000 00000000 00000000
[  136.171458] c120  ee0c4000 ffffffc0 00000002 04208060 00000000 00000000 00000000 00000000
[  136.172210] c140  00000001 00000000 00000001 00000000 fffd7d09 00000000 ee10db00 ffffffc0
[  136.172962] c160  00000000 00000001 00000078 00000078 00000078 00000000 08b986b8 ffffff80
[  136.173716] 
[  136.173716] X8: 0xffffffc0ee10ca00:
[  136.174151] ca00  00000000 00000000 00000000 00000000 00000001 00000000 00000000 00000000
[  136.174902] ca20  090dd150 ffffff80 ee10c100 ffffffc0 090dd150 ffffff80 0909b000 ffffff80
[  136.175655] ca40  ed1d4a80 ffffffc0 08b78010 ffffff80 f7ee6080 ffffffc0 00000000 00000000
[  136.176407] ca60  090d7000 ffffff80 ee10c6f8 ffffffc0 ee0c7d00 ffffffc0 ee0c7d00 ffffffc0
[  136.177159] ca80  08084e7c ffffff80 00000000 00000000 00000000 00000000 00000000 00000000
[  136.177911] caa0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  136.178662] cac0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  136.179413] cae0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  136.180165] 
[  136.180165] X9: 0xffffffc0ee0c7c80:
[  136.180599] 7c80  09104000 ffffff80 091042c0 ffffff80 ec041618 ffffffc0 eca91440 ffffffc0

jgeslin, 2018-06-04:
Hi,

Can you enable libnfc-nci traces, setting NXPLOG_xxx_LOGLEVEL to 0x03 in /etc/libnfc-nxp.conf file ?

Regards,
Jeremy.

XianglongLu, 2018-06-05:
Hi @jgeslin ,

I enable all NXPLOG_xxx_LOGLEVEL to 0x03, the log is as follow. It seems that when I run the "nfcDemoApp poll" for about 22 s, I would see this info "mmc1: Timeout waiting for hardware interrupt.". After then my linux would crash.

# nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

1358499103:482:866 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

1358499103:483:051 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

1358499103:483:559 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

1358499103:484:111 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

0:000:228 - NxpHal:     Entering phNxpNciHal_init_monitor
0:000:442 - NxpHal:     Returning with SUCCESS
0:000:914 - NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC

0:001:361 - NxpTml:     NFC - Assign IO pins

0:001:722 - NxpTml:     Pin /sys/class/gpio/gpio1066

0:002:194 - NxpTml:     Create pin /sys/class/gpio/gpio1066

0:103:289 - NxpTml:     Direction /sys/class/gpio/gpio1066/direction

0:103:629 - NxpTml:     Pin 1066 now an input

0:103:760 - NxpTml:     Edge /sys/class/gpio/gpio1066/edge

0:104:290 - NxpTml:     Edge-string rising - 6

0:104:692 - NxpTml:     Pin 1066 trigger on rising

0:105:009 - NxpTml:     Value /sys/class/gpio/gpio1066/value

0:105:538 - NxpTml:     Pin /sys/class/gpio/gpio1067

0:105:932 - NxpTml:     Create pin /sys/class/gpio/gpio1067

0:206:962 - NxpTml:     Direction /sys/class/gpio/gpio1067/direction

0:207:274 - NxpTml:     Pin 1067 now an output

0:207:512 - NxpTml:     Pin 1067 now off

0:207:763 - NxpTml:     NFCHW - open I2C bus - /dev/i2c-1

0:208:260 - NxpTml:     NFC - open I2C device - 0x28

0:409:393 - NxpTml:     PN54X - Tml Reader Thread Started................

0:409:557 - NxpHal:     thread started
0:409:574 - NxpTml:     PN54X - Read requested.....

0:410:283 - NxpTml:     PN54X - Invoking I2C Read.....

0:409:552 - NxpTml:     PN54X - Tml Writer Thread Started................

0:411:284 - NxpTml:     PN54X - Tml Writer Thread Running................

0:411:888 - NxpTml:     PN54X - Write requested.....

0:412:275 - NxpTml:     PN54X - Invoking I2C Write.....
[... 1990 lines cut ...]
[  156.166381] 5060  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  156.167133] 
[  156.167133] X4: 0xffffff800925dc80:
[  156.167568] dc80  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  156.168319] dca0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  156.169070] dcc0  00000000 00000000 20746f4e 6e696174 00646574 00000000 00000000 00000000
[  156.169822] dce0  00000000 00000000 00000000 00000000 090e8088 ffffff80 00000000 00000000
[  156.170573] dd00  000002c2 00000467 00002065 00000000 00000000 00000021 00000001 00000000
[  156.171324] dd20  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  156.172076] dd40  00000000 00003689 00000008 00000000 00000000 0000000f 00000000 00000000
[  156.172827] dd60  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  156.173579] 
[  156.173579] X7: 0xffffff8008b98500:
[  156.174013] 8500  00000000 00000000 00000000 00000000 00000000 00000000 080c4e84 ffffff80
[  156.174765] 8520  00000000 00000000 00000000 00000000 080cc070 ffffff80 080cc058 ffffff80
[  156.175517] 8540  00000000 00000000 00000000 00000000 00000000 00000000 080cc0d0 ffffff80
[  156.176269] 8560  080cc0bc ffffff80 080cc088 ffffff80 080cc0a4 ffffff80 00000000 00000000
[  156.177021] 8580  40200800 00000080 00000000 00000000 00082040 00000000 0c284860 00000001
[  156.177773] 85a0  2b4b6270 0000010f 4c627078 0002102d 00000000 000003ea 000007be 00000b7d
[  156.178525] 85c0  00000f28 000012be 00001641 000019b0 00001d0d 00002057 0000238f 000026b5
[  156.179278] 85e0  000029ca 00002cce 00002fc2 000032a6 0000357a 0000383e 00003af3 00003d99
[  156.180032] 
[  156.180032] X12: 0xffffff8008b98508:
[  156.180478] 8508  00000000 00000000 00000000 00000000 080c4e84 ffffff80 00000000 00000000
[  156.181230]

jgeslin, 2018-06-05:
Hi,

I doubt that the freeze is caused by the linux-libnfc stack itself.
From the logs, the only thing which looks weird are many traces:
> NxpTml: _i2c_read() [hdr] errno : 6

This tend to say that the Interrupt is not properly handled, so that's the IRQ line GPIO assignment is not valid. Usually, I2c read request should only occurs when IRQ line is HIGH, otherwise PN7150 have nothing to answer leading to read failure.

Regards,
Jeremy.

closed 2018-09-12 by XianglongLu